arendst/Tasmota #4489: 6.3.0 not working on Sonoff-Basic

MilosStanic, 2018-11-30:
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used : esptool
- [ ] Hardware used :  Sonoff Basic R2 v1.0
- [ ] If a pre-compiled release or development binary was used, which one? : 6.3.0
- [x ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
17:53:03 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Room"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:53:03 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:20","StartupUTC":"2018-11-29T16:52:43","Sleep":0,"BootCount":6,"SaveCount":11,"SaveAddress":"F9000"}}
17:53:03 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:53:03 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Digimet",""],"TelePeriod":300,"SetOption":["00008009","55A18000","00000000"]}}
17:53:03 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
17:53:03 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"Main","IPAddress":"192.168.1.46","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:0E:9C:79","Webserver":2,"WifiConfig":5}}
17:53:03 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E9C79","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:53:03 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 29 16:53:03 2018","Local":"Thu Nov 29 17:53:03 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:19","Sunset":"16:57"}}
17:53:03 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-29T17:53:03","Switch1":"ON"}}
17:53:03 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-29T17:53:03","Uptime":"0T00:00:20","Vcc":3.085,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Digimet","BSSId":"70:4D:7B:4C:F2:A8","Channel":6,"RSSI":68}}}
17:53:12 MQT: Attempting connection...

```
Just a heads up to say that 6.3.0 only works when connected via USB. When powered via mains, the Sonoff Basic doesn't connect to WiFi at all. 

Downgrading to 6.2.1 helped. 

Tried both sonof.bin and sonof-basic.bin -- same result.

blakadder, 2018-11-30:
Use the[ latest development .bin](http://thehackbox.org/tasmota/)... currently 6.3.0.14
There are known issues with the release version and some sonoff basics

Tommy-LSA, 2018-11-30:
complete your MQTT Settings. There is no host entered:

`17:53:03 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E9C79","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}`

andrethomas2, 2018-11-30:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-30 by andrethomas2

MilosStanic, 2018-11-30:
@blakadder thanks, I will try that. 
@Tommy-LSA I'm not using MQTT, so I don't know what I would set the "host" value to. Anyway, it works with the same config on 6.2.1 fw.

andrethomas, 2018-11-30:
If it worked on 6.2.1 it will work better on 6.3.0.14 :) Just make sure you have sleep set to 0 because it is no longer needed.

andrethomas, 2018-11-30:
@MilosStanic Also, you can use

`SetOption3 0`

To disable mqtt on the device then it will stop complaining about it :)

MilosStanic, 2018-12-01:
@andrethomas thanks for the SetOption3 tip. 
I haven't had the chance to test latest dev bins. I don't know where to find them. And for the time being, I like how 6.2.1 works.