arendst/Tasmota #4494: KNX: Add the GPIO_xxx_NP buttons

cybermaus, 2018-12-01:
Activating the KNX protocol (great feature!) I found that on Sonof I could select buttons, but on my Shelly1 I could not. The reason turned out that Shelly is configured with GPIO_SWT1_NP instead of GPIO_SWT1, and the KNX source is simply forgetting to enumerate those.

I modded my code locally, but am not setup for GIT. Someone please review and submit this change to the xdrv_11_knx.ino file

Basically I just duplicated the first 2 blocks below to also have the next 2 blocks.
I also tested, and it works for my shelly1

```
  for (int i = GPIO_SWT1; i < GPIO_SWT4 + 1; ++i)
  {
    if (GetUsedInModule(i, my_module.gp.io)) { device_param[i - GPIO_SWT1 + 8].show = true; }
  }
  for (int i = GPIO_KEY1; i < GPIO_KEY4 + 1; ++i)
  {
    if (GetUsedInModule(i, my_module.gp.io)) { device_param[i - GPIO_KEY1 + 8].show = true; }
  }
  **for (int i = GPIO_SWT1_NP; i < GPIO_SWT4_NP + 1; ++i)
  {
    if (GetUsedInModule(i, my_module.gp.io)) { device_param[i - GPIO_SWT1_NP + 8].show = true; }
  }
  for (int i = GPIO_KEY1_NP; i < GPIO_KEY4_NP + 1; ++i)
  {
    if (GetUsedInModule(i, my_module.gp.io)) { device_param[i - GPIO_KEY1_NP + 8].show = true; }
  }**
```

While doing this, I also feel the actual shelly1 definition should have GPIO0, 1, 3 in its definition. They are externally connectable, and usable. Not sure why tasmota has them on Shelly2, but not Shelly1:

```
  { "Shelly 1",         // Shelly1 Open Source (ESP8266 - 2MB) - https://shelly.cloud/shelly1-open-source/
     **GPIO_USER,         // Optional input, NOT SAFE : 3.3V negative referenced to MAINS HOT wire
     GPIO_USER,          // GPIO01 MCP39F501 Serial input : NOT SAFE : 3.3V negative referenced to MAINS HOT wire
     0,
     GPIO_USER,          // GPIO03 MCP39F501 Serial output : NOT SAFE : 3.3V negative referenced to** MAINS HOT wire
     GPIO_REL1,         // GPIO04 Relay (0 = Off, 1 = On)
     GPIO_SWT1_NP,      // GPIO05 SW pin
     0, 0, 0, 0, 0, 0,  // Flash connection
     0, 0, 0, 0, 0, 0
  },
```

ascillato, 2018-12-01:
Hi,

Thanks for pointing out the KNX enhancement. Will add it.

About the shelly, as it is very dangerous, we will not add GPIO_USER gpios for that. A newbie can be electrocuted trying to use that. This is the same case as Sonoff POW and other devices with power measurements.

cybermaus, 2018-12-01:
No prob. Thanks for picking it up.

Slightly related: Do you happen to know the functional difference between GPIO_SWTx and GPIO_SWTx_NP? I suspect it may be because there is No-Pull-up resistor in the Shelly1, so the software needs to set the internal pull-up?

closed 2018-12-01 by ascillato2