arendst/Tasmota #4496: Help for configuration

gabryk91, 2018-12-01:
Hi everybody,
I'm up to make a "smart" thermostat using my old non-smart thermostat, so I can place it away from the boiler.
The idea is: the thermostat relay will activate/deactivate sonoff 1 by acting on the gpio14 and gnd pin (I configured gpio14 as switch1 and set SwitchMode to 3 so it will replicate the pushbutton action), then through knx I will send the state of sonoff 1 to sonoff 2, which will be wired to the boiler (passing by 220v relay). SO when thermostat reach the temperature it will activate sonoff 1, then sonoff 2 will follow.

What I want is be able to use my thermostat programming (timing, temperature, etc) but also make Google start the boiler when I want, but my doubt is that with the actual setting things can go sideways (eg I fear that if I activated the sonoff 1 through google and then the thermostat sends the signal sonoff 1 will be deactivated)
Any advice? If I was unclear I'll try to explain it better, sorry

ascillato, 2018-12-01:
Hi,

You can start with the Thermostat example provided in rules:

https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#9-simple-thermostat-example

Then you can modify it and adapt to your needs.

If you need more help, remember that you can find us in the Tasmota Support Chat at: https://discord.gg/Ks2Kzd4

closed 2018-12-01 by ascillato2

gabryk91, 2018-12-02:
thank you @ascillato but the example is not for my case (much simpler) and the chat seems pretty dead, any wiki about how to configure and use user_config_override? On the rules page it says that I need to edit it but it don't say how and I can't find a reference for this. Thanks

ascillato, 2018-12-02:
The chat pretty dead? What? It is super active. There is hundreds of conversations every day solving issues and giving support.

Rules are enabled by default. You don't need to use any override for that. Neither compile by yourself. 

Just use the latest precompiled bin from the development branch 6.3.0.15