arendst/Tasmota #4502: Dimmer status/value not transmitted in MQTT when switching device on over button

nodus-dev, 2018-12-01:
I have uploaded the lastest Tasmota version to an ESP8266 dimmer switch from "Smart Life" (Which is a Tuya clone. The foto of the round dimmer on the Tuya wiki page describes it pretty well. just a newer version of the ESP module (flipped around).
All works well so far yet i encountered a problem when adding the item into Habpanel/SimpleUI. To get the item working i added a JS transformation to the incoming MQTT definition as there is no "Dimmer" key set in the MQTT return JSON when switching the dimmer off via its center pad/button.
This is not a problem when turning the dimmer off that way. Only when turning on as no Dimmer value is transmitted and all knobs and sliders stay at 0.
When pressing the dimmers on/off button the message in MQTT only reads { POWER : OFF } and likewise { POWER : ON }. But the key "Dimmer" is only set when dimming the device up or down (via the buttons or MQTT) { POWER : ON, Dimmer : 42 }

To reproduce you will need a dimmer module with a dedicatd on/off button.

Could i change it somewhere so it only sends the Dimmer value (0-100)? Or that it is included in all RESULT messages the dimmer propagates?

ascillato, 2018-12-01:
See commands setoption4 and setoption59

With those you can change the behaviour of the resulting mqtt message. 

You will need to use the latest version from the development branch (6.3.0.15)

If those command don't satisfy your needs, you need to use rules. 

Read the wiki for commands and also for rules. 

Thanks 

If you need further assistant please go to the Tasmota Support Chat

closed 2018-12-01 by ascillato2

Jason2866, 2018-12-02:
For help on OpenHab you can visit: https://community.openhab.org
What you want to do is possible in OpenHab with rules

nodus-dev, 2018-12-09:
Thx for the hint with the options! Quite interesting, though not what i was looking for.
And i was aware of the rule based solution. Just thought it might be possible to send the dimmer percentage along with power status changes as the dimmer knows to what setting it returns.