arendst/Tasmota #4505: Please Help! Connect problems for sonoff basic with Home Assistant

pjmsheff, 2018-12-02:
Please help. I am a complete novice on home assistant and sonoff. I have been trying to sort this out for over a week. It’s driving me crazy. I simply can't get the sonoff basic to talk to home assistant. I CAN control the sonoff from sonoff.local

Here is what I get from the sonoff console

Attempting connection...
09:58:09 MQT: Connect failed to brokerIP 192.168.1.112:1883, rc -2. Retry in 10 sec
09:58:20 MQT: Attempting connection...
09:58:26 MQT: Connect failed to brokerIP 192.168.1.112:1883, rc -2. Retry in 10 sec

Here is a reading from the information screen of sonoff settings

[image]

Here are the references in my configuration.yaml file

#  MQTT 
mqtt:
  broker: !secret my_mqtt_address
  port: !secret my_mqtt_port
  username: !secret my_mqtt_user
  password: !secret my_mqtt_pass
  
switch:
  platform: mqtt
  name: "SonoffNumber1"
  command_topic: "cmd/sonoffNo1/power"
  state_topic: "stat/sonoffNo1/POWER"
  payload_on: "ON"
  payload_off: "OFF"
  optimistic: false
  qos: 1
  retain: true

For information - In the configuration.yaml file I have defined the broker as 192.168.1.112 (the same address as my Home assistant.

I am sorry if this is an obvious problem but I have tried everything to try to make it work.

could it be a firewall issue. I have a BT smarthub 

Thanks in anticipation.

mihalski, 2018-12-02:
Hang on.. Have a look at your MQTT broker.. You have:
```
brokerIP 192.168.1.112
```
When it should just be:
```
192.168.1.112
```

pjmsheff, 2018-12-02:
Thanks for your quick replies. I will try them

pjmsheff, 2018-12-02:
WORKING!!!!!!

Thank you so much guys - I really do appreciate your help  Did both upgrade and change the broker issue .

GREAT GREAT GREAT

thanks again

closed 2018-12-02 by pjmsheff