arendst/Tasmota #4506: No IR control possible - "Protocol":"PANASONIC","Bits":48,"Data":218152113

bigshit11elf, 2018-12-02:
Hello, I have a problem sending out IR signals for different Panasonic devices.

My receiver (generic one from Amazon) is connected to a NodeMCU V2 ESP8266-12E board and is able to read signals from remotes. With other brands, I am able to successfully reproduce their IR signals and control those devices.

When I try to do the same with my DENON AV Receiver I get a 48bit Panasonic signal that - if reproduced - does nothing. The same goes for my Blu-Ray player. 48 bit signal that cannot be successfully reproduced, although (in both cases) Tasmota says "Irsend done."

I am using the latest development build and I already tried reading RAW IR data but that does not get fully read. (Console shows three dots "..." at the end of the line and does not complete the RAW recognition.)

I am stuck here: It is really easy to reproduce IR signals from other manufacturers, but I cannot get those Panasonic ones to work. Is there a solution?

Thanks for your replies in advance.

Jason2866, 2018-12-02:
Please fill out issue template. Without we can just guess...

Jason2866, 2018-12-02:
Do you know IR Diode is sending? Connecting IR Diode direct to a GPIO pin will NOT work.
See example how to: https://github.com/altelch/SonoffIR

Jason2866, 2018-12-02:
Please post the log from receiving good signals and from Panasonic

bigshit11elf, 2018-12-02:
IR diode is connected correctly via a transistor (2N2222). It does not draw power from the GPIO (which is connected via a base resistor to the transistor's base), the diode is connected to 5V Vin of the NodeMCU. Positive side has a resistor, negative side is connected via the 2N2222. 
I checked that it is sending IR signals via my mobile's camera - and: it is able to control other devices. 

I don't think it is a hardware issue...

bigshit11elf, 2018-12-02:
The raw data? Or the standard data?

Jason2866, 2018-12-02:
Both

bigshit11elf, 2018-12-02:
Can do that.

Here the logs:

Acer projector:
Good result, can be reproduced:

SetOption58 turned OFF:
14:02:18 MQT: tele/XXX/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":281551326}}

SetOption58 turned ON:
14:03:12 MQT: tele/XXX/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":281551326,"RawData":[9038,4414,612,514,594,514,594,514,594,1640,592,516,594,514,594,514,592,516,592,1640,594,1640,594,516,592,516,592,1642,592,516,592,516,592,516,592,516,594,514,592,1642,594,516,592,516,592,516,592,514,592,1642,592,1640,592,1644,592,516,592,1642,592,1642,592,1642,592,1644,592,508,590],"RawDataInfo":[67,67,0]}}

----------

Now the PANASONIC problem:
Bad result, cannot be reproduced:

SetOption58 turned OFF:
14:05:06 MQT: tele/XXX/RESULT = {"IrReceived":{"Protocol":"PANASONIC","Bits":48,"Data":218103821}}

SetOption58 turned ON:
14:04:24 MQT: tele/XXX/RESULT = {"IrReceived":{"Protocol":"PANASONIC","Bits":48,"Data":218103821,"RawData":[3564,1720,450,416,450,1284,450,416,448,416,448,418,474,390,450,390,474,414,450,416,472,392,448,402,464,416,450,390,476,1272,462,414,450,416,448,416,450,414,450,414,450,416,450,1286,450,1286,474,390,450,1284,450,416,450,416,450,416,450,416,448,416,450,416,450,416,450,414,450,414,450,414,450,416,450,416,448,416,450,416,448,416,450,416,450,416,448,416,448,392,474,414,450,1286,450,1284,450,416,450,1284,450], ...

(BTW: The raw data changes on each press of the exact same button... and does always end in these three dots. It never reaches the point where the "[XX,XX,X]" info is displayed.)

Jason2866, 2018-12-02:
So you have to extend IR buffer!

ascillato, 2018-12-02:
Hi,

Have you managed to solve your issue?