arendst/Tasmota #4507: RNX: Basic Button1 always sends "1" (and 2 more issues)

cybermaus, 2018-12-02:
Trying to make a 7-fold hotel switch, the broadcast of RNX is nice. But I am having some issues getting it active, and in the process I think I uncovered at least 2, maybe 3 bugs.

(BTW: I did get it working using websend rule, but coding a rule with 7 IP's on 7 devices in 7 different IP combinations is horrifying. Yeah broadcast)

Not sure if I should make it into 3 reports. Also not sure if I am maybe just doing something wrong. Anyway, 3 issues are these, more details in sections below:

1) Button1->2/2/1 always sends "1"
2) RNX_TX1->2/2/1 always sends 2-decimals (how to send an 1=On command?)
3) Output1->2/2/1 goes into a loop and hangs the entire network

Hardware: Sonoff Basic
PlatformIO with download of dev-sources from yesterday (6.3.0+latest devs)

```
13:01:48 CMD: status 0
13:01:48 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:01:48 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:25:41","StartupUTC":"2018-12-02T11:36:07","Sleep":0,"BootCount":2,"SaveCount":50,"SaveAddress":"FA000"}}
13:01:48 RSL: STATUS2 = {"StatusFWR":{"Version":"6.3.0.14(sonoff)","BuildDateTime":"2018-12-02T10:38:26","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:01:48 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PriveNet","SpeedTouchNR3974"],"TelePeriod":300,"SetOption":["0A008001","558180C0","00000000"]}}
13:01:48 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":502,"Free":500,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1420BA","FlashMode":3,"Features":["00000809","1FD06790","24038000","02B400C6","000028C0"]}}
13:01:48 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-2104","IPAddress":"192.168.190.69","Gateway":"192.168.190.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.190.1","Mac":"CC:50:E3:02:E8:38","Webserver":2,"WifiConfig":4}}
13:01:48 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 02 12:01:48 2018","Local":"Sun Dec 02 13:01:48 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:28","Sunset":"15:12"}}
13:01:48 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-12-02T13:01:48"}}
13:01:48 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-12-02T13:01:48","Uptime":"0T00:25:41","Vcc":3.161,"LoopSet":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":2,"SSId":"SpeedTouchNR3974","BSSId":"BC:05:43:13:53:B9","Channel":6,"RSSI":100}}}
```

1) Button1->2/2/1 always sends "1"
I noticed pressing the button always turns on the other sonoff, never off. Which is logical if you look in the log:
```
12:37:35 KNX: Button 1 = 1 sent to 2.2.1
12:37:35 RSL: RESULT = {"POWER":"OFF"}
12:37:35 RSL: POWER = OFF
12:37:46 KNX: Button 1 = 1 sent to 2.2.1
12:37:46 RSL: RESULT = {"POWER":"ON"}
12:37:46 RSL: POWER = ON
```
Of course, the button is a toggle button, that when released probably always ends up high, I would argue that either the "Button1->2/2/1" should evaluate the actual state (ideally), or if you are strict, then at the least send a "2" in stead of 1 as that means "toggle" (according to the rnx source)
So out of the 3 issues, I feel this one is most clearly a bug. The other two could be knowledge issues on my part.

2) RNX_TX1->2/2/1 always sends 2-decimals (how to send an 1=On command?)
Abandoning the button, I try to make a rule with RNX_TX1->2/2/1
```
rule1 
on Power1#State=1 do knxtx_val1 1 endon 
on Power1#State=0 do knxtx_val1 0 endon

or simply:

rule1
on Power1#State do knxtx_val1 %value% endon
```
Actually tried several variants of the rule, with %value%; "1"; '1'; %31;
Always this result in RNX_TX1 sending "1.00" instead of "1" and the other sonoff's do not respond:
```
13:31:57 RSL: RESULT = {"POWER":"ON"}
13:31:57 RSL: POWER = ON
13:31:57 RUL: POWER1#STATE performs "knxtx_val1 1"
13:31:57 KNX: KNX TX 1 = 1.00 sent to 2.2.1
13:31:57 RSL: RESULT = {"KnxTx_Val1":"1.00"}
13:32:02 RSL: RESULT = {"POWER":"OFF"}
13:32:02 RSL: POWER = OFF
13:32:02 RUL: POWER1#STATE performs "knxtx_val1 0"
13:32:02 KNX: KNX TX 1 = 0.00 sent to 2.2.1
13:32:02 RSL: RESULT = {"KnxTx_Val1":"0.00"}
```

3) Output1->2/2/1 goes into a loop and hangs the entire network
Both best, and worst, results I had with simply duplicating Output1->2/2/1 and no rules
When using UNCHECK communications enhancements it works, but sometimes not all other sonoff's follow (lost UDP packets)
When using CHECK communications enhancements it works, once, twice, three times.... But then, especially if I switch to quickly, it starts looping, rattling relays, until a ping monitor I have on the background shows the wifi itself is breaking down, probably due to a network storm.

I would probably (eventually) be able and willing to dig through the sources, I may be able to figure it out. But I felt either some feedback (am I doing something wrong?) or some pointers (where in the source?) could be useful.

 
Many thanks
Maurits

ascillato2, 2018-12-02:
Hi,

What is RNX? Tasmota supports only KNX protocol

KNX_TX is for sending values, that is why it has decimals. Also, it is sent as a float value, so all KNX devices will interpret it as a value not as a ON command. 

KNX_CMND is for sending commands

Please review the wiki at KNX support. There are examples of this. 
https://github.com/arendst/Sonoff-Tasmota/wiki/KNX-Features#5-using-rules

Buttons send a 1 because buttons always are for TOGGLE a relay. Buttons are not switches. Buttons are pushbuttons. You push, you send a toggle command.

ascillato2, 2018-12-02:
Buttons are meant to turn on scenes. If you want to sync devices, see the examples in the wiki please.

The loop issue is just because using the same group address for sending and receiving. KNX has a internal wait time to avoid this, so fast switching is ignored. 1 second or more, between turn on and off is only available.

ascillato2, 2018-12-02:
Have you managed to solve your issue?

If not, please provide all the configuration information for all devices and the exact solution you want in order to help you to configure it

cybermaus, 2018-12-02:
Thanks for checking up.
Had some social obligations this afternoon, and need to travel for work this week, so preparing that trip this evening. So it may need to wait till next weekend.

But I did test the  KNX_CMND at least, and that works. So item 2) definitely solved. Need to work on checking out the 1) and 3)

ascillato, 2018-12-02:
Ok, no problem. We can address this when you come back. It is just how to config this.

For you 1) buttons send a 1 because are meant for turning on scenes (scenes don't have off) So, what you want to achieve can be done using another options like send your output status or a rule.

For your 3) you need to use different group address for sending data and another for receiving data. Also fast switching is not supported in Tasmota KNX IP Multicast in favor of Multicast Enhancement.

When you have time, we can address your configuration to do what you need. You can just ask here or in the Tasmota Chat.

ascillato2, 2018-12-02:
Issue put to ON HOLD. Let's reopen when you have time. Thanks. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-02 by ascillato2

goofy79, 2023-11-10:
Is there any updates for issue 1 ? Because i have the same Problem, i alway get 1. 
I want to use the Button as an Status. So i need 0 and 1 (unpressed/pressed) 

Thank you

cybermaus, 2023-11-10:
Wow. Old stuff. I had completely forgotten about this. Anyway, my use case for KNX (syncing the switch-state for multiple tasmotas) was completely replaced since the later tarmota's have native multi-device group sync capabilities. 

However, I do have another trick I was using for something else, that may work for you: Configure the device with an extra "virtual" relay on an unconnected GPIO. (like here) and use the KNX to send the relay state as status.

Not sure if it will work, and no plans to check it, But maybe you can get somewhere with it.

ascillato, 2023-11-12:
@goofy79 

> Is there any updates for issue 1 ? Because i have the same Problem, i alway get 1.
> I want to use the Button as an Status. So i need 0 and 1 (unpressed/pressed)

Hello, by design a button will send a 1 when pressed and that's it. There isn't any message when the button is released.

If you need the button release, the idea from @cybermaus is a very good workaround. Hope it helps you use case.

goofy79, 2023-11-13:
Hi, thanks - can you explain a littlebit more how to configure a virtual relay and send send the status to the KNX ?
Thx

cybermaus, 2023-11-13:
As mentioned, I am not planning to test this, and check if it works at all. Just a hint if this may be useful.

But as a starter:

Go to templates settings, and set a unused GPIO to relay number 2 (or go to the website, and read all about templates). Next make a RULE to make sure that power1 state is always copied to power2 state (or go to the website, and read all about RULES).

And as to how make KNX send the state: It's been 5 year's, I have no idea how that works anymore, and all my devices are nowadays build without KNX, so I cannot check even if I wanted.

Reading the above back, I wonder why you would not simply send the original power1 state instead of the button, but I am assuming you already tried, that.

goofy79, 2023-11-13:
I have a pellet stove that I turn on and off using Shelly. And now I need the status - here I have a relay on the fan and this relay is attached to Button1. This means I have the status in KNX. Because it may be that I switch on the oven (using a button) and the pellets are empty, then of course the fan goes off and the oven goes off too. I want to get that as a status.

Noschvie, 2023-11-13:
Maybe you have to decouple the button from the relay. You are using the Shelly to turn on and off the stove and want to read back the fan using the input of the Shelly, isn't it?

goofy79, 2023-11-13:
I have already decoupled the button from the relay. And that's exactly how it is, I want to get feedback on the status of the fan.

Noschvie, 2023-11-13:
Have you tried to use rules like for a PIR sensor?

```
SwitchMode1 1
SwitchTopic 0
Rule1 on Switch1#state do knxtx_val1 %value% endon
Rule1 1
```

goofy79, 2023-11-13:
Not yet - I'll test it out when I get home this evening.

goofy79, 2023-11-13:
ok, my Switchmode1 was 13 i changed it to 1 ..  I'll test it out when I get home this evening.

goofy79, 2023-11-13:
> ok, my Switchmode1 was 13 i changed it to 1 .. I'll test it out when I get home this evening.

thanks to VPN .. i have tested it - it works excellently. Only the Chance "switchmode1 13" to "switchmode1 1" fix my problem. 
Thank you all