arendst/Tasmota #4508: Sonoff 4ch Pro r2 inching mode. unexpected turn on after reboot

netmaster01, 2018-12-02:
The problem is that during restart of the sonoff 4ch pro R2, the relays are turned on during the boot process. that is causing unwanted behavior with the garage door. Searched in the forums and find similar cases that were left behind, no more feedbacks from the users.

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
find similar problems on forums, related to mqtt broker and HA, remove HA configurations, create a new mqtt topic. using only web tasmota interface and the problem still perssits.

- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4) found info about ghost switches. never experience this. only during reboot proceess.

- [ x] Development/Compiler/Upload tools used : Atom Platfornio espressif8266 @ 1.7.0
- [x ] Hardware used : Sonoff 4ch pro R2
- [x] If a pre-compiled release or development binary was used, which one? :
Version 6.3.0.15(f3d625d-sonoff)-2_3_0
- [x] You have tried latest release or development binaries? :
yes, same problem
- [x ] Provide the output of command ``status 0`` :
```
14:37:59 WIF: Connected
14:38:09 CMD: status 0
14:38:09 SRC: WebConsole from 10.0.1.109
14:38:09 RSL: Received Topic /status, Data Size 1, Data 0
14:38:09 RSL: Group 0, Index 1, Command STATUS, Data 0
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoffgg1","Sonoffgg2","Sonoffgg3","Sonoffgg4"],"Topic":"sonoffgg1dgsdgsdgsdgds","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:54","StartupUTC":"2018-12-02T13:35:15","Sleep":50,"BootCount":550,"SaveCount":2273,"SaveAddress":"F5000"}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS2 = {"StatusFWR":{"Version":"6.3.0.15(f3d625d-sonoff)","BuildDateTime":"2018-12-02T12:00:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["NetXtreme","net-hack"],"TelePeriod":300,"SetOption":["00008008","558180C0","00000001"]}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS4 = {"StatusMEM":{"ProgramSize":501,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS5 = {"StatusNET":{"Hostname":"sonoffgg1dgsdgsdgsdgds-6488","IPAddress":"10.0.1.29","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"60:01:94:A3:59:58","Webserver":2,"WifiConfig":3}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A35958","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 02 13:38:09 2018","Local":"Sun Dec 02 14:38:09 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:23","Sunset":"16:55"}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS10 = {"StatusSNS":{"Time":"2018-12-02T14:38:09"}}
14:38:09 MQT: stat/sonoffgg1dgsdgsdgsdgds/STATUS11 = {"StatusSTS":{"Time":"2018-12-02T14:38:09","Uptime":"0T00:02:54","Vcc":3.196,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"NetXtreme","BSSId":"14:CC:20:8E:C9:E6","Channel":11,"RSSI":100}}}
14:38:20 WIF: Checking connection...
14:38:20 WIF: Connected
14:38:40 WIF: Checking connection...
14:38:40 WIF: Connected
```

New version installed, new mqtt topic created, first boot logs. all normal, no relays triggered in boot proccess, no hard button pressed, only via web interface

00:00:00 CFG: Loaded from flash at FB, Count 2266
00:00:00 CFG: Saved to flash at FB, Count 2267, Bytes 3584
00:00:00 Project sonoff Sonoffgg1 (Topic sonoffgg1dgsdgsdgsdgds, Fallback DVES_A35958, GroupTopic sonoffs) Version 6.3.0.15(f3d625d-sonoff)-2_3_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 NetXtreme in mode 11N as sonoffgg1dgsdgsdgsdgds-6488...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoffgg1dgsdgsdgsdgds-6488.local with IP address 10.0.1.29
00:00:05 CFG: Saved to flash at FA, Count 2268, Bytes 3584
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoffgg1dgsdgsdgsdgds/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoffgg1dgsdgsdgsdgds/POWER = 
00:00:05 MQT: Subscribe to cmnd/sonoffgg1dgsdgsdgsdgds/#
00:00:05 MQT: Subscribe to cmnd/sonoffs/#
00:00:05 MQT: Subscribe to cmnd/DVES_A35958/#
00:00:05 MQT: tele/sonoffgg1dgsdgsdgsdgds/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"6.3.0.15(f3d625d-sonoff)","FallbackTopic":"DVES_A35958","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/sonoffgg1dgsdgsdgsdgds/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffgg1dgsdgsdgsdgds-6488","IPAddress":"10.0.1.29"}
00:00:06 MQT: tele/sonoffgg1dgsdgsdgsdgds/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/POWER1 = OFF
[... 119 lines cut ...]
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoffgg1dgsdgsdgsdgds/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoffgg1dgsdgsdgsdgds/POWER = 
00:00:05 MQT: Subscribe to cmnd/sonoffgg1dgsdgsdgsdgds/#
00:00:05 MQT: Subscribe to cmnd/sonoffs/#
00:00:05 MQT: Subscribe to cmnd/DVES_A35958/#
00:00:06 MQT: tele/sonoffgg1dgsdgsdgsdgds/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"6.3.0.15(f3d625d-sonoff)","FallbackTopic":"DVES_A35958","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/sonoffgg1dgsdgsdgsdgds/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffgg1dgsdgsdgsdgds-6488","IPAddress":"10.0.1.29"}
00:00:06 MQT: tele/sonoffgg1dgsdgsdgsdgds/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/POWER1 = OFF
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/RESULT = {"POWER2":"OFF"}
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/POWER2 = OFF
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/RESULT = {"POWER3":"OFF"}
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/POWER3 = OFF
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/RESULT = {"POWER4":"OFF"}
00:00:06 MQT: stat/sonoffgg1dgsdgsdgsdgds/POWER4 = OFF
00:00:07 APP: (UTC) Sun Dec 02 13:35:22 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
14:35:23 HTP: Console
14:35:25 APP: Boot Count 550
14:35:25 CFG: Saved to flash at F5, Count 2273, Bytes 3584

if you create a new topic and don't activate relay the reboot proccess is normal. no relays are turning on.
Also read about ghost switches, but i only have this unwanted trigger during a restart or a wifi network down. i change my wifi equipments 3 days ago, thats is when I  found the problem. tastmota 4ch was restarting in loop looking for a wifi network. every reboot triggerd the relay4 opening and closing the garage. i have 7 more tasmota units in my house, including a 4ch (not the pro version). only the 4ch pro R2 version witch inching hardware settings experience this behavior

Jason2866, 2018-12-02:
For normal operation with Tasmota the following settings are recommended:
S6: 1
K5: all 1
K6: all 0
(0 and 1 are printed onto the board next to the switch names)

netmaster01, 2018-12-02:
my settings are 
S6: 1
k5: ch1=0 ch2 =1 ch3=0 ch4=1
k6= 1110

This is controlling a motor. if i setup with mqtt timers with homeassistant. will depend on that If mqtt broker is down the motor will be working forever instead of xx seconds. with hardware setup its more reliable.

Jason2866, 2018-12-02:
Retain is off in HA? And mqtt contains no retained messages?

netmaster01, 2018-12-02:
yeah that topic was created just after the upgrade. is the only device that is using. no interaction with home assistant and mqtt messages. only managed via tasmota web interface. 

below decode status 

*** decode-status.py v20180730 by Theo Arends ***
Decoding information for device Sonoffgg1 from status report taken at 2018-12-02T18:12:38

Options
   0 (OFF) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT enabled
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) MQTT serial binary
  24 (OFF) Convert pressure to mmHg
  25 (OFF) KNX enabled
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led
  50 (ON ) Timers enabled
  51 (OFF) Generic ESP8285 GPIO enabled
  52 (OFF) Add UTC time offset to JSON message
  53 (OFF) Show hostname and IP address in GUI
  54 (OFF) Apply SetOption20 to Tuya
  55 (OFF) Use short Hass discovery messages
  56 (OFF) Use wifi network scan at restart
  57 (OFF) Use wifi network rescan regularly
  58 (OFF) Add IR raw data to JSON message
  59 (OFF) Change state topic from tele/STATE to stat/RESULT
  60 (OFF) Enable normal sleep instead of dynamic sleep

Features
  Language LCID = 2057
  MQTT_HOST_DISCOVERY
  MQTT_PUBSUBCLIENT
  USE_ADC_VCC
  USE_ARILUX_RF
  USE_ARMTRONIX_DIMMERS
  USE_BH1750
  USE_BMP
  USE_CSE7766
  USE_DHT
  USE_DISCOVERY
  USE_DISPLAY_LCD
  USE_DISPLAY_MATRIX
  USE_DISPLAY_MODES1TO5
  USE_DISPLAY_SSD1306
  USE_DOMOTICZ
  USE_DS18x20
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_HLW8012
  USE_HOME_ASSISTANT
  USE_HTU
  USE_HX711
  USE_I2C
  USE_IR_RECEIVE
  USE_IR_REMOTE
  USE_LM75AD
  USE_MCP39F501
  USE_MHZ19
  USE_NOVA_SDS
  USE_PMS5003
  USE_PZEM004T
  USE_PZEM_AC
  USE_PZEM_DC
  USE_RC_SWITCH
  USE_RULES
  USE_SENSEAIR
  USE_SERIAL_BRIDGE
  USE_SGP30
  USE_SHT
  USE_SHT3X
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_TUYA_DIMMER
  USE_TX20_WIND_SENSOR
  USE_WEBSERVER
  USE_WS2812
  WEBSERVER_ADVERTISE

ascillato, 2018-12-02:
use command weblog 4 to have more debug information and please re do the test to see the origin of the the toggle. You will see SRC: SWITCH or SRC: MQTT

If SRC: MQTT, your issue is that you have retained values in your broker

ascillato, 2018-12-02:
Ok. If you set the hardware inching mode, Tasmota will never now that. The hardware inching mode is controlled by another mcu. That is why Jason told you to use the recommend config of the wiki. Any other config will do this type of issues.

Sorry, you need to set the internal dip switches as Jason and the wiki states.

netmaster01, 2018-12-02:
ok. any suggestions to make the relays auto off after 10 seconds of operations ? or how to create a pulse of 2 seconds ?

i don't want to relay on homeassistant scripts or mqtt brokers since network some time may be unreliable

ascillato, 2018-12-02:
See pulsetime command in the wiki

ascillato2, 2018-12-02:
Have you managed to solve your issue?

netmaster01, 2018-12-02:
yeah. i change the dip switches  as the suggestion
For normal operation with Tasmota the following settings are recommended:
S6: 1
K5: all 1
K6: all 0
(0 and 1 are printed onto the board next to the switch names)

the configure the pulsetime 
22:51:09 RSL: Received Topic /pulsetime, Data Size 0, Data 
22:51:09 RSL: Group 0, Index 1, Command PULSETIME, Data 
22:51:09 MQT: stat/bugtest/RESULT = {"PulseTime1":"112 (Active 0)"}

that worked and solve the problem. no more relay trigger during boot proccess. Thanks

closed 2018-12-02 by netmaster01