arendst/Tasmota #4509: compile error soft_spi_flg

smittna, 2018-12-02:
Hello,
there is a little problem compiling the source code (6.3.0.15)

sonoff/xdrv_13_display.ino: In function 'boolean Xdrv13(byte)':
sonoff/xdrv_13_display.ino:1069:30: error: 'soft_spi_flg' was not declared
in this scope
if ((i2c_flg || spi_flg || soft_spi_flg) && XdspPresent()) {
^

missing in sonoff.ino something like that:
uint8_t soft_spi_flg = 0;

arendst, 2018-12-02:
You're right! I forgot to merge sonoff.ino which als contains a test for soft spi gpios. Pls forget soft spi for now as I have no access to the correct sonoff.ino at the moment.

andrethomas, 2018-12-02:
@smittna Thanks for pointing it out. I just added the variable to satisfy the compiler for now so please do not use Software SPI until @arendst has completed the remainder of the merge.

closed 2018-12-02 by smittna