arendst/Tasmota #4515: Sonoff Basic new Layout

reloxx13, 2018-12-03:
Heya,

the Sonoff Basics got a new Board Layout.

GPIO14 is gone and now there is a GPIO2 (IO2 and its not a borehole!).

In the Template GPIO2 is disabled, so dunno if u want to add a new Layout or allow GPIO2 for the Sonoff Basic.

Workaround: Use Generic Template and set GPIOs yourself.

Template:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/sonoff_template.h#L520

[image]

[image]

Jason2866, 2018-12-03:
Hello @reloxx13 
is it a real ITEAD Sonoff Basic? For me the typical ITEAD sticker on relais is missing.
Maybe just a (good?) clone...

andrethomas, 2018-12-03:
At least it seems they are addressing the thermal issue since this board appears to have 9 via's from the bottom of the ESP8266 chip coming through to the bottom onto what seems to be a much larger ground plane that in the older versions.

andrethomas, 2018-12-03:
@reloxx13 Is that a ESP8285 on there?

reloxx13, 2018-12-03:
@andrethomas yep, you will find better pictures here: https://smarthome.schimmer-media.de/forum/index.php?thread/2370-neuer-sonoff-basic-oder-ganz-altes-modell/&postID=26177#post26177

@Jason2866 im pretty sure those are orginal itead's
[image]

Some more infos here (in german):
https://forum.creationx.de/forum/index.php?thread/1254-sonoff-basic-neueste-version-gpio14/

https://smarthome.schimmer-media.de/forum/index.php?thread/2370-neuer-sonoff-basic-oder-ganz-altes-modell/

andrethomas, 2018-12-03:
@reloxx13 @Jason2866 

This is good news - so in future we will not have thermal issues with the new sonoff basic's I think.

andrethomas, 2018-12-03:
I think just adding GPIO2 to the Sonoff Basic template will not cause harm as GPIO2 is not used on the previous editions of the Sonoff Basic.

andrethomas, 2018-12-03:
Or it will be easier for user to select different version and template will dictate where POWER is.

ascillato, 2018-12-03:
I think that would be better and less confusing for people to just add gpio2 to the sonoff basic template

ascillato, 2018-12-03:
Lets make a PR and that Theo decides ;)

andrethomas, 2018-12-03:
Yeah, just adding GPIO2 is fine.

[image]

arendst, 2018-12-03:
Just add gpio2 please ;-)

ascillato, 2018-12-03:
@reloxx12 your call ;)

andrethomas, 2018-12-03:
https://github.com/arendst/Sonoff-Tasmota/pull/4518

ascillato2, 2018-12-03:
Closing as the PR has been made. Thanks everyone for sharing theirs ideas.

We also need to add this information to the wiki. I will do that later. Thanks.

closed 2018-12-03 by ascillato2

reloxx13, 2018-12-03:
i wrote sth in the wiki for that, feel free to edit/move it, im not sure where to place it on the basic page.

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic#new-board-layout

Version needs to be added if its released.

jimmyhall82, 2018-12-05:
How do I update the template used on my Tasmota sonoff? I flashed with most recent firmware but I don’t have gpio2 listed in the configuration template

ascillato, 2018-12-05:
Use latest Tasmota version 6.3.0.15 (http://thehackbox.org/tasmota/)

reloxx13, 2018-12-05:
(its not released now, its in the dev version)

jimmyhall82, 2018-12-05:
@ascillato thanks

joydeepsaha05, 2018-12-09:
@reloxx13 I have the same board and was trying to flash Tasmota, but it fails to get detected by my computer. Were you able to successfully flash it? Are the flashing steps different for the new board?

andrethomas, 2018-12-09:
@joydeepsaha05 

> Are the flashing steps different for the new board?

No, exactly the same.

joydeepsaha05, 2018-12-10:
@andrethomas Thanks for confirming. I was able to flash using a Raspberry Pi, must've been something wrong with my FTDI adapter.

hudecitydave, 2018-12-19:
THANKS for the work here!
Now I try to power up a SR505 PIR sensor on GPIO02/IO2. 
Has anyone found a 5V pin to get the needed power on the new board?
Thanks for info!

ascillato, 2018-12-19:
The sonoff basic R2 does not have enough power to power another device. Sorry. The hardware was not designed for that.

hudecitydave, 2018-12-23:
> 
> 
> The sonoff basic R2 does not have enough power to power another device. Sorry. The hardware was not designed for that.

Just a small update to that: I managed to solder a small cable to GPIO2/IO2, and ordered some HC-SR501 PIR Sensors. Bringing them to power with 3,3V works as charme. (like this: https://randomnerdtutorials.com/modifying-cheap-pir-motion-sensor-to-work-at-3-3v/)

Now we have the next room hassio enabeld :-)

andrethomas, 2018-12-23:
@hudecitydave Sure, it may work (for an undeterminable time), but it is not recommended.

hudecitydave, 2018-12-23:
> @hudecitydave Sure, it may work (for an undeterminable time), but it is not recommended.

You both are right. The device produces many false switch actions now. Not stable, not wife compatible.

But ... how to solve the "switch light on room movement" without an extra esp8266? Any hints for me?
Thanks, folks!

andrethomas, 2018-12-23:
> But ... how to solve the "switch light on room movement" without an extra esp8266? Any hints for me?

I think the obvious conclusion is an extra esp8266/wemos d1mini (or whatever you normally use for bare installations)

tlpbu, 2018-12-26:
@hudecitydave I am using [this AM312 PIR](https://www.aliexpress.com/item/Mini-IR-Pyroelectric-Infrared-PIR-Motion-Human-Sensor-Automatic-Detector-Module-high-reliability-12mm-x-25mm/32749737125.html?spm=a2g0s.9042311.0.0.27424c4dFWe9sq&dp=99791c0a21e48cf8853fef7b5a93156e&af=843361&cv=47843&afref=&mall_affr=pr3&dp=99791c0a21e48cf8853fef7b5a93156e&af=843361&cv=47843&afref=&mall_affr=pr3&aff_platform=aaf&cpt=1545862862805&sk=VnYZvQVf&aff_trace_key=10c9ef61636c4ec09ded881b548aa18e-1545862862805-03435-VnYZvQVf&terminal_id=033a0522a0184d888e9f8fdf12e0547f) with Sonoff Basic (older model without GPIO2) for 5 months or so.
Setup Tasmota rule to control PIR as you and your wife needs and you’re golden.

andrethomas, 2018-12-28:
@ivanmarban It should work the same as GPIO14 - I don't know why they changed the pin from GPIO14 to GPIO2 - perhaps it was easier to route on the new pc board design.

eric-void, 2018-12-28:
@ivanmarban @andrethomas Looking at tasmota documentation, it's not true that "it should work the same as GPIO14"
In https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#restrictions, you can read "GPIO2 can’t be low at boot, so you can’t connect a switch to it".
If the documentation tells the truth, using GPIO2 as a switch could give some problems (if the device reboots for any reason and the switch is off the device will not boot normally).

ivanmarban, 2018-12-28:
@eric-void @andrethomas any workaround ?

Jason2866, 2018-12-28:
Yes, there are restrictions with GPIO2. 
So it is not recommended to use switches, just only momentary switches.
If you want tu use a other GPIO you have to solder direct on the ESP chip.

freemiumdev, 2019-01-05:
how do you flash the new model? I can not get it into FLASH MODE
when i try to fash with Arduino ide i get this error:
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

andrethomas, 2019-01-05:
No difference to the old one.

Disconnect all power, hold down push button while powering up usb - keep button down for 2 seconds and then flash as per wki... no difference.

Some picky devices want you to hold the button down during the entire process - not sure if you maybe have one of these.

czarnikjak, 2019-01-06:
Any luck With connecting DHT22 to GPIO02 on these new boards?
Can’t get it to work, tasmota doesn’t see the sensor at all.

akasma74, 2019-01-07:
Yeah, tried to use DHT22 (on a board) with a new version of Sonoff Basic.
Shows null/null in web interface.
Ended up connecting the sensor to the old Sonoff Basic, and it works a treat.

Still want to have a solution for new boards as it's a bit odd and looks like a bug.

czarnikjak, 2019-01-07:
Mine is not even showing anything in the Web interface.
As if dht22 is not getting enough power.
I am supplying the 3.3v to it from the sonoff itself ( using vcc and ground
pins next to tx and rx)

freemiumdev, 2019-01-09:
> No difference to the old one.
> 
> Disconnect all power, hold down push button while powering up usb - keep button down for 2 seconds and then flash as per wki... no difference.
> 
> Some picky devices want you to hold the button down during the entire process - not sure if you maybe have one of these.

i've tried, but not working

cosmos1978, 2019-01-10:
So no more switches with the new layout.
Can someone tell me which pin is gpio14 ?

andrethomas, 2019-01-10:
[image]

Its a corner pin so if you're good you will achieve it :)

andrethomas, 2019-01-10:
[image]

andrethomas, 2019-01-10:
Actually, GPIO4 (D2) is also a good possible candidate...

[image]

andrethomas, 2019-01-10:
[image]

cosmos1978, 2019-01-10:
Wow many thanks. I know what to do tomorrow. 😀

andrethomas, 2019-01-10:
You're welcome - remember to use flux paste to avoid solder bridges between the pins and inspect your work with a magnifier after you have soldered, then tack the wire down with some hot glue a cm or so away from the chip so it does not come loose easily.

1300371, 2019-01-12:
> @eric-void @andrethomas any workaround ?
I use `GPIO3` (RX) and `GND` connected to a wall switch and it works great.
Moreover, it doesn't require additional soldering as i already solder it to flash.

czarnikjak, 2019-01-13:
Any possibility to use RX or TX pins to connect DHT sensor?

Jason2866, 2019-01-13:
Why not? connect it to RX

petersendk, 2019-01-13:
Hi forgot to write i tried 3,3v i tried 5,5v i tried adress 48, i tried adress 49 without luck. i also tried generic sonoff, but after configuring  and at first boot it stopped.

petersendk, 2019-01-14:
It seems that a scan for i2c sensors take about 5 minutes first time after pwr up, after this everything ok.

cosmos1978, 2019-01-14:
Does gpio 3 (rx) have a buildin pullup resistor ?
I have a sonnof with long cables now on gpio 2 and since there is no pullup lots of ghost switching.

andrethomas, 2019-01-15:
@cosmos1978 Yes

cbugeja, 2019-01-16:
Apologies but I'm still very very new to all this...
I need to set up a Sonoff Basic with 2 external sensors (reed switches) to work independently from the relay. The cable from the sensors to the sonoff needs to be 10mtrs max. 
I was thinking of using GPIO4 and GPIO14. Is this a good choice once the sensors are not momentary switches?
Would I need to attach any resistors to these to avoid what @cosmos1978 referred to as ghost switching?

georgia088, 2019-01-18:
> @ivanmarban
> 
> > @eric-void @andrethomas any workaround ?
> 
> I use `GPIO3` (RX) and `GND` connected to a wall switch and it works great.
> Moreover, it doesn't require additional soldering as i already solder it to flash.

I am new to all this, so I am trying to understand.  You are soldering wires from GPIO3 (RX) and GND to a standard Wall Switch (not momentary/rocker).  Then setting GPIO3 to (GPIO3 Serial In - Switch1 (09)) in Tasmota?

Is this correct? Does it work as it should?  

Thanks!!

BrunoZumba, 2019-01-18:
> PIO3 Serial In

Yes, my friend. That's perfect. I just did it this morning and works like a charm.
As I said in my reply just above this one: i just want someone to confirm that this setting has no other problems. But i can confirm that is working so far :)

BrunoZumba, 2019-01-18:
> I am new to all this, so I am trying to understand. You are soldering wires from GPIO3 (RX) and GND to a standard Wall Switch (not momentary/rocker). Then setting GPIO3 to (GPIO3 Serial In - Switch1 [...]

Yes, my friend. That's perfect. I just did it this morning and works like a charm.
As I said in my reply just above this one: i just want someone to confirm that this setting has no other problems. But i can confirm that is working so far :)

georgia088, 2019-01-18:
> Yes, my friend. That's perfect. I just did it this morning and works like a charm.
> As I said in my reply just above this one: i just want someone to confirm that this setting has no other problems. [...]

Awesome! Do you have to install a pull up resistor, or does it work with a built in one?
P.S. I posted this same basic question as a new issue because this thread shows "closed".  I wasn't sure if it would get viewed.  I apologize!

BrunoZumba, 2019-01-19:
> Awesome! Do you have to install a pull up resistor, or does it work with a built in one?
> P.S. I posted this same basic question as a new issue because this thread shows "closed". I wasn't sure if it [...]

No, I didn't install any pull up resistor. It hasn't exploded nor set my house on fire yet, so far so good :)
 
But a few posts above AndreThomas said it has a buildin pull resistor already.

2016for, 2019-01-25:
> Just a small update to that: I managed to solder a small cable to GPIO2/IO2, and ordered some HC-SR501 PIR Sensors. Bringing them to power with 3,3V works as charme. (like this: [...]

Installing PIR on GPIO2 isn't causing any problems like should be low on boot or something?
Someone suggested me to use GPIO1/TX for PIR, would that be ok/better/same as your GPIO2 connection?

[13 more replies]