arendst/Tasmota #4516: Is it me or is every version since 5.14 not as stable on Sonoff Basics

mcc05, 2018-12-03:
So I've tried all the 6 versions on around 10 sonoff basics, and they all show similar signs of instability.

Even on the latest 6.3.0, the wi-fi connection keeps dropping, the web interface is unresponsive, and the longer the device is active the worse it seems to get. On each when I go back to 5.14 everything is good again.

andrethomas, 2018-12-03:
Some sonoff basic's do not dissipate heat away from the processor as efficiently as it should and there was a lot of drivers and system functionality added since 5.14.

The result is that some sonoff basic's need sleep enabled in order to remain stable during the critical time when wifi is busy connecting.

Set your sleep to 50 by using command:
`sleep 50`

Upgrade to http://thehackbox.org/tasmota/sonoff.bin (6.3.0.15)

After upgrade is complete confirm that sleep is still set to 50.

ascillato, 2018-12-03:
Latest 6.3.0.15 works great for me. The previous had some issues in just one of my devices.

ascillato2, 2018-12-03:
Please, try latest 6.3.0.15 and let us know if works fine for you. Several users has reported good news about latest version. Hope would be the same for you.

Looking forward to your answer

closed 2018-12-03 by ascillato2

mcc05, 2018-12-03:
Ok just flashed 6.3.0.15, on the first sonoff basic, and had issues, but tried again this time erased the flash first. First impressions it does seem to be running a lot faster on the wifi and web interface. Is there a link for this version of the code, can't see a tagged version of 6.3.0.15 in the Tasmota git hub?

ascillato, 2018-12-03:
6.3.0.15 is the development version. Then after some more testing and fixes will become next release.

6.3.0 is latest master release.
6.3.0.15 is latest development (it has more than 400 commits than 6.3.0)

ascillato, 2018-12-03:
Which issues do you have?