arendst/Tasmota #4519: Sonoff outside access impossible.

Oliv59, 2018-12-03:
Hello there,

I've searched, but didn't find any information about this, and my issue is pretty simple, but impossible for me to fix it : i can't access my sonoff basic, flashed with Tasmota Firmware (Sonoff-Tasmota 6.2.1 by Theo Arends) a few days ago. I can access it form inside (same wifi network connection), but once my wifi is disconnected, and i'm connected through data (4G), no way to access it, always the ERR_CONNECTION_TIMED_OUT error. Tried through many browsers (Chrome, FF, IE), tried to forward different ports from my box to my sonoff (80, 82, 84), no way. Same ports redirected to other devices work fine from outside. I triple checked the destination IP forwarding, nothing wrong.

```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
00:15:03 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:15:03 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:04","StartupUTC":"","Sleep":0,"BootCount":74,"SaveCount":97,"SaveAddress":"FA000"}}
00:15:03 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:15:03 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXX","YYYYYY"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:15:03 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
00:15:03 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5338","IPAddress":"192.168.1.ZZZ","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:3A:E8:4F:94:DA","Webserver":2,"WifiConfig":5}}
00:15:03 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4F94DA","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:15:03 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:15:03 1970","Local":"Thu Jan 01 00:15:03 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:15:03 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:15:03"}}
00:15:03 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:15:03","Uptime":"0T00:14:04","Vcc":3.140,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXX","RSSI":78,"APMac":"00:E0:4C:71:59:77"}}}

```

Any lead, any further test appreciated ;)
Thanks,

Olivier.

ascillato, 2018-12-03:
Tasmota is not meant for being reached from outside.

You need to use your home automation software for that.

Jason2866, 2018-12-03:
If you  do port forwarding in your router to reach the device you have a security problem!
Not recommended and no support...

ascillato2, 2018-12-03:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-03 by ascillato2

Oliv59, 2018-12-03:
Sorry, isn't it the entire point of that sort of devices ? I don't have a home automation device at this place, but, even if i had a jeedom, for example, that wouldn't fix any security issues, right ? I don't see what are the differences between the one or the other ? Thanks for your help ;)

Jason2866, 2018-12-03:
If you setup a Home Automation correct, you dont expose Tasmota and other devices to the "outside" world. You have a secured way to your HA System which is in the "inside" world and is controlling the 
this "things"
So for secure use from apart a HA System is a must have...

Oliv59, 2018-12-06:
Hey again, thanks for your explanations. Can you give me one (or more, if you feel patient enough today ;) ) examples of securities differences between this simple Sonoff (tasmota based, or original firmware), and a HA system ? For example, i use Jeedom at home (but this one is not for me), and i can't find why my jeedom is more secure than this simple Sonoff basic ?
Thanks for your help ;)

Jason2866, 2018-12-06:
I didnt say that a setup with a HA system is general more secure.
BUT on the platform where the HA is running are enough resources from hardware to do this via
TLS VPN or or... use a firewall which only one route to one device! Security does need performance.
Esp82xx just hasnt enough ressources to do this AND running Tasmota or similar stuff

ascillato, 2018-12-06:
Remember that Tasmota has OTA capacity so, you can update the firmware from web. That means that if you port forward your Tasmota, a hacker can easily upload its own firmware to your sonoff and hack all your network broadcasting all your passwords and stuff. So, do not port forward a device that was not meant for that.

ascillato, 2018-12-06:
Please, search and read about securing your local network

ascillato, 2018-12-06:
A brief description of networking.

Your local network can't be reached using IP addresses from internet unless you port forward ports in your router. Doing that, you are opening paths for hackers to attack and steal information from you. Tasmota is not meant to be exposed to internet. 

Tasmota is meant to be part of your home automation system, so you need a mqtt broker for that. If your broker is local (like running in a raspberry pi) all your information is not being shared with anyone outside you own LAN.

The case of the stock software of the sonoff devices, use a server that is in china. So they connect to it and also your phone app connects to the same server. So, everything you do with your devices is being shared with iTead.

That is why, having an automation software that runs in your home only in a raspberry pi for example, will give you independence from all the manufacturers. So, you will own your own information. 

Then, this home automation softwares (like openhab, domoticz, home assistant, node red, etc etc) have some options for you to control them safely from Internet.

Hope this helps.

Oliv59, 2018-12-08:
Hello,

First, thanks to all of you for all those information. Indeed, the MQTT and firmware update possibilities didn't come to my mind, that can be a good explanation. Thus, as this devis was bought especially for this part (and i don't have, and don't plan to but a complete HA), is there a firmware, working on this sonoff basic Device, who would JUST allow me to do what i planned to do basically by a web interface, meaning just send an impulse to the out ? Or is it totally unsafe / unreal dream ?
Once more, thanks for your answers ;)

Jason2866, 2018-12-08:
Short answer, unsafe not recommended. Because device is limited in resources....
To be on the safe side communication from device has to be from Inside to Outside.
This can be reached via own HA system Inside or using a Cloud Service Outside (like eWelink)
If you communicate direct from Outside to Inside to your device you have to Open your Inside world -> Unsafe