arendst/Tasmota #4521: MPU-6050 no sensor reading

paspo, 2018-12-03:
**Describe the bug**
The values from the sensor aren't updated.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff SV + MPU-6050 (sda/scl on gpio 4&5)
- [x] _Development/Compiler/Upload tools used_ : atom + PlatformIO
- [x] _If a pre-compiled release or development binary was used, which one?_ :
- [x] _You have tried latest release or development binaries?_ : yes
- [x] _Provide the output of command_``status 0`` :
```
20:43:41 CMD: status 0
20:43:41 RSL: stat/sonoff6/STATUS = {"Status":{"Module":3,"FriendlyName":["Sonoff"],"Topic":"sonoff6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:43:41 RSL: stat/sonoff6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://REDACTED/release/sonoff.bin","RestartReason":"Power on","Uptime":"9T08:40:39","StartupUTC":"2018-11-24T11:03:02","Sleep":50,"BootCount":11,"SaveCount":66,"SaveAddress":"FA000"}}
20:43:41 RSL: stat/sonoff6/STATUS2 = {"StatusFWR":{"Version":"6.3.0.7(sonoff)","BuildDateTime":"2018-11-22T17:25:57","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:43:41 RSL: stat/sonoff6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"REDACTED","LogPort":514,"SSId":["iot",""],"TelePeriod":30,"SetOption":["00008009","5581C000","00000000"]}}
20:43:41 RSL: stat/sonoff6/STATUS4 = {"StatusMEM":{"ProgramSize":488,"Free":512,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FCAE794","24018001","23B617CE","00003BC8"]}}
20:43:41 RSL: stat/sonoff6/STATUS5 = {"StatusNET":{"Hostname":"sonoff6-7764","IPAddress":"REDACTED","Gateway":"REDACTED","Subnetmask":"255.255.255.0","DNSServer":"REDACTED","Mac":"REDACTED","Webserver":2,"WifiConfig":2}}
20:43:41 RSL: stat/sonoff6/STATUS6 = {"StatusMQT":{"MqttHost":"REDACTED","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_029E54","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:43:41 RSL: stat/sonoff6/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 03 19:43:41 2018","Local":"Mon Dec 03 20:43:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:24","Sunset":"16:55"}}
20:43:41 RSL: stat/sonoff6/STATUS10 = {"StatusSNS":{"Time":"2018-12-03T20:43:41","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}}
20:43:41 RSL: stat/sonoff6/STATUS11 = {"StatusSTS":{"Time":"2018-12-03T20:43:41","Uptime":"9T08:40:39","Vcc":2.980,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iot","BSSId":"02:9F:C2:F7:CD:A9","Channel":10,"RSSI":40}}}
```

**To Reproduce**
Connect the sensor via I2C and check the returned values in the webui/console

**Expected behavior**
I expected the values to change (it's a gyro+temperature sensor, so any movement should change the values)

**Screenshots**
An extract from the console:
```
20:44:18 RSL: tele/sonoff6/SENSOR = {"Time":"2018-12-03T20:44:18","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
20:45:02 RSL: tele/sonoff6/SENSOR = {"Time":"2018-12-03T20:45:02","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
20:45:53 RSL: tele/sonoff6/SENSOR = {"Time":"2018-12-03T20:45:53","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
20:46:44 RSL: tele/sonoff6/SENSOR = {"Time":"2018-12-03T20:46:44","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
20:47:35 RSL: tele/sonoff6/SENSOR = {"Time":"2018-12-03T20:47:35","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
```
On reboot I get a different read, but the Temperature is always wrong and most of the values are zero.

ascillato, 2018-12-03:
Hi,

This issue is happening to you after this fix (https://github.com/arendst/Sonoff-Tasmota/issues/4412) ?

ascillato, 2018-12-03:
Hi, your version is 6.3.0.7 but the fixes were applied in the 6.3.0.12

Please, update to latest 6.3.0.15

ascillato, 2018-12-03:
If after updating, you still have issues, we can ask to @VerboteneZone @Bunkerschild for help.

paspo, 2018-12-04:
my fault: I've reported the 'status 0' of the wrong sonoff...
In fact I was trying with 2 different releases, one of which was 6.3.0.12.

At the moment I have 2 sonoffs SV, both upgraded to 6.3.0.15, both with I2C on GPIO4&5, both with an MPU6050 connected.

On both devices, I2Cscan reports:
```
11:37:35 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x68"}
```
On both devices i get:
```
11:38:35 MQT: tele/sonoff6/SENSOR = {"Time":"2018-12-04T11:38:35","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
```
```
11:38:29 MQT: tele/sonoff/SENSOR = {"Time":"2018-12-04T11:38:29","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
```

paspo, 2018-12-04:
maybe this help:
on sonoff reboot, I get a different value for temperature (35.5) and then it goes back to 74.7:
```
tele/sonoff/SENSOR {"Time":"2018-12-04T11:52:25","MPU6050":{"Temperature":35.5,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
tele/sonoff6/SENSOR {"Time":"2018-12-04T11:52:27","MPU6050":{"Temperature":35.5,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
tele/sonoff/SENSOR {"Time":"2018-12-04T11:52:55","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
tele/sonoff6/SENSOR {"Time":"2018-12-04T11:52:57","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
tele/sonoff/SENSOR {"Time":"2018-12-04T11:53:25","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
tele/sonoff6/SENSOR {"Time":"2018-12-04T11:53:27","MPU6050":{"Temperature":74.7,"AccelXAxis":13312.00,"AccelYAxis":0.00,"AccelZAxis":0.00,"GyroXAxis":0.00,"GyroYAxis":0.00,"GyroZAxis":0.00},"TempUnit":"C"}
```

ascillato, 2018-12-04:
In your previous issue, your sensor was working fine right? What happened?

ascillato, 2018-12-04:
Ok, so we need the help of @VerboteneZone and @Bunkerschild.

paspo, 2018-12-04:
Maybe I found the problem: shitty chinese sensors.
Let me explain.
I connected one of the sensors to a raspberry-pi for testing, I wrote a small python script to retrieve the values and that was working. For some minutes. Ok, the values were somewhat random, but they're flowing... Until I started to shake the sensor. After some shaking, something wrong happened.
Disconnect, reconnect and the shaking is still knocking off the sensor.
I tried the other sensor, which is somewhat worse (it "requires" less shaking).
So I re-done my soldering, on both sensors, with the same results.
Then I changed the wires, with no luck.

I made a short video to explain: https://www.dropbox.com/s/pwnrooxhc8qf9wm/_mpu6050-fsck.mp4?dl=0

I'm shaking only the sensor, so I suppose the raspberry/sonoff/arduino side is ok.
Now I just bought a new sensor, from Amazon, that should arrive tomorrow (5,99€ vs 0,77€ :(  I really hope this works).
So, let's pause this and recheck when the new sensors arrives...

ascillato2, 2018-12-04:
Ok, That is good news and bad news. So, if it is hardware issue, the driver is OK. It is bad because you need to buy another.

Please, let us know if it works with the new sensors. Thanks.

paspo, 2018-12-05:
bad news...
the sensor arrived, and the results are the same...
This is a different sensors-board with different components (I hope better quality, due to 7x the cost)

ascillato2, 2018-12-05:
@VerboteneZone, @Bunkerschild 

Help please

Staars, 2018-12-06:
> bad news...
> the sensor arrived, and the results are the same...
> This is a different sensors-board with different components (I hope better quality, due to 7x the cost)

Did You try to upload a minimal working example (one of the various ARDUINO sketches on the internet) to your ESP8266 and did this work reliably? I just did this and while the simple sketch runs without a hitch as I just type this, my TASMOTA-builds with the MPU-6050 from today all crash.
If You can confirm this, I would try to dig a little deeper.

paspo, 2018-12-06:
With the help of a friend, I did a new set of tests, with both sensors (2 chinese + 1 amazon).
All my sensors are ok (they're reading correctly and the data makes sense).
As for the "shaking" problem: this doesn't happen with the arduino. I can't understand why.

In the evening, or maybe tomorrow, I'll try to flash the exact arduino sketch on both the sonoff-sv and a real arduino, and hope for the best.

Staars, 2018-12-06:
I will do some further tests. 
ATM it seems as it might be a software problem because I have rewritten some parts of the driver and get some sensor readings now.
But I am still not absolutely sure, that this is not some kind coincidence. More debugging is necessary.

paspo, 2018-12-07:
I'll update my experience.
I had ALL of my mpu-6050 connected to an arduino, with the example mpu-6050 sketch, and they're working out-of-the box.
Disconnected from the arduino, I connected Vcc,gnd,SDA (to gpio4), SCL (to gpio5) to a sonoff SV.
The only software difference is this: I replaced 
```
  Wire.begin();
```
with
```
  Wire.begin(4,5);
```
because I needed to specify the correct pins for I2C.

On the arduino I get:
```
=======================================================
temp : 24.22
accX : 0.01	accY : -0.04	accZ : 1.13
gyroX : -0.25	gyroY : 0.07	gyroZ : 0.08
accAngleX : -2.02	accAngleY : -0.60
gyroAngleX : -1.26	gyroAngleY : 1.10	gyroAngleZ : 1.08
angleX : -2.15	angleY : -0.71	angleZ : 1.08
=======================================================
```
on the sonoff I get:
```
=======================================================
temp : 36.50
accX : -0.00	accY : -0.00	accZ : -0.00
gyroX : -0.00	gyroY : -0.00	gyroZ : -0.00
accAngleX : -135.00	accAngleY : 135.00
gyroAngleX : -0.00	gyroAngleY : -0.00	gyroAngleZ : -2.27
angleX : -135.00	angleY : 135.00	angleZ : -2.27
=======================================================
```
Interesting fact: if I disconnect the sensor from the sonoff, when I power it on, I get the same values...
I'm not going in the right direction... any hint?

Staars, 2018-12-07:
Questions:
1. Did you compile yourself? What Tasmota-Version?
2. What does the web-page show?

ascillato, 2018-12-07:
@paspo

From your last post. You upload the arduino sample sketch to an arduino and also to the esp8266 and in the esp8266 didn't work right? 

Are you powering the sensor from another power source? Or using the 3.3v from your sonoff? There are some sensors that requiere 5v. May be you are using 5v from your arduino to your sensor and 3.3v from your sonoff to your sensor. 

Can you try power it up from your arduino (vcc and gnd) but the I2C pins and gnd from your sonoff? (the gnd from arduino needs to be connected also to your sonoff for this test)

Staars, 2018-12-08:
> I'll update my experience.
> I had ALL of my mpu-6050 connected to an arduino, with the example mpu-6050 sketch, and they're working out-of-the box.
> Disconnected from the arduino, I connected [...]

It is still a bit unclear, but it seems, that the sensor readings where always working under the hood, but the construction of the tele message is not working (in my case it even crashes).
The showing of the values on the web page are working on my setup (self compiled with 3.0.16).

Can you confirm this?

Staars, 2018-12-08:
Okay, so far so good:
`18:19:25 MQT: tele/sonoff/SENSOR = {"Time":"2018-12-08T18:19:25","MPU6050":{"Temperature":22.6,"AccelXAxis":4828.00,"AccelYAxis":-7124.00,"AccelZAxis":13420.00,"GyroXAxis":-130.00,"GyroYAxis":259.00,"GyroZAxis":75.00},"TempUnit":"C"}`

I am really scratching my head, why I had to change some things in the code to prevent a repeatable TASMOTA-crash and why it does not crash in the same way for you @paspo.
As I never used this sensor in TASMOTA before, I don't know if this is new or when it broke.

I will try to repeat and extend the tests and then maybe I will do a PR.

ascillato, 2018-12-08:
Please, share your changes. May be we can help to determine the crash you experienced.

Staars, 2018-12-08:
Okay, but please understand, that I changed some things in parallel and that it took me a while to find THE crash reason  ... and I am still not totally shure.

Basically I did the following:

#define MQTT_MAX_PACKET_SIZE   1200  (from 1000) in sonoff_post.h

snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s,\"%s\":{\"Temperature\":%s%s%s%s%s%s%s}"),
in the MPU-driver
before: snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s,\"%s\":{\"" D_JSON_TEMPERATURE "\":%s%s%s%s%s%s%s}"),

and reduced the sizes for the char[] above the last line from 40 to 30.

I am still in the middle of changing values for testing and have one or two additional ideas for the driver.

ascillato, 2018-12-08:
cool, thanks for sharing.

Just an observation:

the `#define MQTT_MAX_PACKET_SIZE 1000` shouldn't produce a crash. 1000 is high enough for the data. If it is unsufficient, the mqtt data will be cutted only.

Please, try with 1000 again. Thanks.

Staars, 2018-12-08:
Yes, you are right. It still works with 1000.
After counting the chars I also reduced the array sizes from 30 to 25 (22 should be enough).
For me the weird thing at the moment is D_JSON_TEMPERATURE vs Temperature.

ascillato, 2018-12-08:
D_JSON_TEMPERATURE is just a text

Staars, 2018-12-08:
Yes, I find this strange too and have no explanation how this breaks it for me.

ascillato, 2018-12-08:
When you are compiling by yourself, take into account to erase before flash, and to use core 2.3.0 or 2.4.2

paspo, 2018-12-10:
@ascillato 
This is my current setup. I have a small board with a micro-USB connector for powering (I power it from a 5V 2A power supply). From there, I get the 5V to power the MPU and the SonoffSV separately.

Same results with Tasmota.

[image]

Staars, 2018-12-10:
To my knowledge the operating voltage supply for the MPU-6050 should be 2,375 - 3,46 V. Is it really 5V compliant?
I use a 3.3V-pin of my NodeMCU, which itself is powered via USB.

paspo, 2018-12-10:
@Staars all the MPU-6050 boards I have here have got  small 3v3 voltage regulator onboard. From the spec sheet the input voltage range is 2 to 10V.

paspo, 2018-12-10:
I managed to have it working with a very simple arduino sketch without using any external libraries.
I'm just reading values via I2C and this is working (except I don't have calibration and so on, I'm reading values directly from sensor).

So I modified this function in xsns_32_mpu6050.ino:
```
void MPU_6050PerformReading(void)
{
  uint8_t buffer[14];

  I2Cdev::readBytes(0x68, 0x3B, 14, buffer);
  MPU_6050_ax = (((int16_t)buffer[0]) << 8) | buffer[1];
  MPU_6050_ay = (((int16_t)buffer[2]) << 8) | buffer[3];
  MPU_6050_az = (((int16_t)buffer[4]) << 8) | buffer[5];
  MPU_6050_temperature = (((int16_t)buffer[6]) << 8) | buffer[7];
  MPU_6050_gx = (((int16_t)buffer[8]) << 8) | buffer[9];
  MPU_6050_gy = (((int16_t)buffer[10]) << 8) | buffer[11];
  MPU_6050_gz = (((int16_t)buffer[12]) << 8) | buffer[13];

}
```
And while readings are correct now (yeah!), I found thins in the console:
```
MPU6050 found at 0x69
```
which is incorrect and maybe the cause of the errors...
In fact, if I use MPU_6050_address instead of 0x68, I get wrong data.

So:
- I think MPU_6050Detect() should be called only once, possibly at powerup, not on FUNC_PREP_BEFORE_TELEPERIOD. This is clearly a workaround: I have only one MPU6050 connected at 0x68 and I don't know why I get that message in the console.
- I think that it's reasonable to slim down the mpu6050 library just to the functions we actually need.

Even if this solves my problem (I have to detect a 90 degrees tilt, and get a temperature), I still believe we have to better integrate it into tasmota, maybe with 2 devices support.

Staars, 2018-12-10:
The issue with `MPU6050 found at 0x69` is fixed in my new version.
Maybe then you should try it.

Staars, 2018-12-10:
If you have questions regarding DMP, then feel free to ask.

Staars, 2018-12-10:
@paspo I have done the PR. I would be nice, if you can confirm that it works for you - or not.

ascillato2, 2018-12-10:
@paspo 

Please, try the PR https://github.com/arendst/Sonoff-Tasmota/pull/4581

Thanks

closed 2018-12-10 by ascillato2

elik745i, 2021-01-01:
Hi guys, Im working on weather station and experimenting with this module 6050, but cant compile bin file it exits with error code. Can you please have a look what is the problem?

WelterRocks, 2021-01-13:
Hi elik745i,

which kind of error code? Please send the error log.

elik745i, 2021-01-14:
> Hi elik745i,
> 
> which kind of error code? Please send the error log.

All good, sorted, using VisualStudio from now on, the problem was with ArduinoIDE.