arendst/Tasmota #4523: Sonoff Basic fail to start while connected to mains

ohadbenita, 2018-12-03:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff Basic
- [x ] _Development/Compiler/Upload tools used_ : NodeMCU Python Flasher
- [x] _If a pre-compiled release or development binary was used, which one?_ : sonoff.bin from the official tasmota github
- [x] _You have tried latest release or development binaries?_ : I've tried several versions including 5.9.1, 6.2.0 & 6.3.1
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

I connected the unit to my computer's serial output to check for possible clues, here's what I got :

00:00:04 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

**To Reproduce**
1. Burn with a tool of your choice
2. Connect to USB --> everything is working as expected
3. Connect to mains --> unit won't start

**Expected behavior**
Unit is expected to start

ascillato, 2018-12-03:
Seems that you have a hardware issue. Sorry

ascillato, 2018-12-03:
Please, search in issues, some users have found that their devices came with a faulty capacitor

ascillato, 2018-12-03:
If you need further assistance, please find us in the Tasmota Support chat. Thanks

closed 2018-12-03 by ascillato2