arendst/Tasmota #4525: Mulltiple triggers

AalianKhan, 2018-12-04:
Hello, I was playing with rules and thought it would be helpful to have multiple triggers. I have been using `events` and it is helpful but it will take up more space compared to just listing multiple triggers. Thanks in advance.

andrethomas, 2018-12-04:
Can you give an example of what you mean in contrast to what is currently available? I don't quite follow what you are requesting.

ascillato, 2018-12-04:
For simplicity of rules, it is not supported to chain several trigger conditions in the same ON.... ENDON rule. That is why, there is the posibility of using ON.... BREAK for doing that. Please, search in issues for this new type of rules. It is not in the wiki yet because was recently added. Thanks

ascillato2, 2018-12-04:
https://github.com/arendst/Sonoff-Tasmota/issues/4477