arendst/Tasmota #4526: DNS: Attempting connection...

mihalski, 2018-12-04:
I am concerned by the many `DNS: Attempting connection...` entries after startup and would like to know if there is something I need to fix to remove or at least reduce them. I have included the boot messages until `DNS: Initialized` below the template information.

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Firmware updated from 5.1.1 using the WebUI
- [x] Hardware used : Sonoff Basic
- [x] If a pre-compiled release or development binary was used, which one? : sonoff-classic
- [x] You have tried latest release or development binaries? : No
- [x] Provide the output of command ``status 0`` : Available below
```
17:57:28 MQT: stat/study_backlight/STATUS = {"Status":{"Module":1,"FriendlyName":["Study Backlight"],"Topic":"study_backlight","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:57:28 MQT: stat/study_backlight/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.198:5555/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:53","StartupUTC":"2018-12-04T07:51:35","Sleep":0,"BootCount":305,"SaveCount":26072,"SaveAddress":"F9000"}}
17:57:28 MQT: stat/study_backlight/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:35:18","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:57:28 MQT: stat/study_backlight/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["DobNet","BobNet"],"TelePeriod":30,"SetOption":["542A0009","55818000","00000001"]}}
17:57:28 MQT: stat/study_backlight/STATUS4 = {"StatusMEM":{"ProgramSize":483,"Free":520,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","6050A790","008003A8","210217C6","000001C0"]}}
17:57:28 MQT: stat/study_backlight/STATUS5 = {"StatusNET":{"Hostname":"study_backlight-2387","IPAddress":"192.168.1.18","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:07:E9:53","Webserver":2,"WifiConfig":4}}
17:57:28 MQT: stat/study_backlight/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07E953","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:57:28 MQT: stat/study_backlight/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 04 07:57:28 2018","Local":"Tue Dec 04 17:57:28 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10}}
17:57:28 MQT: stat/study_backlight/STATUS10 = {"StatusSNS":{"Time":"2018-12-04T17:57:28"}}
17:57:28 MQT: stat/study_backlight/STATUS11 = {"StatusSTS":{"Time":"2018-12-04T17:57:28","Uptime":"0T00:05:53","Vcc":3.198,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"DobNet","BSSId":"80:2A:A8:8A:21:B3","Channel":11,"RSSI":40}}}
```

Here are the messages from start up until success:
```
00:00:00 Project sonoff Study Backlight (Topic study_backlight, Fallback DVES_07E953, GroupTopic sonoffs) Version 6.3.0-2_3_0
00:00:00 WIF: Connecting to AP1 DobNet in mode 11N as study_backlight-2387...
00:00:11 WIF: Connected
00:00:11 DNS: Attempting connection...
00:00:11 HTP: Web server active on study_backlight-2387 with IP address 192.168.1.18
00:00:12 DNS: Attempting connection...
00:00:12 MQT: Attempting connection...
00:00:12 MQT: Connected
00:00:12 MQT: tele/study_backlight/LWT = Online (retained)
00:00:12 MQT: cmnd/study_backlight/POWER = 
00:00:12 MQT: tele/study_backlight/INFO1 = {"Module":"Sonoff Basic","Version":"6.3.0","FallbackTopic":"DVES_07E953","GroupTopic":"sonoffs"}
00:00:12 MQT: tele/study_backlight/INFO2 = {"WebServerMode":"Admin","Hostname":"study_backlight-2387","IPAddress":"192.168.1.18"}
00:00:12 MQT: tele/study_backlight/INFO3 = {"RestartReason":"Software/System restart"}
00:00:13 MQT: stat/study_backlight/RESULT = {"POWER1":"OFF"}
00:00:13 MQT: stat/study_backlight/POWER1 = OFF
00:00:13 MQT: homeassistant/switch/study_backlight_1/config =  (retained)
00:00:13 MQT: homeassistant/light/study_backlight_1/config = {"name":"Study Backlight","command_topic":"cmnd/study_backlight/POWER1","state_topic":"stat/study_backlight/RESULT","value_template":"{{value_json.POWER1}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/study_backlight/LWT","payload_available":"Online","payload_not_available":"Offline"} (retained)
00:00:13 MQT: homeassistant/switch/study_backlight_2/config =  (retained)
00:00:13 MQT: homeassistant/light/study_backlight_2/config =  (retained)
00:00:13 MQT: homeassistant/switch/study_backlight_3/config =  (retained)
00:00:13 MQT: homeassistant/light/study_backlight_3/config =  (retained)
00:00:13 MQT: homeassistant/switch/study_backlight_4/config =  (retained)
00:00:13 MQT: homeassistant/light/study_backlight_4/config =  (retained)
00:00:13 MQT: homeassistant/switch/study_backlight_5/config =  (retained)
00:00:13 MQT: homeassistant/light/study_backlight_5/config =  (retained)
00:00:13 MQT: homeassistant/switch/study_backlight_6/config =  (retained)
00:00:13 MQT: homeassistant/light/study_backlight_6/config =  (retained)
00:00:13 MQT: homeassistant/switch/study_backlight_7/config =  (retained)
00:00:13 MQT: homeassistant/light/study_backlight_7/config =  (retained)
00:00:13 MQT: homeassistant/switch/study_backlight_8/config =  (retained)
00:00:13 MQT: homeassistant/light/study_backlight_8/config =  (retained)
00:00:15 DNS: Attempting connection...
00:00:16 DNS: Attempting connection...
00:00:20 DNS: Attempting connection...
00:00:21 DNS: Attempting connection...
00:00:22 DNS: Attempting connection...
17:51:54 DNS: Attempting connection...
17:51:55 DNS: Attempting connection...
17:51:56 DNS: Attempting connection...
17:51:57 MQT: tele/study_backlight/STATE = {"Time":"2018-12-04T17:51:56","Uptime":"0T00:00:21","Vcc":3.196,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"DobNet","BSSId":"80:2A:A8:8A:21:B3","Channel":11,"RSSI":20}}
17:51:57 DNS: Attempting connection...
17:51:58 DNS: Attempting connection...
17:51:59 DNS: Attempting connection...
17:52:00 DNS: Attempting connection...
[... 75 lines cut ...]
17:53:20 DNS: Attempting connection...
17:53:21 DNS: Attempting connection...
17:53:22 DNS: Attempting connection...
17:53:23 DNS: Attempting connection...
17:53:24 DNS: Attempting connection...
17:53:25 DNS: Attempting connection...
17:53:26 DNS: Attempting connection...
17:53:27 DNS: Attempting connection...
17:53:28 DNS: Attempting connection...
17:53:29 DNS: Attempting connection...
17:53:30 DNS: Attempting connection...
17:53:31 DNS: Attempting connection...
17:53:32 DNS: Attempting connection...
17:53:33 MQT: tele/study_backlight/STATE = {"Time":"2018-12-04T17:53:33","Uptime":"0T00:01:58","Vcc":3.196,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"DobNet","BSSId":"80:2A:A8:8A:21:B3","Channel":11,"RSSI":20}}
17:53:33 DNS: Attempting connection...
17:53:34 DNS: Attempting connection...
17:53:35 DNS: Attempting connection...
17:53:36 DNS: Attempting connection...
17:53:37 DNS: Attempting connection...
17:53:38 DNS: Attempting connection...
17:53:39 DNS: Attempting connection...
17:53:40 DNS: Attempting connection...
17:53:41 DNS: Attempting connection...
17:53:43 DNS: Initialized
```

ascillato, 2018-12-04:
That depends on your router. If your router does not support mDNS or it has that disabled, this Tasmota feature will not work. Nothing to do from Tasmota side. If you are not using this feature, it is better that you disable it in my_user_config.h

mihalski, 2018-12-04:
Doesn't quite explain why other Tasmota devices do not appear to be having this issue:
```
00:00:00 Project sonoff Office (Topic office, Fallback DVES_0404B1, GroupTopic sonoffs) Version 6.3.0-2_3_0
00:00:00 WIF: Connecting to AP1 DobNet in mode 11N as office-1201...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on office-1201.local with IP address 192.168.1.204
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
```
I'm fairly confident mDNS is supported. For what it's worth avahi-daemon is running on the router.

mihalski, 2018-12-04:
Appears to have had no impact on this issue. I upgraded the Office device which continued to work as previously and I also updated the Study Backlight device which also continued to work as it had previously:
```
00:00:00 Project sonoff Study Backlight (Topic study_backlight, Fallback DVES_07E953, GroupTopic sonoffs) Version 6.3.0.15(fd8af87-classic)-2_3_0
00:00:00 WIF: Connecting to AP1 DobNet in mode 11N as study_backlight-2387...
00:00:04 WIF: Connected
00:00:04 DNS: Attempting connection...
00:00:04 HTP: Web server active on study_backlight-2387 with IP address 192.168.1.18
00:00:05 DNS: Attempting connection...
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: study_backlight/tele/LWT = Online (retained)
00:00:05 MQT: study_backlight/cmnd/POWER = 
00:00:05 MQT: study_backlight/tele/INFO1 = {"Module":"Sonoff Basic","Version":"6.3.0.15(fd8af87-classic)","FallbackTopic":"DVES_07E953","GroupTopic":"sonoffs"}
00:00:05 MQT: study_backlight/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"study_backlight-2387","IPAddress":"192.168.1.18"}
00:00:05 MQT: study_backlight/tele/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: study_backlight/stat/RESULT = {"POWER1":"OFF"}
00:00:05 MQT: study_backlight/stat/POWER1 = OFF
00:00:05 MQT: homeassistant/switch/study_backlight_1/config =  (retained)
00:00:05 MQT: homeassistant/light/study_backlight_1/config = {"name":"Study Backlight","command_topic":"study_backlight/cmnd/POWER1","state_topic":"study_backlight/tele/STATE","value_template":"{{value_json.POWER1}}","payload_off":"OFF","payload_on":"ON","availability_topic":"study_backlight/tele/LWT","payload_available":"Online","payload_not_available":"Offline"} (retained)
00:00:06 MQT: homeassistant/switch/study_backlight_2/config =  (retained)
00:00:06 MQT: homeassistant/light/study_backlight_2/config =  (retained)
00:00:06 MQT: homeassistant/switch/study_backlight_3/config =  (retained)
00:00:06 MQT: homeassistant/light/study_backlight_3/config =  (retained)
00:00:06 MQT: homeassistant/switch/study_backlight_4/config =  (retained)
00:00:06 MQT: homeassistant/light/study_backlight_4/config =  (retained)
00:00:06 MQT: homeassistant/switch/study_backlight_5/config =  (retained)
00:00:06 MQT: homeassistant/light/study_backlight_5/config =  (retained)
00:00:06 MQT: homeassistant/switch/study_backlight_6/config =  (retained)
00:00:06 MQT: homeassistant/light/study_backlight_6/config =  (retained)
00:00:06 MQT: homeassistant/switch/study_backlight_7/config =  (retained)
00:00:06 MQT: homeassistant/light/study_backlight_7/config =  (retained)
00:00:06 MQT: homeassistant/switch/study_backlight_8/config =  (retained)
00:00:06 MQT: homeassistant/light/study_backlight_8/config =  (retained)
00:00:06 DNS: Attempting connection...
22:29:25 DNS: Attempting connection...
22:29:26 DNS: Attempting connection...
22:29:27 DNS: Attempting connection...
22:29:28 DNS: Attempting connection...
22:29:29 DNS: Attempting connection...
22:29:30 DNS: Attempting connection...
22:29:31 DNS: Attempting connection...
22:29:32 MQT: study_backlight/tele/STATE = {"Time":"2018-12-04T22:29:32","Uptime":"0T00:00:14","Vcc":3.208,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"DobNet","BSSId":"80:2A:A8:8A:21:B3","Channel":11,"RSSI":68}}
22:29:32 DNS: Attempting connection...
22:29:33 DNS: Attempting connection...
22:29:34 DNS: Attempting connection...
22:29:35 DNS: Attempting connection...
22:29:36 DNS: Attempting connection...
22:29:37 DNS: Attempting connection...
22:29:38 DNS: Attempting connection...
22:29:39 DNS: Attempting connection...
22:29:40 DNS: Attempting connection...
22:29:41 DNS: Attempting connection...
22:29:42 DNS: Attempting connection...
22:29:43 DNS: Attempting connection...
22:29:44 DNS: Attempting connection...
22:29:45 DNS: Attempting connection...
22:29:46 DNS: Attempting connection...
22:29:47 DNS: Attempting connection...
22:29:48 DNS: Attempting connection...
22:29:49 DNS: Attempting connection...
22:29:50 DNS: Attempting connection...
22:29:51 DNS: Attempting connection...
22:29:52 DNS: Attempting connection...
22:29:53 DNS: Attempting connection...
22:29:54 DNS: Attempting connection...
22:29:55 DNS: Attempting connection...
22:29:56 DNS: Attempting connection...
22:29:57 DNS: Attempting connection...
22:29:58 DNS: Attempting connection...
22:29:59 DNS: Attempting connection...
[... 51 lines cut ...]
22:30:49 DNS: Attempting connection...
22:30:50 DNS: Attempting connection...
22:30:51 DNS: Attempting connection...
22:30:52 DNS: Attempting connection...
22:30:53 DNS: Attempting connection...
22:30:54 DNS: Attempting connection...
22:30:55 DNS: Attempting connection...
22:30:56 DNS: Attempting connection...
22:30:57 DNS: Attempting connection...
22:30:58 DNS: Attempting connection...
22:30:59 DNS: Attempting connection...
22:31:00 DNS: Attempting connection...
22:31:01 DNS: Attempting connection...
22:31:02 MQT: study_backlight/tele/STATE = {"Time":"2018-12-04T22:31:02","Uptime":"0T00:01:44","Vcc":3.208,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"DobNet","BSSId":"80:2A:A8:8A:21:B3","Channel":11,"RSSI":72}}
22:31:02 DNS: Attempting connection...
22:31:03 DNS: Attempting connection...
22:31:04 DNS: Attempting connection...
22:31:05 DNS: Attempting connection...
22:31:06 DNS: Attempting connection...
22:31:07 DNS: Attempting connection...
22:31:08 DNS: Attempting connection...
22:31:09 DNS: Attempting connection...
22:31:10 DNS: Attempting connection...
22:31:11 DNS: Initialized
```

ascillato2, 2018-12-04:
Have you tried also to reboot the router?

_Just for your information: you are using sonoff-classic version that it is not the full version. That is a legacy version. If you want to have all the features, please use sonoff.bin_

mihalski, 2018-12-04:
I was only using it because son off.bin failed to flash. I assume because of size? I'll try it again now, and I'll reboot the router.

ascillato, 2018-12-04:
Flashing by OTA? Yes, you need to use minimal first to free program space and then flash the full version.

By wire, you can flash the full version right away. 

Another thing we can try is to erase all flash before flashing again. You can use esptool.py as explained in the wiki

mihalski, 2018-12-04:
No change:
```
00:00:00 Project sonoff Study Backlight (Topic study_backlight, Fallback DVES_07E953, GroupTopic sonoffs) Version 6.3.0.15(fd8af87-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 DobNet in mode 11N as study_backlight-2387...
00:00:07 WIF: Connected
00:00:07 DNS: Attempting connection...
00:00:07 HTP: Web server active on study_backlight-2387 with IP address 192.168.1.18
00:00:08 DNS: Attempting connection...
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: study_backlight/tele/LWT = Online (retained)
00:00:08 MQT: study_backlight/cmnd/POWER = 
00:00:08 MQT: study_backlight/tele/INFO1 = {"Module":"Sonoff Basic","Version":"6.3.0.15(fd8af87-sonoff)","FallbackTopic":"DVES_07E953","GroupTopic":"sonoffs"}
00:00:08 MQT: study_backlight/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"study_backlight-2387","IPAddress":"192.168.1.18"}
00:00:08 MQT: study_backlight/tele/INFO3 = {"RestartReason":"Software/System restart"}
00:00:08 MQT: study_backlight/stat/RESULT = {"POWER1":"OFF"}
00:00:08 MQT: study_backlight/stat/POWER1 = OFF
00:00:08 MQT: homeassistant/switch/study_backlight_1/config =  (retained)
22:47:34 MQT: homeassistant/light/study_backlight_1/config = {"name":"Study Backlight","command_topic":"study_backlight/cmnd/POWER1","state_topic":"study_backlight/tele/STATE","value_template":"{{value_json.POWER1}}","payload_off":"OFF","payload_on":"ON","availability_topic":"study_backlight/tele/LWT","payload_available":"Online","payload_not_available":"Offline"} (retained)
22:47:34 MQT: homeassistant/switch/study_backlight_2/config =  (retained)
22:47:34 MQT: homeassistant/light/study_backlight_2/config =  (retained)
22:47:34 MQT: homeassistant/switch/study_backlight_3/config =  (retained)
22:47:34 MQT: homeassistant/light/study_backlight_3/config =  (retained)
22:47:34 MQT: homeassistant/switch/study_backlight_4/config =  (retained)
22:47:34 MQT: homeassistant/light/study_backlight_4/config =  (retained)
22:47:34 MQT: homeassistant/switch/study_backlight_5/config =  (retained)
22:47:34 MQT: homeassistant/light/study_backlight_5/config =  (retained)
22:47:34 MQT: homeassistant/switch/study_backlight_6/config =  (retained)
22:47:34 MQT: homeassistant/light/study_backlight_6/config =  (retained)
22:47:34 MQT: homeassistant/switch/study_backlight_7/config =  (retained)
22:47:34 MQT: homeassistant/light/study_backlight_7/config =  (retained)
22:47:34 MQT: homeassistant/switch/study_backlight_8/config =  (retained)
22:47:34 MQT: homeassistant/light/study_backlight_8/config =  (retained)
22:47:34 DNS: Attempting connection...
22:47:35 DNS: Attempting connection...
22:47:36 DNS: Attempting connection...
22:47:37 DNS: Attempting connection...
22:47:38 DNS: Attempting connection...
22:47:39 DNS: Attempting connection...
22:47:40 DNS: Attempting connection...
22:47:41 DNS: Attempting connection...
22:47:42 MQT: study_backlight/tele/STATE = {"Time":"2018-12-04T22:47:42","Uptime":"0T00:00:17","Vcc":3.235,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"DobNet","BSSId":"80:2A:A8:8A:21:B3","Channel":11,"RSSI":74}}
22:47:42 DNS: Attempting connection...
22:47:43 DNS: Attempting connection...
22:47:44 DNS: Attempting connection...
22:47:45 DNS: Attempting connection...
22:47:46 DNS: Attempting connection...
22:47:47 DNS: Attempting connection...
22:47:48 DNS: Attempting connection...
22:47:49 DNS: Attempting connection...
22:47:50 DNS: Attempting connection...
22:47:51 DNS: Attempting connection...
22:47:52 DNS: Attempting connection...
22:47:53 DNS: Attempting connection...
22:47:54 DNS: Attempting connection...
22:47:55 DNS: Attempting connection...
22:47:56 DNS: Attempting connection...
22:47:57 DNS: Attempting connection...
22:47:58 DNS: Attempting connection...
22:47:59 DNS: Attempting connection...
22:48:00 DNS: Attempting connection...
22:48:01 DNS: Attempting connection...
22:48:02 DNS: Attempting connection...
22:48:03 DNS: Attempting connection...
22:48:04 DNS: Attempting connection...
22:48:05 DNS: Attempting connection...
22:48:06 DNS: Attempting connection...
22:48:07 DNS: Attempting connection...
22:48:08 DNS: Attempting connection...
22:48:09 DNS: Attempting connection...
[... 52 lines cut ...]
22:49:00 DNS: Attempting connection...
22:49:01 DNS: Attempting connection...
22:49:02 DNS: Attempting connection...
22:49:03 DNS: Attempting connection...
22:49:04 DNS: Attempting connection...
22:49:05 DNS: Attempting connection...
22:49:06 DNS: Attempting connection...
22:49:07 DNS: Attempting connection...
22:49:08 DNS: Attempting connection...
22:49:09 DNS: Attempting connection...
22:49:10 DNS: Attempting connection...
22:49:11 DNS: Attempting connection...
22:49:12 MQT: study_backlight/tele/STATE = {"Time":"2018-12-04T22:49:12","Uptime":"0T00:01:47","Vcc":3.213,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"DobNet","BSSId":"80:2A:A8:8A:21:B3","Channel":11,"RSSI":76}}
22:49:12 DNS: Attempting connection...
22:49:13 DNS: Attempting connection...
22:49:14 DNS: Attempting connection...
22:49:15 DNS: Attempting connection...
22:49:16 DNS: Attempting connection...
22:49:17 DNS: Attempting connection...
22:49:18 DNS: Attempting connection...
22:49:19 DNS: Attempting connection...
22:49:20 DNS: Attempting connection...
22:49:21 DNS: Initialized
```
And for the short term my LAN is a harrowing nightmare of hostname resolution problems.

mihalski, 2018-12-04:
I'll give esptool.py a go tomorrow. I've used it before (along with every other imaginable flashing technique). I'm determined to figure out what's causing this.

ascillato, 2018-12-04:
So, as you said, seems that is your network setup that is overflowed. Try to set some devices to reserved IP in your router (like your home assistant, tv, etc) and use the IP instead. May be too much name resolution is overflowing your router.

You said that avahi is running in your router, so that means that your router is a raspberry pi? May be it is at high cpu usage and it is making this issue in your network.

ascillato, 2018-12-04:
Please, let us know your findings that will be very useful for other users.

Thanks for all the testings. Hope you will find the mDNS issue of your network.

andrethomas2, 2018-12-04:
This can also be a problem if your local DNS (on your router) is directly connected to internet - Many devices that run a local DNS service like that are subject to DDOS attacks - It can reach the ISP's dns but it will not be reachable from internet side.

mihalski, 2018-12-04:
But how is name resolution even a factor in this?
Currently my router is an ASUS RT-AC68U.. I jump between routers because I can't find anything that satisfies me. CPU usage on the router is less than 10% most of the time. Plus other Tasmota devices don't have this issue.. I'll update them ALL to the current dev firmware and see how it goes..

mihalski, 2018-12-04:
@andrethomas2 My DNS is not open. I'm using Cloudflare for my dns externally. Nothing so far has addressed why this issue is not affecting other Tasmota devices.. I guess I have to take a closer look at other Sonoff Basic devices in particular.

ascillato, 2018-12-04:
Try the erase before flash and flash by wire to avoid any incorrect data left in the flash at the SDK memory block (that is outside Tasmota)

mihalski, 2018-12-04:
Will do.. Project for tomorrow morning.. This has me both intrigued and frustrated. Hopefully it won't be too much longer before we "solve" the mystery.

andrethomas, 2018-12-04:
No idea to be honest as I cannot replicate what you are experiencing on any of my 63 devices with the current dev bin. So you'll need to dig a little with the information that has been given to you as hints  as to where to look.

I think that's about all that can be done at this stage but both @ascillato  and I have not been able to reproduce this problem so its not a firmware issue.

To proceed I agree with @ascillato 

1. Try reset 5 - This will erase the storage flash and revert to defaults whilst retaining your wifi credentials.
2. If the above does not do anything, do reset 3 - This will clear the SDK settings
3. If this does not solve it, follow the complete erase and reflash procedure as documented in the Wiki
4. If this does not work, then find the probem on your network.
5. If you cannot find any problem and other devices is not doing the same then replace the device.

ascillato, 2018-12-04:
If that stills produces this issue, try the precompiled bin with the 2.4.2 esp8266 core at http://thehackbox.org/tasmota/020402/

andrethomas, 2018-12-04:
Yeah, insert that last step of @ascillato between step 4 and 5 :)

ascillato2, 2018-12-04:
Please, let us know your findings that will be very useful for other users.

Thanks for all the testings. Hope you will find the mDNS issue of your network.

closed 2018-12-04 by ascillato2

mihalski, 2018-12-04:
It seems to me as if `reset 5` was the solution. After startup everything went perfectly and as I updated settings to match the previous configuration (as I believed it to be) it continued to start perfectly.

I guess `reset 5` is my new go to when I have problems I can't explain before coming and bothering the good folk here. Thank you!

andrethomas, 2019-02-01:
It erases the portion of flash not used by the firmware itself but retains the wifi configuration.

smartboart, 2019-07-08:
I have also this issue but only with generic 18 modules.. Like wemo d1 mini or 5 slot powerpluggs... I use different firmware.. Self-made and the pre compiled firmware.. 6.4.1.18 core 2.5.0 and self made 6.5.0.16 core 2.5.2. All other sonoff devices do not have this issue..

bobdig, 2020-12-01:
Thanks, had this problem too with all my DeLock Power Sockets.