arendst/Tasmota #4527: MQTT search via SD-DNS and link-local IP address when DHCP is not available

amontefusco, 2018-12-04:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
I want to stress a couple of points in support of mDNS search, and above all SD-DNS discovery, of the MQTT server:

1. isolated networks: i.e., networks operating on an ISOLATED broadcast domain: in such environment no services are available, just mDNS and SD-DNS, so the end nodes should be able to reach the MQTT server using Service Discovery; this is my case (avahi-browse -D -a -r output shown)

= wlan0 IPv4 Mosquitto MQTT server on cm3home _mqtt._tcp local
  hostname = [cm3home.local]
  address = [169.254.55.66]
  port = [1883]
  txt = ["info=Publish, Publish! Read all about it! mqtt.org"]

so, here two  features are in order:
a. discover the service _mqtt._tcp
b. assign a link-local IPv4 to the interface when DHCP  is not responding

2. even in case a DHCP service is available, discovering the MQTT broker via SD-NDS is very useful where you have more than one broker to implement redundancy.

**Describe the solution you'd like**
- Add a new configuration option that search for SD-DNS service name, not via the mDNS hostname, and doesn't cache the IP address so achieved.
The search has to be done each time the connection  with the broker is lost and/or at startup
- Add a new configuration option that , when DHCP is not answering

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
isolated networks and multiple redundant brokers availability.

amontefusco, 2018-12-04:
it seems that the code for SD-DNS MQTT broker search is already there:

xdrv_02_mqtt.ino @amontefusco 
line 214

`#ifdef USE_DISCOVERY
#ifdef MQTT_HOST_DISCOVERY
boolean MqttDiscoverServer(void)
{
  if (!mdns_begun) { return false; }

  int n = MDNS.queryService("mqtt", "tcp");  // Search for mqtt service
`
just the call  at line 505 should be slightly changed in order to force the rediscovery each time.

ascillato, 2018-12-04:
Hi

Rediscovery is implemented. Let's change your feature request to troubleshooting issue. 

Which sonoff devices are you using?
How is your setup?
Which Tasmota version are you using? Latest 6.3.0.15?
What do you see in the console output? 
Are you using weblog to level 4? (command weblog 4 in console to have more debug information) 
Are you using the precompiled bins from development branch? 
Have you compiled yourself?

amontefusco, 2018-12-04:
> Hi
> 
> Rediscovery is implemented.
> 
> Let's change your feature request to troubleshooting issue.
> 
>     * Which sonoff devices are you using?
tele/sonoff-0657/LWT Online
cmnd/sonoff-0657/POWER (null)
tele/sonoff-0657/INFO1 {"Module":"**Sonoff Touch**","Version":"**6.3.0.15**(sonoff)","FallbackTopic":"DVES_97E291","GroupTopic":"sonoffs"}

>     * How is your setup?
> 
>     * Which Tasmota version are you using? Latest 6.3.0.15?
Version":"**6.3.0.15**

>     * What do you see in the console output?

`
00:00:00 CFG: Loaded from flash at F8, Count 204
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff-0657, Fallback DVES_97E291, GroupTopic sonoffs) Version 6.3.0.15(sonoff)-2_3_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 MRKDEV in mode 11N as sonoff-0657-0657...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
00:00:10 APP: Boot Count 21
00:00:10 CFG: Saved to flash at F7, Count 205, Bytes 3584
00:00:10 WIF: Checking connection...
00:00:10 WIF: Attempting connection...
00:00:11 WIF: Checking connection...
00:00:11 WIF: Connect failed with AP timeout
00:00:11 WIF: Patch issue 2186
00:00:11 WIF: Connecting to AP1 MRKDEV in mode 11N as sonoff-0657-0657...
00:00:12 WIF: Checking connection...
00:00:12 WIF: Attempting connection...
00:00:13 WIF: Checking connection...
00:00:13 WIF: Attempting connection...
00:00:14 WIF: Checking connection...
00:00:14 WIF: Attempting connection...
00:00:15 WIF: Checking connection...
00:00:15 WIF: Attempting connection...
00:00:16 WIF: Checking connection...
00:00:16 WIF: Attempting connection...
00:00:17 WIF: Checking connection...
00:00:17 WIF: Connected
00:00:17 DNS: Initialized
00:00:17 HTP: Web server active on sonoff-0657-0657.local with IP address 192.168.0.18
00:00:18 MQT: Attempting connection...
00:00:19 DNS: Query done. MQTT services found 1
00:00:19 DNS: MQTT service found on cm3home, IP Address 192.168.0.15, Port 1883
00:00:19 MQT: Connected
00:00:19 APP: (UTC) Tue Dec 04 21:16:21 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
22:16:22 MQT: tele/sonoff-0657/LWT = Online (retained)
22:16:23 MQT: cmnd/sonoff-0657/POWER = 
22:16:23 MQT: Subscribe to cmnd/sonoff-0657/#
22:16:23 MQT: Subscribe to cmnd/sonoffs/#
22:16:23 MQT: Subscribe to cmnd/DVES_97E291/#
22:16:23 MQT: tele/sonoff-0657/INFO1 = {"Module":"Sonoff Touch","Version":"6.3.0.15(sonoff)","FallbackTopic":"DVES_97E291","GroupTopic":"sonoffs"}
22:16:23 MQT: tele/sonoff-0657/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0657-0657","IPAddress":"192.168.0.18"}
22:16:23 MQT: tele/sonoff-0657/INFO3 = {"RestartReason":"Software/System restart"}
22:16:23 MQT: stat/sonoff-0657/RESULT = {"POWER":"ON"}
22:16:23 MQT: stat/sonoff-0657/POWER = ON
22:16:23 CFG: Saved to flash at F6, Count 206, Bytes 3584
22:16:27 HTP: Main Menu
22:16:31 MQT: tele/sonoff-0657/STATE = {"Time":"2018-12-04T22:16:31","Uptime":"0T00:00:28","Vcc":3.204,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"MRKDEV","BSSId":"06:8D:DB:66:3B:20","Channel":11,"RSSI":82}}
22:16:31 HTP: Console
22:16:42 WIF: Checking connection...
22:16:42 WIF: Connected
22:16:47 HTP: Main Menu
22:16:48 HTP: Console

`

 
>     * Are you using weblog to level 4? (command `weblog 4` in console to have more debug information)
>yes

>     * Are you using the precompiled bins from development branch?
No 
>     * Have you compiled by yourself?
yes, on Linux with only the following changeset

'
andrew@andrewp:/tmp/Sonoff-Tasmota$ git diff
diff --git a/platformio.ini b/platformio.ini
index bb68e98..98f95cb 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -84,7 +84,7 @@ build_flags =
 monitor_speed = 115200
 
 ; *** Upload Serial reset method for Wemos and NodeMCU
-upload_speed = 115200
+upload_speed = 74880
 upload_resetmethod = nodemcu
 upload_port = COM5

'

amontefusco, 2018-12-04:
> Thanks for answering. But in the log output is not shown your reconnecting issue. Please, show the output also when your issue happens. Please also add the STATUS 0 output.

00:00:10 APP: Boot Count 22
00:00:10 CFG: Saved to flash at FA, Count 210, Bytes 3584
00:00:10 WIF: Checking connection...
00:00:10 WIF: Attempting connection...
00:00:11 WIF: Checking connection...
00:00:11 WIF: Connect failed with AP timeout
00:00:11 WIF: Patch issue 2186
00:00:11 WIF: Connecting to AP1 MRKDEV in mode 11N as sonoff-0657-0657...
00:00:12 WIF: Checking connection...
00:00:12 WIF: Attempting connection...
00:00:13 WIF: Checking connection...
00:00:13 WIF: Attempting connection...
00:00:14 WIF: Checking connection...
00:00:14 WIF: Attempting connection...
00:00:15 WIF: Checking connection...
00:00:15 WIF: Attempting connection...
00:00:16 WIF: Checking connection...
00:00:16 WIF: Attempting connection...
00:00:17 WIF: Checking connection...
00:00:17 WIF: Connected
00:00:17 DNS: Initialized
00:00:17 HTP: Web server active on sonoff-0657-0657.local with IP address 192.168.0.19
00:00:18 MQT: Attempting connection...
00:00:19 APP: (UTC) Tue Dec 04 21:27:25 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
22:27:30 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:27:38 HTP: Main Menu
22:27:40 SRC: WebGui from 192.168.0.56
22:27:40 RSL: stat/sonoff-0657/RESULT = {"POWER":"OFF"}
22:27:40 RSL: stat/sonoff-0657/POWER = OFF
22:27:41 SRC: WebGui from 192.168.0.56
22:27:41 RSL: stat/sonoff-0657/RESULT = {"POWER":"ON"}
22:27:41 RSL: stat/sonoff-0657/POWER = ON
22:27:41 MQT: Attempting connection...
22:27:47 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:27:47 HTP: Configuration
22:27:49 HTP: Configure MQTT
22:27:56 WIF: Checking connection...
22:27:56 WIF: Connected
22:27:59 MQT: Attempting connection...
22:28:05 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:28:16 MQT: Attempting connection...
22:28:22 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:28:28 WIF: Checking connection...
22:28:28 WIF: Connected
22:28:33 MQT: Attempting connection...
22:28:39 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:28:50 MQT: Attempting connection...
22:28:56 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:29:00 WIF: Checking connection...
22:29:00 WIF: Connected
22:29:07 MQT: Attempting connection...
22:29:13 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:29:24 MQT: Attempting connection...
22:29:30 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:29:32 WIF: Checking connection...
22:29:32 WIF: Connected
22:29:41 MQT: Attempting connection...
22:29:47 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:29:58 WIF: Checking connection...
22:29:58 WIF: Connected
22:29:58 MQT: Attempting connection...
22:30:04 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:30:15 MQT: Attempting connection...
22:30:21 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:30:30 WIF: Checking connection...
22:30:30 WIF: Connected
22:30:32 MQT: Attempting connection...
22:30:38 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:30:49 MQT: Attempting connection...
22:30:55 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:31:02 WIF: Checking connection...
22:31:02 WIF: Connected
22:31:06 MQT: Attempting connection...
22:31:12 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:31:23 MQT: Attempting connection...
22:31:29 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:31:34 WIF: Checking connection...
22:31:34 WIF: Connected
22:31:40 MQT: Attempting connection...
22:31:46 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:31:57 MQT: Attempting connection...
22:32:03 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:32:03 HTP: Configuration
22:32:06 WIF: Checking connection...
22:32:06 WIF: Connected
22:32:09 HTP: Main Menu
22:32:11 HTP: Console
22:32:14 MQT: Attempting connection...
22:32:20 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:32:31 MQT: Attempting connection...
22:32:37 MQT: Connect failed to 192.168.0.15:1883, rc -2. Retry in 10 sec
22:32:38 WIF: Checking connection...
22:32:38 WIF: Connected

ascillato, 2018-12-04:
Thanks.

please type in the console `MqttHost 0`

amontefusco, 2018-12-04:
> Thanks.
> 
> please type in the console `MqttHost 0`

22:40:05 CMD: MqttHost 0
22:40:05 SRC: WebConsole from 192.168.0.56
22:40:05 RSL: Received Topic /MqttHost, Data Size 1, Data 0
22:40:05 RSL: Group 0, Index 1, Command MQTTHOST, Data 0
22:40:05 MQT: stat/sonoff-0657/RESULT = {"MqttHost":""}

ascillato, 2018-12-04:
The actual behaviour is that Tasmota will look for a MQTT Broker only IF you don't have set any MQTT Broker before.

The command MQTTHOST 0 will delete your Broker Information and Tasmota will look for another MQTT Server.

[image]

If you want is this to be ALL the time, that is another behaviour different than the default. That is also supported by Tasmota using the rules feature. Rules are there to "modify/adapt" the actual behaviours of Tasmota.

Please, go to the console and type:

```
rule1 1
rule1 on Mqtt#Disconnected do MqttHost 0 endon
```

ascillato2, 2018-12-04:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-04 by ascillato2

amontefusco, 2018-12-04:
The rule1 you suggested nailed it !
Now when the cluster takeover happens and therefore the MQTT servicce is not anymore reachable, the Tasmota Sonoff Touch is detecting the unreachability, reset itself and finally rediscover the new MQTT server via SD-DNS !

For the sake of the completeness:

00:00:00 CFG: Loaded from flash at F4, Count 232
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff-0657, Fallback DVES_97E291, GroupTopic sonoffs) Version 6.3.0.15(sonoff)-2_3_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 MRKDEV in mode 11N as sonoff-0657-0657...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
00:00:10 APP: Boot Count 28
00:00:10 CFG: Saved to flash at FB, Count 233, Bytes 3584
00:00:10 WIF: Checking connection...
00:00:10 WIF: Attempting connection...
00:00:11 WIF: Checking connection...
00:00:11 WIF: Connect failed with AP timeout
00:00:11 WIF: Patch issue 2186
00:00:11 WIF: Connecting to AP1 MRKDEV in mode 11N as sonoff-0657-0657...
00:00:12 WIF: Checking connection...
00:00:12 WIF: Attempting connection...
00:00:13 WIF: Checking connection...
00:00:13 WIF: Attempting connection...
00:00:14 WIF: Checking connection...
00:00:14 WIF: Attempting connection...
00:00:15 WIF: Checking connection...
00:00:15 WIF: Attempting connection...
00:00:16 WIF: Checking connection...
00:00:16 WIF: Attempting connection...
00:00:17 WIF: Checking connection...
00:00:17 WIF: Attempting connection...
00:00:18 WIF: Checking connection...
00:00:18 WIF: Attempting connection...
00:00:19 WIF: Checking connection...
00:00:19 WIF: Attempting connection...
00:00:20 WIF: Checking connection...
00:00:20 WIF: Connected
00:00:20 DNS: Initialized
00:00:20 HTP: Web server active on sonoff-0657-0657.local with IP address 192.168.0.26
00:00:21 MQT: Attempting connection...
00:00:22 DNS: Query done. MQTT services found 1
00:00:22 DNS: MQTT service found on cm3home, IP Address 192.168.0.55, Port 1883
00:00:22 MQT: Connected
00:00:23 MQT: tele/sonoff-0657/LWT = Online (retained)
00:00:23 MQT: cmnd/sonoff-0657/POWER = 
00:00:23 MQT: Subscribe to cmnd/sonoff-0657/#
00:00:23 MQT: Subscribe to cmnd/sonoffs/#
00:00:23 MQT: Subscribe to cmnd/DVES_97E291/#
00:00:23 MQT: tele/sonoff-0657/INFO1 = {"Module":"Sonoff Touch","Version":"6.3.0.15(sonoff)","FallbackTopic":"DVES_97E291","GroupTopic":"sonoffs"}
00:00:23 MQT: tele/sonoff-0657/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0657-0657","IPAddress":"192.168.0.26"}
00:00:23 MQT: tele/sonoff-0657/INFO3 = {"RestartReason":"Software/System restart"}
00:00:23 MQT: stat/sonoff-0657/RESULT = {"POWER":"ON"}
00:00:23 MQT: stat/sonoff-0657/POWER = ON
00:00:23 CFG: Saved to flash at FA, Count 234, Bytes 3584
00:00:23 APP: (UTC) Tue Dec 04 21:56:20 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
22:56:28 MQT: tele/sonoff-0657/STATE = {"Time":"2018-12-04T22:56:28","Uptime":"0T00:00:30","Vcc":3.221,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"MRKDEV","BSSId":"06:8D:DB:66:3B:20","Channel":11,"RSSI":82}}
22:56:36 HTP: Main Menu
22:56:38 HTP: Console
22:56:39 WIF: Checking connection...
22:56:39 WIF: Connected
22:56:59 WIF: Checking connection...
22:56:59 WIF: Connected

Many thanks for your support.

amontefusco, 2018-12-04:
I updated the example section of Rules page.

https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#18-rule-to-force-automatic-reconnection-to-mqtt-server-via-sd-dns

ascillato, 2018-12-05:
Thanks :+1: very nice from you!

MaxBec, 2019-04-17:
Unfortunately this isn't working with me... I install ahavi on my raspberry but with mqtthost 0 it doesn't find the server...

amontefusco, 2019-04-17:
> Unfortunately this isn't working with me... I install ahavi on my raspberry but with mqtthost 0 it doesn't find the server...

Did you followed the wiki ?
Is the service mqtt visible on sd-dns ? (avahi-browse -D -a)

dachshund-digital, 2021-05-15:
So the default behavior is NOT standard to what every other solution uses?  That FQDN should always be resolvable?  When proper DNS servers are present and authoritative?  Why make users jump through hops?