arendst/Tasmota #4528: Flashing sonoff touch does nothing

kisdaniel, 2018-12-04:
First I have saved the original image:

```
./esptool.py read_flash 0 0x100000 sonoff_touch_original.bin
esptool.py v2.6-beta1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
MAC: dc:4f:22:c6:da:00
Uploading stub...
Running stub...
Stub running...
1048576 (100 %)
1048576 (100 %)
Read 1048576 bytes at 0x0 in 94.9 seconds (88.4 kbit/s)...
Hard resetting via RTS pin...
```

Than I have erased the flash:

```
./esptool.py erase_flash 
esptool.py v2.6-beta1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
MAC: dc:4f:22:c6:da:00
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Hard resetting via RTS pin...

```

Finally I tried to upload tasmota firmware from platformIO:

```
Building .pioenvs/sonoff/firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=======   ]  66.1% (used 54160 bytes from 81920 bytes)
PROGRAM: [=====     ]  47.6% (used 498883 bytes from 1048576 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/sonoff/firmware.bin
Uploading 503024 bytes from .pioenvs/sonoff/firmware.bin to flash at 0x00000000
................................................................................ [ 16% ]
................................................................................ [ 32% ]
................................................................................ [ 48% ]
................................................................................ [ 65% ]
................................................................................ [ 81% ]
................................................................................ [ 97% ]
............                                                                     [ 100% ]
================================================================================================================ [SUCCESS] Took 57.92 seconds ================================================================================================================

========================================================================================================================== [SUMMARY] ==========================================================================================================================
Environment sonoff              [SUCCESS]
Environment sonoff-minimal      [SKIP]
Environment sonoff-basic        [SKIP]
Environment sonoff-classic      [SKIP]
Environment sonoff-knx          [SKIP]
Environment sonoff-sensors      [SKIP]
Environment sonoff-display      [SKIP]
Environment sonoff-BG           [SKIP]
Environment sonoff-BR           [SKIP]
Environment sonoff-CN           [SKIP]
Environment sonoff-CZ           [SKIP]
Environment sonoff-DE           [SKIP]
Environment sonoff-ES           [SKIP]
Environment sonoff-FR           [SKIP]
Environment sonoff-GR           [SKIP]
Environment sonoff-HE           [SKIP]
Environment sonoff-HU           [SKIP]
Environment sonoff-IT           [SKIP]
Environment sonoff-NL           [SKIP]
Environment sonoff-PL           [SKIP]
Environment sonoff-PT           [SKIP]
Environment sonoff-RU           [SKIP]
Environment sonoff-TR           [SKIP]
Environment sonoff-TW           [SKIP]
Environment sonoff-UK           [SKIP]
================================================================================================================ [SUCCESS] Took 57.92 seconds ================================================================================================================

```

my problem is that nothing happened, it seems to me that original firmware has not beed overridden:
When I read back the content of the flash

./esptool.py read_flash 0 0x100000 after_flashing.bin

than I checked the md5 sum of the file, it is exactly same as  sonoff_touch_original.bin.

How can I override the flash?

curzon01, 2018-12-04:
First try a ready compiled bin and manual entered flahs command:

```
wget http://thehackbox.org/tasmota/release/sonoff.bin
esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
```

If it's working you can try flash your own compiled bin using same command
`esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 .pioenvs/sonoff/firmware.bin`

Jason2866, 2018-12-04:
Flash erase did fail. Erase flash was done in 0 sec. This is a indicator for defect spi flash chip
You hav a issue with uploading firmware.... It looks good but it doesnt work
I had the same issue already, after changing SPI Flash it worked

kisdaniel, 2018-12-04:
I have solved this: the error was that I had not disconnected GPIO0 from GND after start. 

The erase and the write was successfull.

This video not mention that I have to disconnect GPIO0 from GND after 2-3 sec after boot.
https://www.youtube.com/watch?v=LPUq-SOzI-o

closed 2018-12-04 by kisdaniel