arendst/Tasmota #4529: DNS: Attempting connection

crispy78, 2018-12-04:
One device that is running 6.3.0.15 has issues with its DNS connection. After numerous restarts and ipaddress1 till 4 configurations (fixed ip 192.168.178.50, gateway 192.168.178.1, subnet 255.255.255 and dns 8.8.8.8) I’ve updated the device with the http://thehackbox.org/tasmota/sonoff-minimal.bin update, then the regular version http://thehackbox.org/tasmota/sonoff.bin but still some errors connecting with DNS. All my other devices (+/- 25) are running 6.3.0.15 without any issue. All have similar settings. What could be the issue?

A snippet of the log:
18:51:51 DNS: Attempting connection...
18:51:52 DNS: Attempting connection...
18:51:53 DNS: Attempting connection...
18:51:54 DNS: Attempting connection...
18:51:55 DNS: Attempting connection...
18:51:56 DNS: Attempting connection...
18:51:57 DNS: Attempting connection...
18:51:58 DNS: Attempting connection...
18:51:59 DNS: Attempting connection...
18:52:00 DNS: Attempting connection...
18:52:01 DNS: Attempting connection...
18:52:02 DNS: Attempting connection...
18:52:03 DNS: Attempting connection...
18:52:04 DNS: Attempting connection...
18:52:05 DNS: Attempting connection...
18:52:06 DNS: Attempting connection...
18:52:07 DNS: Attempting connection...
18:52:08 DNS: Attempting connection...
18:52:09 DNS: Attempting connection...
18:52:10 DNS: Attempting connection...
18:52:11 DNS: Attempting connection...
18:52:12 DNS: Attempting connection...
18:52:13 DNS: Attempting connection...
18:52:14 DNS: Attempting connection...
18:52:15 DNS: Attempting connection...
18:52:16 DNS: Attempting connection...
18:52:17 DNS: Attempting connection...
18:52:18 DNS: Attempting connection...
18:52:19 DNS: Attempting connection...
18:52:20 DNS: Attempting connection...
18:52:21 DNS: Attempting connection...
18:52:22 DNS: Attempting connection...
18:52:23 DNS: Attempting connection...
18:52:24 DNS: Attempting connection...
18:52:25 DNS: Attempting connection...
18:52:26 DNS: Attempting connection...
18:52:27 DNS: Attempting connection...
18:52:28 DNS: Attempting connection...
18:52:29 DNS: Attempting connection...
18:52:30 DNS: Attempting connection...
18:52:31 DNS: Attempting connection...
18:52:32 DNS: Attempting connection...
18:52:33 DNS: Attempting connection...
18:52:34 DNS: Attempting connection...
18:52:35 DNS: Attempting connection...
18:52:36 DNS: Attempting connection...
18:52:37 DNS: Attempting connection...
18:52:38 DNS: Attempting connection...
18:52:39 DNS: Attempting connection...
18:52:40 DNS: Attempting connection...
18:52:41 DNS: Attempting connection...
18:52:42 DNS: Attempting connection...
18:52:43 DNS: Attempting connection...
18:52:44 DNS: Attempting connection...
18:52:45 DNS: Attempting connection...
18:52:46 DNS: Attempting connection...
18:52:47 DNS: Attempting connection...
18:52:48 DNS: Attempting connection...
18:52:49 DNS: Attempting connection...
18:52:50 DNS: Attempting connection...
18:52:51 DNS: Attempting connection...
18:52:52 DNS: Attempting connection...
18:52:53 DNS: Attempting connection...
18:52:54 DNS: Attempting connection...
18:52:55 DNS: Attempting connection...
18:52:56 DNS: Attempting connection...
18:52:57 DNS: Attempting connection...
18:52:58 DNS: Attempting connection...
18:52:59 DNS: Attempting connection...
18:53:00 DNS: Attempting connection...
18:53:01 DNS: Attempting connection...
18:53:02 DNS: Attempting connection...
18:53:03 DNS: Attempting connection...
18:53:04 DNS: Attempting connection...
18:53:05 DNS: Attempting connection...
18:53:06 DNS: Attempting connection...
18:53:07 DNS: Attempting connection...
18:53:08 DNS: Attempting connection...
18:53:09 DNS: Attempting connection...
18:53:10 DNS: Attempting connection...
18:53:11 DNS: Attempting connection...
18:53:12 DNS: Attempting connection...
18:53:13 DNS: Attempting connection...
18:53:14 DNS: Attempting connection...
18:53:15 DNS: Attempting connection...
18:53:16 DNS: Attempting connection...
18:53:17 DNS: Attempting connection...
18:53:18 DNS: Attempting connection...
18:53:19 DNS: Attempting connection...
18:53:20 DNS: Attempting connection...
18:53:21 DNS: Attempting connection...
18:53:22 DNS: Attempting connection...
18:53:23 DNS: Attempting connection...
18:53:24 DNS: Initialized
18:56:46 MQT: tele/50/STATE = {"Time":"2018-12-04T18:56:46","Uptime":"0T00:05:14","Vcc":3.170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":”XXXXXXXXXXXXX","BSSId":"XX:XX........","Channel":9,"RSSI":56}}

ascillato2, 2018-12-04:
Hi,

duplicated issue: https://github.com/arendst/Sonoff-Tasmota/issues/4526

closed 2018-12-04 by ascillato2

crispy78, 2018-12-04:
Looks like the same issue, with that difference that this device has the same settings just like all my other devices that don't have issues. That rules out my network in my opinion.

I've uploaded the 6.3.0.15 2.4.2 core version to all my devices including this one, still no difference with this particular device. With NodeMCU PyFlasher I wiped the memory of this particular unit and installed the 6.3.0.15 2.4.2 firmware on the device, then set all setting again. Now it's working again.