arendst/Tasmota #4532: 70% packet loss with 6.3.0 and esp8266 v 2.4.2

carlotodeschini, 2018-12-04:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE on Ubuntu 18.04
- [x] Hardware used : intel core i5-2410M
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` :

``21:41:22 CMD: status 0``
``21:41:22 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}``
``21:41:22 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:13:01","StartupUTC":"2018-12-04T20:28:21","Sleep":0,"BootCount":14,"SaveCount":45,"SaveAddress":"F7000"}}``
``21:41:22 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-12-04T21:13:22","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}``
``21:41:22 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<DELETED>",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}``
``21:41:22 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}``
``21:41:22 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2078","IPAddress":"<DELETED>","Gateway":"<DELETED>","Subnetmask":"255.255.255.0","DNSServer":"<DELETED>","Mac":"<DELETED>","Webserver":2,"WifiConfig":5}}``
``21:41:22 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"<DELETED>","MqttPort":1883,"MqttClientMask":"sonoff-01","MqttClient":"sonoff-01","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}``
``21:41:22 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 04 20:41:22 2018","Local":"Tue Dec 04 21:41:22 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:26","Sunset":"16:54"}}``
``21:41:22 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-12-04T21:41:22"}}``
``21:41:22 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-12-04T21:41:22","Uptime":"0T00:13:01","Vcc":3.580,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MaHaHa","BSSId":"A0:63:91:2F:CB:72","Channel":6,"RSSI":82}}}``
``21:41:23 MQT: Attempting connection...``
``21:41:27 MQT: Connected``

Compiled without problems with Arduino IDE and uploaded to Sonoff Basic. The board reboot and I can reach it. But the web interface works intermittently and the ping reveal a 70% packet loss:

``[...]``
``64 bytes from 192.168.10.65: icmp_seq=949 ttl=255 time=2.77 ms``
``^C``
``--- 192.168.10.65 ping statistics ---``
``961 packets transmitted, 241 received, +15 errors, 74% packet loss, time 977336ms``
``rtt min/avg/max/mdev = 1.707/232.600/6229.856/840.118 ms, pipe 7``

The wifi signal is good.
What am I doing wrong?
Thanks,
tode

ascillato, 2018-12-04:
Hi,

I could not reproduce your issue.

Try restarting your router and also erasing all flash of your sonoff basic with esptool.py and flashing again but using the latest Tasmota v6.3.0.15 from development branch at http://thehackbox.org/tasmota/

ascillato2, 2018-12-04:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-04 by ascillato2

carlotodeschini, 2018-12-06:
Flashed Tasmota v6.3.0.15 from development branch with esptool.py solved the problem, thanks.

mzealey, 2018-12-06:
I was getting this too with v6.3.0, sonoff basic and core 6.3.0. I could't track down the issue it looked like it may be related to TIMER compile options. It was not displaying on ESP12-F type standalone modules, only the sonoff basics that I tested it on. But current development version worked...