arendst/Tasmota #4535: Sending learned RFcode does not work with RFbridge

fverschure, 2018-12-05:
Hi All, 

To start with I love the tasmota firmware. The amount of supported devices is huge. It fits my needs of getting rid of firmware i do not trust with a open source one to connect with my open source domotica systems (Domoticz, HA, Node-Red)

Apologies if I am raining an issue which has been discussed before. 
I need some guidance on getting my RFbridge to work with a cheap set of RF outlets.

Target device: Sonoff RFBridge
Firmware: Sonoff-Tasmota v6.3.0 (precompiled)
Configuration: moduletype 25 (RF bridge)

Issue description:
Learning codes from the remote is successful as also seen in console.
When sending the leaned code the switch does not respond.
What am I missing here?

I did found issue #1387 where it was reported that the learned code was incorrect however I could not read how this was resolved.

Thanks for your guidance.

KR, Floor

meingraham, 2018-12-05:
I read in a recent post in the broader community something about needing to send the code as a hex value formatted accordingly (e.g., 0xHHHH). Just a suggestion. Not sure that's applicable to your situation.

fverschure, 2018-12-05:
I thought i  tried that as well, with same result but will try again tonight.

ascillato, 2018-12-05:
Please, use the latest Tasmota version 6.3.0.15 from the development branch (http://thehackbox.org/tasmota/) and try again.

If you stills experience the issue, please try the Tasmota Chat or open a new issue with the troubleshooting templete completed. Thanks.

ascillato2, 2018-12-05:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-05 by ascillato2

Jason2866, 2018-12-06:
Flash latest Portisch firmware to the RF chip. This is NOT the ESP chip!
Original Firmware has issues decoding Intertechno code (used in cheap Power RF Sockets)
See https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#rf-chip-firmware-upgrade