arendst/Tasmota #4543: Tuya Dimmer - need help

h4nc, 2018-12-06:
Hi, 

I followed these instructions to set up my new tuya dimmer:

https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer

I have two issues:

1) The dimmer will be mounted in a bedroom. Is there away to turn off the led's when the dimmer is off?

2) after setting up gpio13 to Rx and gpio15 to tx the device doesn't swith on off when I press the button on it. 
Before it worked, but the dimming buttons did not. 
I selectet the device ID SetOption34 2.
Will this always be "2" or is it an ID of the device itself? Where can I find it?

Jason2866, 2018-12-06:
Since there is not "the One" Tuya Dimmer nobody cant help without knowing which type you have

h4nc, 2018-12-06:
You are right, I'm sorry. 

I have the EU model. I also noticed that the layout of my pcb is a little different. The esp is faced the opposite as the picture showed it the documentation. 

[image]

Jason2866, 2018-12-06:
@chaosmaster  Can you help? I have no Tuya device (yet)....

chaosmaster, 2018-12-06:
@h4nc
>     The dimmer will be mounted in a bedroom. Is there away to turn off the led's when the dimmer is off?

Sorry, that's not possible, the LEDs are controlled by the MCU and can only switch between red and blue.
Only the wifi-led (green) LED is controlled by Tasmota.

>     after setting up gpio13 to Rx and gpio15 to tx the device doesn't swith on off when I press the button on it.
This dimmer uses GPIO1 as TX and GPIO3 as RX (the default for "Tuya Dimmer").

>     I selectet the device ID SetOption34 2.
>     Will this always be "2" or is it an ID of the device itself? Where can I find it?
Everything else should be detected automatically in the current development branch.
Otherwise `SetOption34 2` is correct for this device.
The ID depends on the type of Tuya-device and currently there are devices using _2_ or _3_

h4nc, 2018-12-06:
@chaosmaster 
Thanks for the fast reply. 

> Sorry, that's not possible, the LEDs are controlled by the MCU and can only switch between red and blue.

Is there a line I can cut to disable this three leds that are not managed by tasmota. 
If there is one line to cut, maybe its possible to connect it to the esp, so that it can be controlled by tasmota? Which pin would you recommend? 

So I have it configured like this now (see picture), but the dimmer-buttons don't work. 
[image]
What else could be wrong?

h4nc, 2018-12-06:
Every led has 4 pins. GND, BLUE, RED and?

It does seem like there is one line I could cut to trigger all blue and another to trigger als red lights.
I still want to use it in my bedroom, but the the leds lightning the whole room in blue in not an option.

How would you to it? Desolder the led's? Cut the lines to the leds?
I could also only cut the blue ones.
Best way for my still would be to be able to use them in tasmota by connecting them to the esp.

Any ideas are welcome!

Jason2866, 2018-12-06:
- The easiest mod would be cut trace(s)
- For connecting to the ESP (it seems GPIO 12-16 are free)
   you have to add a transistor. GPIO from ESP doesnt deliver enough current to connect it direct.
   The function of the LEDs could be done...

ascillato2, 2018-12-06:
Great to know, that now the dimmer is working for you.

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-06 by ascillato2

h4nc, 2018-12-06:
I marked the Pins of the Power Switch in this picture.

As I would expect a two color led needs 3 pins. So why are there 4 pins per led?

I think a will cut the traces, but first I need to now what the traces are for.

[image]

chaosmaster, 2018-12-06:
@h4nc 
You can check [this post](https://github.com/arendst/Sonoff-Tasmota/issues/469#issuecomment-399006536).
It describes what is connected where.
Although the LEDs aren't directly connected to the MCU, they are connected via the CR302-chip.
They are switched between blue and red by pulling the GPIO HIGH or LOW.
To disable them the GPIO can be configured as input.
You can also check out my custom MCU-firmware for this device, to disable the LEDs.
But be warned it isn't completely stable and I haven't had time to look further into it.

I suppose the have 4 pins since, they are all in fact RGB LEDs, although I believe only the top LED has green wired.

Here you can find my custom MCU-firmware:
https://github.com/chaosmaster/tuya_mcu

h4nc, 2018-12-06:
I still have one open question. I don't want to open a new issue for that.

Why do we have to connect NRST to GND for flashing the device? What does NRST stand for?

chaosmaster, 2018-12-06:
@h4nc 
NRST is the reset-pin for the MCU it's part of the MCUs programming-interface.
It needs to be pulled LOW during flashing the ESP, because the MCU is connected to the ESPs UART.

TommyChausson, 2019-01-24:
@h4nc I've got a few question about the EU model, i manage to flash it with the last Tasmota firmware. It work when piloting it by button, by mqtt, by tasmota interface but when i use the switch (toggle or up/down button) no informations are sent by mqtt ? Is there a specific configuration in "Module parameters" interface ? Actually i have : 
- GPIO0 - Button 1
- GPIO1 - Tuya Tx
- GPIO3 - Tuya Rx
- GPI14 -  Led1
 
Thank's

TommyChausson, 2019-01-24:
@h4nc  Yes, VERY strange, when it boot it send MQTT infos (state etc...), That prove that MQTT info sending is working, only when i touch a button nothing is sent ! What's your tasmota version ?

kajmaj, 2019-02-19:
I have Tuya Dimmer with  6.4.1(release-sensors), 2.3.0/1.5.3.
Everything works well but after restart (change configuration) it everytime went ON.
PowerOnState is 0

In hassio config retain: false
Is there anything possible to do with it?

h4nc, 2019-02-19:
Maybe adding this code to your automation.yaml could help, this will call the current state of all your sonoff. You could also change the topic to your tuya dimmer to only call its state.

```
- alias: 'Sonoff - State after Restart'
  initial_state: 'on'
  trigger:
  - event: start
    platform: homeassistant
  action:
    - service: mqtt.publish
      data:
        topic: cmnd/sonoffs/POWER
```

kajmaj, 2019-02-19:
I already have this automation:
```
  - alias: "power_state_on_ha_start_up"
    initial_state: "on"
    trigger:
    - platform: homeassistant
      event: start
    action:
    - service: mqtt.publish
      data:
        topic: "cmnd/sonoffs/power1"
        payload: ""
    - service: mqtt.publish
      data:
        topic: "cmnd/sonoffs/power2"
        payload: ""
    - service: mqtt.publish
      data:
        topic: "cmnd/sonoffs/Power3"
        payload: "" 
    - service: mqtt.publish
      data:
        topic: "cmnd/sonoffs/power4"
        payload: ""
    - service: mqtt.publish
      data:
        topic: "cmnd/sonoffs/power"
        payload: ""
```
In the case of Tuya it does not have any influence.

And this tuya config:

```
light:
- platform: mqtt
  name: "bedroom"
  state_topic: "stat/bedroom_dim/POWER"
  command_topic: "cmnd/bedroom_dim/POWER"
  availability_topic: "tele/bedroom_dim/LWT"
  brightness_state_topic: "stat/bedroom_dim/RESULT"
  brightness_command_topic: "cmnd/bedroom_dim/Dimmer"
  brightness_scale: 100
  brightness_value_template: >
    {% if value_json.Dimmer is defined %}
      {{ value_json.Dimmer }}
    {% else %}
        {{ state_attr('light.bedroom_dim','brightness') / 255 * 100 }}
    {% endif %}
  qos: 1
  payload_on: "ON"
  payload_off: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
  retain: false
```

h4nc, 2019-02-19:
Try:

```
    - service: mqtt.publish
      data:
        topic: cmnd/sonoffs/STATE
```

kajmaj, 2019-02-19:
response of Tuya Dimmer for STATE: 
`stat/bedroom_dim/RESULT = {"Time":"2019-02-19T18:35:15","Uptime":"0T08:27:51","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":53,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"HXXNET","BSSId":"F4:F8:6D:9B:94:A2","Channel":6,"RSSI":60}}`

h4nc, 2019-02-19:
I use state to get an update of the DIMMER value.

So this too did not help you with your problem, right?

kajmaj, 2019-02-19:
Well, after first restart not. After restart it is OK. It seems to help
Will see during a day or two.
Thank you

kajmaj, 2019-02-20:
Everything is back, Tuya dimmer powers on after every reboot.
With mqtt disabled it is OK, with mqtt enabled it is back.
On Tuya,  6.4.1.16(8cb33b8-sensors), 2_5_0/3.0.0-dev(c0f7b44)
PowerRetain OFF
```
19:06:21 CMD: powerretain
19:06:21 MQT: stat/dimmer_1/RESULT = {"PowerRetain":"OFF"}
```
On Hassio config:
`retain: false`
In addition each Hassio reboot:
```
    action:
    - service: mqtt.publish
      data:
        topic: "cmnd/sonoffs/power"
```

h4nc, 2019-02-20:
Looks like something triggers your dimmer to turn on after the restart.
I do not have an empty payload like you showed in the example above in my config.
Maybe you want to comment that out and try again.

kajmaj, 2019-02-20:
I stopped the whole automation with empty payload. It has no influence. Reseting Tuya toggles power to on.

h4nc, 2019-02-20:
Ok, I have no more ideas. Maybe a clean flash could help, but I don’t know.

joggs, 2019-03-07:
If you've had retain enabled in your settings earlier, that setting will be cached in hass/mqtt, no matter if you set it to disabled later on. When I had this problem I reinstalled my mqtt-service in home assistant and everything started working. I saw simpler ways to achieve this later on, but I cant remember where.

joggs, 2019-03-07:
> @h4nc I've got a few question about the EU model, i manage to flash it with the last Tasmota firmware. It work when piloting it by button, by mqtt, by tasmota interface but when i use the switch [...]

Did you try to set "SetOption34 2" in the console? After that change, my dimmer buttons started working as expected

marrobHD, 2019-05-12:
Mine wont flash :(.
```
resetting board
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        read 0, requested 1
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
```

telmomarques, 2019-08-15:
@TommyChausson did you ever solve your issue? I'm having the same problem, everything works but the Tasmota UI is not updated when I change something manually using the dimmer buttons...

TommyChausson, 2019-08-16:
@telmomarques , Yes and it was a problem with my domotic system sent too much MQTT messages (i had something like infinite loop).

tamirh-2, 2019-10-31:
hi,
i have the YDM_WIFI_SV4 model, how do i put it into flashing mode?

jtauscher, 2020-06-05:
> hi,
> i have the YDM_WIFI_SV4 model, how do i put it into flashing mode?

GPIO0 & 15 need to be grounded.

Dragao75, 2020-07-09:
Hi all, I have the same device (YDM_WIFI_SV4 
But I can't seem to get it flashed... 
I think I am in program mode (I see three blue led and the top one is also green) 
but I am unable to communicate with the device. 

I have tried grounding GPIO0 & 15 (also tried only grounding GPIO0)  
esptool will not communicate. used baud rate 115200 

What else can I do?

Dragao75, 2020-07-10:
Although this topic is closed I am hoping someone will read my post :) 
I was able to flash the dimmer with tasmota. But I am unable to dimm the light. 
I can dimm the light via the buttons but not via webinterface. I even don't have a slider in the webinterface. 
So my guess is I am using the wrong module (Using Tuya MCU 54) 

How can I troubleshoot this? 
(when I touch the buttons nothing is shown in the console)

ascillato, 2020-07-10:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

jtauscher, 2020-07-11:
> Although this topic is closed I am hoping someone will read my post :)
> I was able to flash the dimmer with tasmota. But I am unable to dimm the light.
> I can dimm the light via the buttons but not [...]

I think I had the same problem and it really sucked - took me hours to find out whats missing but actually its pretty simple.
Go to the console in the web interface and enter those two commands one by one:

**TuyaMCU 21,2
DimmerRange 1 255**

That should enable the slider etc. Hope that helps.

Cheers

hamodcraft22, 2020-08-18:
heyyyy so i just stumbled across this and i have a simple question....

 will this work with a "Tuya Fan" Device? 

after further investigation every thing is the same about the "motherboard" like just take a look 
[image]
[image]

the only differences are the (2018) which in mine is 2019 , and the face plate which has a fan rather then a light icon

peepshow-21, 2020-12-19:
Hi,

Struggling a bit with that last part of this :( 

Issue 1.

I have installed Tasmota & configured it. I can see the slider;
[image]

If I move it, I see events in the logs;
[image]

But the lights do not change they just stay on.

If I set it to zero, it goes off. any other value is on MAX. Toggle button works fine too. The dimmer will go up and down using the hardware buttons on the switch, although it can be via sluggish in showing those events in the console.

I've turned weblog 4 on and used the hardware buttons to find the range;
[image]

Which goes from 11 to 255, so I've done this;
*dimmerrange 11,255

No matter what I try, I can seem to get the light to do anything other than on/off via Tasmoto.

Issue 2.

If I press the ON button on the switch, it fades in up the values to come on slow. Same with OFF, if I press that, it fades down the values until off. This is VERY annoying, especially off. When you press it, it looks like nothing has happened so you press it again. Such a stupid implementation. Is there a config, so you press off and it just goes off? I think I'll remove them if there isn't, or wife will never let me add anything else in house EVER!! :)

Any ideas?

peepshow-21, 2021-01-11:
Just as an update for anyone else. GPIO12 was the issue. In another thread on this dimmer it said it needed set, as above. It doesn't, it need to be 'None'!! then it works.

I needed GPIO14 set to 3 for me, to turn the green LED off on button 1, otherwise the top one glows Cyan.

[image]

Only thing I wish though was that the hardware off button turned it off instantly, not fade to off. It's annoying...

jtauscher, 2021-02-08:
Problem is, the buttons dont talk with your ESP but withe the STM chip on your PCB. That cant be changed, as far as I know.

I wonder if there is a dimmer with same optics, touch etc but **trailing edge** instead of leading edge - anybody?

Trailing edge is much better for LEDs lights in general.

Xxyzzy, 2022-01-15:
Recently bought a konesky smart dimmer (from amazon). Date on the board 20210930. Still works with tasmota (flashed via serial interface).
There are problems with recent tasmota versions. Requires remapping of dimming for the MCU, etc and still does not  turn off when dim value is set to <= 9. Reports OFF via mqtt but does not turn off.
However it works flawlessly out of the box with older tasmota versions. I'm using tasmota-6.5.0-core-2.4.2.

alecsandes, 2023-01-12:
Hello!

I got the same dimmer and I want to flash out the tuya fw. 
Doing some digging, I didn't find where to connect what.

Ok, RX and TX is clear, than GND is GND and VCC is 3v3
Now, GPIO0 to GND? 
And the upper shunt, what is it?
I see in the picture that the wires are soldered o the chip, can't I use the side pins? if yes, which ones are to connect to?

Date on board 20220930.
Will esphome run on it as well?

TIA