arendst/Tasmota #4544: Tuya Dimmer in Home Assistant

h4nc, 2018-12-06:
Could someone provide a ready to use code of the tuya dimmer for home assistant?

Thanks!

ascillato, 2018-12-06:
Hi,

You have several examples provided in the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

Please, check them and adapt it to your configuration. Thanks

ascillato2, 2018-12-06:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-06 by ascillato2

h4nc, 2018-12-06:
So I used this example and changed it to my configuration and it says "not available", but I can receive the mqtt data on MQTTBox.

[image]

```
  - platform: mqtt
    name: "Dimmer"
    command_topic: "cmnd/sonoff_dimmer/POWER"
    state_topic: "stat/sonoff_dimmer/RESULT"
    state_value_template: "{{value_json.POWER}}"
    availability_topic: "tele/sonoff_dimmer/LWT"
    brightness_command_topic: "cmnd/sonoff_dimmer/Dimmer"
    brightness_state_topic: "stat/sonoff_dimmer/RESULT"
    brightness_scale: 100
    on_command_type: "brightness"
    brightness_value_template: "{{value_json.Dimmer}}"
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    qos: 1
    retain: false
```

h4nc, 2018-12-06:
Commenting the availability_topic out lets me control the device from ha. However it shows "unavailable" in the history. So something is wrong.