arendst/Tasmota #4548: 433 MHz RF Bridge vs my plug

mastix, 2018-12-06:
Hi guys,

I've searched through this whole list of (similar) issues, but non of the tipps worked. :(

Tasmota Version: 6.3.0

I have a 433 MHz remote controller power plug. The console recognized the ON and OFF commands:

```
00:34:58 MQT: sonoff/RESULT = {"RfReceived":{"Sync":10430,"Low":310,"High":850,"Data":"BC1C0F","RfKey":1}}
00:35:02 MQT: sonoff/RESULT = {"RfReceived":{"Sync":10440,"Low":300,"High":860,"Data":"BC1C0E","RfKey":2}}
```
And as you can see, I already set the rfkey1 and 2 (1 = ON = BC1C0F, 2 = OFF = BC1C0E).

Now I'd like to trigger the plug via MQTT, so I tried this first:

```
MQTT: sonoff/cmnd/rfkey1
=> 00:41:47 MQT: sonoff/RESULT = {"RfKey1":"Learned sent"}
===> Nothing happens. Same with rfkey2
```

So I digged deeper and found the backlog... So I checked some common timings and sent the following mqtt command (all other issues mentioned a different topic cmnd/sonoff/backlog, but that doesn't work for me):

```
MQTT: sonoff/cmnd/backlog
RfSync 10430; RfLow 310; RfHigh 850; RfCode BC1C0F;
=> 00:35:42 MQT: sonoff/RESULT = {"RfSync":10430}
=> 00:35:42 MQT: sonoff/RESULT = {"RfLow":310}
=> 00:35:42 MQT: sonoff/RESULT = {"RfHigh":850}
=> 00:35:43 MQT: sonoff/RESULT = {"RfCode":"BC1C0F"}
```
Nothing happend either. I'm totally lost now...

So my basic questions are:

What am I missing?
Has the topic changed from cmnd/sonoff... to /sonoff/cmnd/...?
Am I using rfcode wrong? When I enter rfcode into the console, I get the following number (always):

```
00:48:49 CMD: rfcode
00:48:49 MQT: sonoff/RESULT = {"RfCode":12327951}
```

I'd really appreciate it if you could point me into the right direction!

Thanks in advance!

ascillato2, 2018-12-07:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-07 by ascillato2

mastix, 2018-12-08:
Hi @ascillato2,

today I've flashed the Portish firmware to the RF chip and tried it again... no chance.

This is the one I've flashed: https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/generic_bucket/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

I've tried sending the rkey1 and rfkey2 directly:

`11:01:15 MQT: sonoff/RESULT = {"RfKey1":"Learned sent"}`

Nothing happens.

I've tried sending the whole package:

MQTT-TOPIC: sonoff/cmnd/backlog
MESSAGE: RfSync 10430; RfLow 240; RfHigh 800; RfCode #BC1C0E

`
11:03:21 MQT: sonoff/RESULT = {"RfSync":10430}
11:03:21 MQT: sonoff/RESULT = {"RfLow":240}
11:03:22 MQT: sonoff/RESULT = {"RfHigh":800}
11:03:22 MQT: sonoff/RESULT = {"RfCode":"#BC1C0E"}
`
Nothing happens.

Here's the info from the bridge:

`

Program Version | 6.3.0
-- | --
Build Date & Time | 2018-10-30T17:33:53
Core/SDK Version | 2_3_0/1.5.3(aec24ac9)
Uptime | 0T00:07:58
Flash write Count | 84 at FB000
Boot Count | 34
Restart Reason | Power on
  |  
AP1 SSId (RSSI) | MyWIFI(100%)
Hostname | sonoff-1343.local
IP Address | 10.0.0.27
Gateway | 10.0.0.1
Subnet Mask | 255.255.255.0
DNS Server | 10.0.0.114
MAC Address | DC:4F:23:C8:25:3F
  |  
MQTT Host | my.domain.com
MQTT Port | 1883
MQTT Client &  Fallback Topic | DVES_C9253F
MQTT User | myuser
MQTT Topic | sonoff
MQTT Group Topic | sonoffs
MQTT Full Topic | sonoff/cmnd/
  |  
Emulation | None
mDNS Discovery | Enabled
mDNS Advertise | Web Server
  |  
ESP Chip Id | 13182271
Flash Chip Id | 1327185
Flash Size | 1024kB
Program Flash Size | 1024kB
Program Size | 489kB
Free Program Space | 512kB
Free Memory | 13kB
`

Jason2866, 2018-12-08:
Have you deleted the learned keys? You have to learn it again if you change efm firmware!

Jason2866, 2018-12-08:
If this doesnt help try this efm firmware:
https://github.com/arendst/Sonoff-Tasmota/blob/development/tools/fw_efm8bb1/RF-Bridge-EFM8BB1-20181127.hex
This version works for my for problematics rf power sockets

Jason2866, 2018-12-08:
Do you can open the Rf Power sockets without damaging it?
If yes please do and look what rf chip is in there. If we know what type it is we know which code it uses

Jason2866, 2018-12-08:
Finally if this doesnt work we can try via bucket sniffing...
This needs more explanation and we need to do this on Discord Chat.
This is the wrong place here. The issue you have is NOT Tasmota related
It is a limitation from the RF Bridge device!!!

mastix, 2018-12-08:
Hey @Jason2866 ,

you might be right... I'll give it a try and let you know what the deal is... but on a different channel! :)

Thanks for the hints and tips! :)

Sascha