arendst/Tasmota #4549: JSON payload within a RULE?

poldim, 2018-12-07:
How can I include JSON formatting in my rule?  

- Should the payload below be in single quotes?
- Is there another way to force a `SENSOR` update to its normal topic on a state change of Switch2?

`rule1 on Switch2#State=1 do publish2 homeassistant/garage/door/SENSOR {"Switch2":"ON"} endon on Switch2#State=0 do publish2 homeassistant/garage/door/SENSOR {"Switch2":"OFF"} endon
1`

curzon01, 2018-12-07:
* Payload needs no extra quotes, your `publish2` command ist correct.
* You can not force SENSOR topic, but you can use `Status 8` command to force the same as in SENSOR topic

Did you set Rule1 4? Please post `Rule1` and `Status 0` output.

For further info about `Rule1 4` please read the Rule<x> command descriptions in cmnd [Management](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management)

ascillato2, 2018-12-07:
Hi,

I have tested your publish syntax and works fine.

```
08:47:42 CMD: publish homeassistant/garage/door/SENSOR {"Switch2":"ON"}
08:47:42 SRC: WebConsole from 192.168.1.131
08:47:42 RSL: Topic Recibido /publish, Tamaño de Datos 49, Datos homeassistant/garage/door/SENSOR {"Switch2":"ON"}
08:47:42 RSL: Grupo 0, Índice 1, Comando PUBLISH, Datos homeassistant/garage/door/SENSOR {"Switch2":"ON"}
08:47:42 MQT: homeassistant/garage/door/SENSOR = {"Switch2":"ON"}
```

[image]

So, you are having another issue if you are not seeing your message in your broker.

Please, complete the **_troubleshooting template_** with the actual outputs of your Tasmota console

closed 2018-12-07 by ascillato2

poldim, 2018-12-08:
> Did you set Rule1 4? Please post `Rule1` and `Status 0` output.

I did not set Rule1 4 as I don't need it to continuously blast the broker when the door is open, just on state change.  Is this not correct?

```
10:12:39 CMD: Rule1
10:12:39 MQT: homeassistant/garage/door/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":337,"Rules":"on switch2#state=1 do publish2 homeassistant/garage/door/SENSOR {"Switch2":"ON"} endon on switch2#state=0 do publish2 homeassistant/garage/door/SENSOR {"Switch2":"OFF"} endon"}
10:12:54 CMD: Status 0 
10:12:54 MQT: homeassistant/garage/door/STATUS = {"Status":{"Module":18,"FriendlyName":["WEMOS Garage Door"],"Topic":"homeassistant/garage/door","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":3,"SaveData":1,"SaveState":0,"SwitchTopic":"reedsensor","SwitchMode":[0,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
10:12:54 MQT: homeassistant/garage/door/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"2T17:20:09","StartupUTC":"2018-12-06T00:52:45","Sleep":0,"BootCount":60,"SaveCount":129,"SaveAddress":"3FB000"}}
10:12:54 MQT: homeassistant/garage/door/STATUS2 = {"StatusFWR":{"Version":"6.2.1.9","BuildDateTime":"2018-09-29T15:20:36","Boot":5,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
10:12:54 MQT: homeassistant/garage/door/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.1.100","LogPort":514,"SSId":["HAN",""],"TelePeriod":60,"SetOption":["000A8128","55818000","00000000"]}}
10:12:54 MQT: homeassistant/garage/door/STATUS4 = {"StatusMEM":{"ProgramSize":450,"Free":2620,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0F502790","00000000","000000C6","000000C0"]}}
10:12:54 MQT: homeassistant/garage/door/STATUS5 = {"StatusNET":{"Hostname":"WEMOS, Garage Door","IPAddress":"192.168.3.78","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"60:01:94:10:A7:1A","Webserver":2,"WifiConfig":5}}
10:12:54 MQT: homeassistant/garage/door/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.3.100","MqttPort":1883,"MqttClientMask":"WEMOS_garage_door","MqttClient":"WEMOS_garage_door","MqttUser":"poldim","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:12:54 MQT: homeassistant/garage/door/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 08 18:12:54 2018","Local":"Sat Dec 08 10:12:54 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-8,"Sunrise":"23:30","Sunset":"07:53"}}
10:12:54 MQT: homeassistant/garage/door/STATUS10 = {"StatusSNS":{"Time":"2018-12-08T10:12:54","Switch2":"ON","AM2301":{"Temperature":76.8,"Humidity":36.8},"TempUnit":"F"}}
10:12:54 MQT: homeassistant/garage/door/STATUS11 = {"StatusSTS":{"Time":"2018-12-08T10:12:54","Uptime":"2T17:20:09","Vcc":3.106,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HAN","BSSId":"8A:8A:20:21:20:3D","Channel":11,"RSSI":54}}}
```

ascillato, 2018-12-08:
No need to use rule1 4 (that means once)

So, what is the issue now? Your rule is ok and it is sending a JSON message to your broker.

> status 8 will output on a different topic so it wont work for immedietly update the status of the sensor in HomeAssistant.

Ok, publish to another topic like publish homeassistant/garage/door/MYSENSOR {"Switch2":"ON"}

ascillato, 2018-12-08:
you can:

* Add a rule in Tasmota for SYSTEM#BOOT to publish the actual value of your sensor when Tasmota starts.
* Add an automation in Home Assistant to be triggered when HA starts to ask Tasmota for that value (see example in the wiki at homeassistant)

curzon01, 2018-12-09:
> No need to use rule1 4 (that means once)

That means disable one :wink: 
`Rule1 5` means once

poldim, 2018-12-10:
Thanks Adrian.
Is there a way for me to obtain from the device where it's expecting the
trigger when HA starts?  Like I mentioned earlier I've set my config to not
use prefixes and my typical cmnd topic `homeassistant/garage/door/cmnd/POWER`
doesn't seem to work.

ascillato, 2018-12-10:
> Is there a way for me to obtain from the device where it's expecting the
> trigger when HA starts?

Yes, please, check the example of the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#tip-sync-auto-discovered-devices-including-power-state-version-with-power-state-usage