arendst/Tasmota #4551: connect to mqtt-Broker without password

UltiCa, 2018-12-07:
tasmota 6.3.0

MQT: Verbindung fehlgeschlagen aufgrund von 10.100.44.96:1883, rc -2

My MQTT-Broker works fine. (other devices can connect, send, and receive Messages)

my-user-config.h
#define MQTT_USE               1
#define MQTT_HOST              "192.168.4.96"
#define MQTT_FINGERPRINT1      "A5 02 xxxxx
#define MQTT_FINGERPRINT2      "A5 02 xxxxx
#define MQTT_PORT             1883       
#define MQTT_USER              ""            
#define MQTT_PASS              ""

the MQTT-Broker is used without User and Password

What is the MQTT_Fingerprint for?
What are the next steps to find the problem?

thx for help

curzon01, 2018-12-07:
Next steps are please read issue template and fill out the issue template :wink: (you completely have deleted the issue template).

rc -2 means `MQTT_CONNECT_FAILED - the network connection failed` - see [CONNECTION FAILED](https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview#connect-failed) within [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki)

MQTT_FINGERPRINT is useless as long you do not have TLS enable.

ascillato2, 2018-12-07:
Hi,

You don't need to modify anything in the code. You can configure your MQTT Broker inside Tasmota using the WEB MENU or by commands.

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-07 by ascillato2