arendst/Tasmota #4553: {"WARNING":"Enable weblog 2 if response expected"} but "weblog 2" is set

mvincm, 2018-12-07:
**Describe the bug**
When using "backlog" with http request (e.g. http://192.168.22.162/cm?cmnd=Backlog%20Power;Status%200) the returned information is "{"WARNING":"Enable weblog 2 if response expected"}" but "weblog 2" is set for sure. It is important to notice that commands are done on sonoff side but there is no correct return information.  

- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : sonoff basic 
- [x] _Development/Compiler/Upload tools used_ : esptool
- [x] _If a pre-compiled release or development binary was used, which one?_ : 6.2.1 but tested also on 6.3.x
- [x] _You have tried latest release or development binaries?_ : no
- [x] _Provide the output of command_``status 0`` :

```
16:50:08 CMD: STATUS 0
16:50:08 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff_mm"],"Topic":"sonoff_1","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
16:50:08 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"7T22:28:51","StartupUTC":"2018-11-29T17:21:17","Sleep":0,"BootCount":26,"SaveCount":61,"SaveAddress":"F7000"}}
16:50:08 RSL: STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:50:08 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["KoroFlat",""],"TelePeriod":60,"SetOption":["04008000","55818000","00000000"]}}
16:50:08 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
16:50:08 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff_mm","IPAddress":"192.168.22.162","Gateway":"192.168.22.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.22.1","Mac":"84:F3:EB:48:6D:51","Webserver":2,"WifiConfig":5}}
16:50:08 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 07 15:50:08 2018","Local":"Fri Dec 07 16:50:08 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:29","Sunset":"16:53"}}
16:50:08 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-12-07T16:50:08","DS18B20":{"Temperature":20.9},"TempUnit":"C"}}
16:50:08 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-12-07T16:50:08","Uptime":"7T22:28:51","Vcc":3.185,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"KoroFlat","RSSI":82,"APMac":"34:CE:00:63:20:41"}}}
```

**To Reproduce**
http://192.168.22.162/cm?cmnd=Backlog%20Power;Status%200

**Expected behavior**
I expected some JSON with return information from "Power" and "Status 0" commands.

**Additional context**
There is no difference which commands I used. In my opinion problem is with "backlog".

Best regards,
MvincM

ascillato, 2018-12-07:
Hi,

That is just a message. Not an error. That is that if you want to see any error messages displayed you need a weblog 3 or 4.

This was already addressed a couple of times in issued.

I will propose a PR to delete this message that is confusing some users.

Thanks.

closed 2018-12-07 by ascillato2

mvincm, 2018-12-07:
Ok. Right. Now is clear for me!