arendst/Tasmota #4556: RGB strip turn always on after reboot

papperone, 2018-12-07:
**Describe the bug**
If MQTT in activated, when a reboot is happening even if the POWER message retained is OFF the LED Strip is setup to the latest "color" value; this is annoying as after a power outage all my stip led are ON as soon as electricity is back;

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ :  Arilux, H801 and selfmade module with power mosfet
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [X] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
18:01:14 MQT: ledtvsalotto/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["LEDTVSalotto"],"Topic":"ledtvsalotto","ButtonTopic":"ledtvsalotto","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
18:01:14 MQT: ledtvsalotto/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:32:39","StartupUTC":"2018-12-07T16:28:35","Sleep":0,"BootCount":38,"SaveCount":363,"SaveAddress":"F9000"}}
18:01:14 MQT: ledtvsalotto/stat/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:01:14 MQT: ledtvsalotto/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR81",""],"TelePeriod":300,"SetOption":["00108088","55818000","00000000"]}}
18:01:14 MQT: ledtvsalotto/stat/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
18:01:14 MQT: ledtvsalotto/stat/STATUS5 = {"StatusNET":{"Hostname":"ledtvsalotto-0221","IPAddress":"192.168.1.106","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"5C:CF:7F:B7:40:DD","Webserver":2,"WifiConfig":5}}
18:01:14 MQT: ledtvsalotto/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.231","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B740DD","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:01:14 MQT: ledtvsalotto/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 07 17:01:14 2018","Local":"Fri Dec 07 18:01:14 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:29","Sunset":"16:53"}}
18:01:14 MQT: ledtvsalotto/stat/STATUS10 = {"StatusSNS":{"Time":"2018-12-07T18:01:14"}}
18:01:14 MQT: ledtvsalotto/stat/STATUS11 = {"StatusSTS":{"Time":"2018-12-07T18:01:14","Uptime":"0T00:32:39","Vcc":2.823,"POWER":"OFF","Dimmer":100,"Color":"FFFFFF","HSBColor":"0,0,100","Channel":[100,100,100],"Scheme":0,"Fade":"ON","Speed":3,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"NETGEAR81","RSSI":68,"APMac":"A4:2B:8C:F5:15:E0"}}}
```

**To Reproduce**
Set RGB Strip to any color, turn then the strip off (cmnd/power off) and reboot the module, the strip is turned on with latest color setup and the topic stat/POWER is set to ON

**Expected behavior**
at reboot the presence of the "cmnd/color" topic should not trigger the power ON command

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-12-07:
Please, use the command weblog 4 to have more debug information in the Tasmota console. After that, reboot your device and you will see `SRC: [something]` before the power on.

That will give you the clue why it is turning on.

May be, you have retained messages in your MQTT broker.

ascillato2, 2018-12-07:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-07 by ascillato2

papperone, 2018-12-07:
ok the issue is the "/cmnd/scheme" that is retained and being "0" it triggers power on, how can I fix this behaviour?

papperone, 2018-12-07:
I don't want to turn off retention of my mosquitto as I will have a lot of issues in my setup so this is *not* a solution to a bug which seems quite obvious...

ascillato2, 2018-12-07:
Hi, not a bug. It is designed like that. See commands at the wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Scheme    0    (default) Use a single color for all pixels in light/strip/ring