arendst/Tasmota #4558: PWM Color/Dimmer

DetlefR, 2018-12-07:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-12-07:
Please read the wiki. Thanks

closed 2018-12-07 by ascillato2

DetlefR, 2018-12-07:
Hello,

I use a MagicHome controller with 3 separate LED stripes. To use the FADE feature SetOption15 is ON .
To change the the PWM value I use the CHANNEL>1-3> command.
My problem. Every time I set  a new value for a channel the other two channels are reduced.
A example. Channel 2 is set by a PIR. Every command reduced channel 1 and 3.

'stat.dhaled3.color':'F8007D'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F6F97C'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F5F97B'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F4F97A'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F2F978'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F1F977'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F3F979'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F0F976'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'EFF975'

Is there a solution.
Thanks