arendst/Tasmota #4563: Adjust delta temperature

sommo, 2018-12-08:
It could be nice to adjust temperature reading by adding or removing for example +2 to the temperature value (or -2, -1.. and so on)

Jason2866, 2018-12-08:
You can do this via rules

ascillato, 2018-12-08:
There is an specific example of this in rules. Please, check the wiki. Thanks

ascillato2, 2018-12-08:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-08 by ascillato2

sommo, 2018-12-08:
i tried to modifiy si7021 read temperature with the rule example, and publish it but it seems that it complete change the output command (i'm using rule example n. 8b):
original without rule:
16:42:58 MQT: tele/mansarda/SENSOR = {"Time":"2018-12-08T16:42:58","ANALOG":{"A0":4},"SI7021":{"Temperature":24.5,"Humidity":56.9},"TempUnit":"C"}

and after:
16:37:58 MQT: tele/mansarda/SENSOR = {"Time":"2018-12-08T16:37:58","ANALOG":{"A0":4},"SI7021":{"Temperature":24.4,"Humidity":56.8},"TempUnit":"C"}
16:37:58 RUL: TELE-SI7021#TEMPERATURE performs "backlog backlog var1 24.4; add1 -2; event sendtemp"
16:37:59 MQT: stat/mansarda/RESULT = {"Var1":"24.4"}
16:37:59 MQT: stat/mansarda/RESULT = {"Add1":"22.40"}
16:37:59 MQT: stat/mansarda/RESULT = {"Event":"Done"}
16:37:59 RUL: EVENT#SENDTEMP performs "publish tele/mansarda/SENSOR 22.40"

it do the job but doesnt publish the complete output like:

MQT: tele/mansarda/SENSOR = {"Time":"2018-12-08T16:37:58","ANALOG":{"A0":4},"SI7021":{"Temperature":**22.40**"Humidity":56.8},"TempUnit":"C"}

cause something else need to read from this type of format string (and when rule is on it reads incorrect value).
i'm guessing right that it publish the modified value via rule in a different way than before?

ascillato, 2018-12-08:
Tasmota will publish the value of the sensor to tele/mansarda/SENSOR every teleperiod time.

the rule in the example adds another publish value with the modified temperature, so you will need to publish under another topic and let your home automation software to read your Modified Temp Topic instead.

you can publish for example to something like tele/mansarda/TEMP

sommo, 2018-12-08:
no way it get error reading anyway, 0, i successfull create a new publish, like very looklike the original:
but i need the %var1% with the same decimal like the original one:

original: 24.2
modified by rule 24.20

i think this is what cause the issue,
i'm looking to the wiki

sommo, 2018-12-08:
i'm tring calcres but i can't figure it out

ascillato, 2018-12-08:
CalcRes command is the solution for that. But that is only available in latest version 6.3.0.16. Please update from http://thehackbox.org/tasmota/

sommo, 2018-12-08:
thank you, for now i use another script to get the modified temperature value (i found a way from the other side), until waiting this CalcRes will be implemented into official tasmota firmware release,
thank you!