arendst/Tasmota #4565: WS2812 wrong Color

danst0, 2018-12-08:
Hi,

Something seems to be wrong with my WS2812 strip on my Tasmota with Wemos D1 mini.
I send a HSB Color of 120,100,70 via MQTT (or console) gives me red LEDs (instead of green). I think the color channels seem to be mixed up...
Is there a way to change the assignment?

Daniel

Jason2866, 2018-12-08:
You have to compile your own version and change in my_user_config.h
`  #define USE_WS2812_CTYPE     NEO_GRB           // WS2812 Color type (NEO_RGB, NEO_GRB, NEO_BRG, NEO_RBG, NEO_RGBW, NEO_GRBW)`

ascillato2, 2018-12-08:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-08 by ascillato2

jeroenst, 2025-10-28:
Thank you, yes an old thread which I used before for info about this. At least it's now up to date again.