arendst/Tasmota #4567: OBI socket v2 parameter assignment

gitterman, 2018-12-08:
I frequently see those screen shots of module parameter assignments like here: https://github.com/arendst/Sonoff-Tasmota/wiki/OBI-Socket-2
Is there a list which describes the meaning of the selectable parameter values?

I have flashed the latest 6.3.0 with https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.3.0/sonoff-DE.bin and selected the generic device.
The OBI v2 has the relay on GPIO4 and the button S1 on GPIO5.
There is a green LED driven by GPIO12 which is labelled "Link" and a red LED driven by GPIO13 which is labelled "Power" and which shows the relay status.
I tried various assignments for GPIO12 and GPIO13 but none worked as expected.
With the setup from the 1st link the green LED never lights and the red LED is always blinking no matter what the relay status is.
I ended up assigning "Relay 1" to the red LED on GPIO13 to make it go along with the relay.
Alas for the wifi link I cannot find a suitable setting.

BTW: is there an US-english binary which corresponds to sonoff-DE.bin?

ascillato, 2018-12-08:
> Is there a list which describes the meaning of the selectable parameter values?

Yes, in the wiki

> is there an US-english binary which corresponds to sonoff-DE.bin?

Yes, **sonoff.bin** (The difference between files are explained in the release notes)

If you need further support, please find us in the Tasmota Support Chat at https://discord.gg/Ks2Kzd4 Thanks

andrethomas2, 2018-12-08:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-08 by andrethomas2

chaosmaster, 2018-12-10:
@gitterman I suggest you assign GPIO12 to LED1. That will be used to indicate connection state.
See https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main (LedState) to see when it will blink.
For the power LED (GPIO13) it seems you have already found a viable solution.
If that configuration works for you as expected, you may want to update the information in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/OBI-Socket-2)

devZer0, 2019-01-20:
@gitterman, did that work for you? can you independently use power and wifi led?

gitterman, 2019-01-20:
@devZer0, no it did not work. GPIO13 needs to be connected to LED1 to show the correct power state.

devZer0, 2019-01-20:
> Closing this issue as it has been answered.
> ### Support Information
> 
> See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
> See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

please don't close tickets which have not been solved. providing an answer is not always a solution and you scare away people reporting bugs/issues with this, because they don't feel welcome to have reported a bug.  it's sort of "is there a problem? no there is no problem - look at the tickets..."  - 16 open bugtickets and 4250 closed one looks interesting to me. i have never seen such numbers in open-source projects, it looks that there is sort of very consequent "bug closing policy".

devZer0, 2019-01-20:
gitterman replied that it did not work. for me it did not work either. so can someone please re-open this bugticket?

arendst, 2019-01-20:
Instead of opening this ticket it would be more useful if you provided a working template for the Obi V2.

So find out where the link and power leds are connected and enable wifi/mqtt status using ``SetOption31 0``

chaosmaster, 2019-01-20:
It looks like the current template already has both LEDs defined:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/sonoff_template.h#L1685-L1686

devZer0, 2019-01-20:
ah - ok, yes, i can choose Obi Socket 2 (61) now and whith default options it's blinking because of mqtt - but when i disable mqtt the green led is permanently off. but it should be on, as there is a wifi link. so it still seems buggy..

devZer0, 2019-01-20:
i found a working configuration:

  Generic 18
  D2 GPIO4 - Relay1 (21)
  D1 GPIO5 - Button1 (17)
  D6 GPIO12 Relay1 Led1i (56)
  D7 GPIO13 Led1i Led2 (53)
  

@gitterman, can you try & confirm ?

chaosmaster, 2019-01-20:
@devZer0 The status LED only blinks when there is **NO** WiFi and/or MQTT-connection and stays off otherwise IIRC.
Check the [commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main) (specifically _LedState_ and _SetOption31_ as @arendst has mentioned.

devZer0, 2019-01-20:
i would expect a green wifi status led to be "on" when connected to the wifi and would expect blinking when there is an error/disconnect.

others seem to expect this, too,  https://github.com/arendst/Sonoff-Tasmota/issues/2230

maybe you could otherwise explain the "philosophy" when it's off intentionally.

unfortunately, i don't have the original firmware to test default behaviour and the manual doesn't provide that information.

the manual for the other obi socket (type 1) with the blue led is telling "blue led=on  when configuration is successful"

chaosmaster, 2019-01-20:
Well it doesn't really matter, what the manual for the original firmware says, since tasmota handles things differently.
With _LedState 0_ it will only blink when WiFi and/or MQTT-Connection issues occur.
With _LedState 1_ it will be lit when the device is powered on.
I like the default behavior of not lighting the LED when WiFi is connected.
As you have figured out you can invert the behavior by selecting _GPIO_LED1_INV_ instead of _GPIO_LED1_.

devZer0, 2019-01-20:
ok, can live with that. but then let's document that in the wiki, ok?

chaosmaster, 2019-01-20:
I believe it's already documented well enough.
But feel free to add this information to the wiki-page for the device.

devZer0, 2019-01-20:
lets wait for the next release. i'm using 6.4.1.9 (pulled lates development version) and it seems to work ok, but 6.4.1 has issues with the green led and i cannot manually override appropriately

gitterman, 2019-01-20:
@devZer0, I cannot confirm your settings. 
D7 GPIO13 Led1i  needs to be LED1 to show the red light if the plug is ON.
I have disabled MQTT to prevent blinking as I have no MQTT but using the device standalone.
I can live without a wifi status but I would appreciate if the green LED showed the Wifi Status as steady green to indicate a working wifi and blinking green to show an error condition.

devZer0, 2019-01-20:
dear repository owner, why did you delete my comment regarding issue handling without any furhter notice? 

because i dared to comment/criticise?

feeling muzzled.

gitterman, 2019-01-21:
@devZer0 - I started the firmware update via webserver and all I got was 6.4.1.

devZer0, 2019-01-21:
you may wait for the next release then...