arendst/Tasmota #4568: SerialSend3

mcc05, 2018-12-08:
If I use SerialSend3 to issue the following with an escape character TEST\r the output the is generated is TEST<CR>r, the r character is output in serial text

ascillato2, 2018-12-10:
Hi,

Escape character is set using command `serialdelimiter` (see wiki at commands)

Could not reproduce your Baudrate issue.

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-10 by ascillato2

mcc05, 2018-12-10:
@ascillato Hi so the code states serialsend3 is 
else if (3 == index) {
          uint16_t dat_len = data_len;
          Serial.printf("%s", Unescape(dataBuf, &dat_len));  // "Hello\f"
        }
So if I send serialsend Hello\r is would expect Hello with a CR, my delimiter was set a 13, this still does not explain the extra r in the string i.e. HelloCRr

And on the baudrate, so I set it to 9600 using the baudrate command in the console. If I then restart the device it reverts to 115200.