arendst/Tasmota #4571: Gosund SP1 reboot loop / forgot settings / self overwrite settings

DojokunOss, 2018-12-09:
I have 4 pieces of Gosund SP1 flashed with dev release 6.3.0.14, self compiled with Atom.

After different time intervals each of the device do an automatic reboot and forget some settings. There is no manual action on the device and there is nothing on MQTT. I use Mosquitto on a Raspi, have full log enabled and the one and only inside the mosquitto log is device register and disconnect messages.

If I set "weblog 4" for a detailed logging and "setoption13 1" (taken from issue #1612 and #773 ), I can see, that there is no reboot, but the setoption13 value switch back to "0" automatically. I have no idea, what the source of the value switch is. After reboot, the weblog value is forgotten also. Last issue is, that there are some Power on / off switches without any action via MQTT or manual doing.

At first, here is the output from "status 0"
`12:23:17 CMD: status 0
12:23:17 MQT: stat/Dose3/STATUS = {"Status":{"Module":55,"FriendlyName":["Dose3"],"Topic":"Dose3","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:23:17 MQT: stat/Dose3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:41","StartupUTC":"2018-12-09T11:22:36","Sleep":0,"BootCount":1,"SaveCount":6,"SaveAddress":"F6000"}}
12:23:17 MQT: stat/Dose3/STATUS2 = {"StatusFWR":{"Version":"6.3.0.14(sonoff)","BuildDateTime":"2018.12.04 17:07:56","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:23:18 MQT: stat/Dose3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HolzGarage","MediaServer5"],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
12:23:18 MQT: stat/Dose3/STATUS4 = {"StatusMEM":{"ProgramSize":504,"Free":496,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","0FDAE794","240383A0","23B617CE","00003BC0"]}}
12:23:18 MQT: stat/Dose3/STATUS5 = {"StatusNET":{"Hostname":"Dose3-3184","IPAddress":"192.168.1.53","Gateway":"192.168.1.99","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.99","Mac":"60:01:94:BA:4C:70","Webserver":2,"WifiConfig":4}}
12:23:18 MQT: stat/Dose3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.48","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BA4C70","MqttUser":"Dose3","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:23:18 MQT: stat/Dose3/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 09 11:23:18 2018","Local":"Sun Dec 09 12:23:18 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:31","Sunset":"16:53"}}
12:23:18 MQT: stat/Dose3/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:23:18 MQT: stat/Dose3/STATUS10 = {"StatusSNS":{"Time":"2018-12-09T12:23:18","ENERGY":{"TotalStartTime":"2018-12-09T12:22:33","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":225,"Current":0.000}}}
12:23:18 MQT: stat/Dose3/STATUS11 = {"StatusSTS":{"Time":"2018-12-09T12:23:18","Uptime":"0T00:00:42","Vcc":3.174,"LoopSet":50,"LoadAvg":27,"POWER":"ON","Wifi":{"AP":2,"SSId":"MediaServer5","BSSId":"C0:4A:00:8D:6B:47","Channel":6,"RSSI":86}}}`

And here a tracing, that you can, what I've described (Important: There was no button pressing in this time, bat the log says "SRC: Button" with differnt type of topics):
`00:00:00 Projekt sonoff Dose3 (Topic Dose3, Fallback DVES_BA4C70, GroupTopic sonoffs) Version 6.3.0.14(sonoff)-2_3_0
00:00:00 WIF: verbinden mit AP1 HolzGarage in Modus 11N wie Dose3-3184...
00:00:06 RSL: stat/Dose3/RESULT = {"POWER":"OFF"}
00:00:06 RSL: stat/Dose3/POWER = OFF
00:00:10 WIF: Verbindung fehlgeschlagen, da der AP nicht antwortet (timeout)
00:00:10 WIF: verbinden mit AP2 MediaServer5 in Modus 11N wie Dose3-3184...
00:00:13 WIF: verbunden
00:00:13 DNS: initialisiert
00:00:13 HTP: Web-Server aktiv bei Dose3-3184.local mit IP-Adresse 192.168.1.53
00:00:14 MQT: Verbindungsversuch...
00:00:14 MQT: verbunden
00:00:14 MQT: tele/Dose3/LWT = Online (beibehalten)
00:00:14 MQT: cmnd/Dose3/POWER = 
00:00:14 MQT: tele/Dose3/INFO1 = {"Module":"Gosund SP1 v23","Version":"6.3.0.14(sonoff)","FallbackTopic":"DVES_BA4C70","GroupTopic":"sonoffs"}
00:00:14 MQT: tele/Dose3/INFO2 = {"WebServerMode":"Admin","Hostname":"Dose3-3184","IPAddress":"192.168.1.53"}
00:00:14 MQT: tele/Dose3/INFO3 = {"RestartReason":"Software/System restart"}
00:00:14 MQT: stat/Dose3/RESULT = {"POWER":"OFF"}
00:00:14 MQT: stat/Dose3/POWER = OFF
10:29:41 CMD: WEBLOG 4
10:29:41 MQT: stat/Dose3/RESULT = {"WebLog":4}
10:29:42 CFG: in Flash gespeichert am F5, zählen 7, Bytes 3584
10:29:45 MQT: tele/Dose3/STATE = {"Time":"2018-12-08T10:29:45","Uptime":"0T00:00:23","Vcc":3.186,"LoopSet":50,"LoadAvg":21,"POWER":"OFF","Wifi":{"AP":2,"SSId":"MediaServer5","BSSId":"C0:4A:00:8D:6B:47","Channel":6,"RSSI":76}}
10:29:45 MQT: tele/Dose3/SENSOR = {"Time":"2018-12-08T10:29:45","ENERGY":{"TotalStartTime":"2018-12-08T10:29:19","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
10:29:45 CMD: setoption13 1
10:29:45 SRC: WebConsole from 192.168.1.8
10:29:45 RSL: empfangenes topic /setoption13, Datengröße 1, Daten 1
10:29:45 RSL: Gruppe 0, Index 13, Befehl SETOPTION, Daten 1
10:29:45 MQT: stat/Dose3/RESULT = {"SetOption13":"ON"}
10:29:46 CFG: in Flash gespeichert am F4, zählen 8, Bytes 3584
10:29:55 WIF: Prüfe Verbindung...
10:29:55 WIF: verbunden
10:30:13 SRC: Button
10:30:13 RSL: empfangenes topic /SetOption13, Datengröße 1, Daten 0
10:30:13 RSL: Gruppe 0, Index 13, Befehl SETOPTION, Daten 0
10:30:13 MQT: stat/Dose3/RESULT = {"SetOption13":"OFF"}
10:30:14 CFG: in Flash gespeichert am FB, zählen 9, Bytes 3584
10:30:16 WIF: Prüfe Verbindung...
10:30:16 WIF: verbunden
10:30:24 SRC: Button
10:30:24 MQT: stat/Dose3/RESULT = {"POWER":"ON"}
10:30:24 MQT: stat/Dose3/POWER = ON
10:30:24 CFG: in Flash gespeichert am FA, zählen 10, Bytes 3584
10:30:24 APP: Knopf1 Mehrfachdruck 1
10:30:36 WIF: Prüfe Verbindung...
10:30:36 WIF: verbunden
10:30:53 SRC: Button
10:30:53 MQT: stat/Dose3/RESULT = {"POWER":"OFF"}
10:30:53 MQT: stat/Dose3/POWER = OFF
10:30:53 APP: Knopf1 Mehrfachdruck 1
10:30:54 CFG: in Flash gespeichert am F9, zählen 11, Bytes 3584
10:30:56 SRC: Button
10:30:56 MQT: stat/Dose3/RESULT = {"POWER":"ON"}
10:30:56 MQT: stat/Dose3/POWER = ON
10:30:56 APP: Knopf1 Mehrfachdruck 1
10:30:56 CFG: in Flash gespeichert am F8, zählen 12, Bytes 3584
10:30:56 WIF: Prüfe Verbindung...
10:30:56 WIF: verbunden
10:31:17 WIF: Prüfe Verbindung...
10:31:17 WIF: verbunden
10:31:24 CMD: setoption32 1
10:31:24 SRC: WebConsole from 192.168.1.8
10:31:24 RSL: empfangenes topic /setoption32, Datengröße 1, Daten 1
10:31:24 RSL: Gruppe 0, Index 32, Befehl SETOPTION, Daten 1
10:31:24 MQT: stat/Dose3/RESULT = {"SetOption32":"1"}
10:31:24 CFG: in Flash gespeichert am F7, zählen 13, Bytes 3584
10:31:38 WIF: Prüfe Verbindung...
10:31:38 WIF: verbunden
10:31:47 SRC: Button
10:31:47 MQT: stat/Dose3/RESULT = {"POWER":"OFF"}
10:31:47 MQT: stat/Dose3/POWER = OFF
10:31:47 APP: Knopf1 Mehrfachdruck 1
10:31:47 CFG: in Flash gespeichert am F6, zählen 14, Bytes 3584
10:31:48 SRC: Button
10:31:48 RSL: empfangenes topic /Reset, Datengröße 1, Daten 1
10:31:48 RSL: Gruppe 0, Index 1, Befehl RESET, Daten 1
10:31:48 MQT: stat/Dose3/RESULT = {"Reset":"Reset and Restarting"}
10:31:49 CFG: Standard verwenden`

Jason2866, 2018-12-09:
Have you erased whole flash before flashing Tasmota?
If not do so. Fragments left from old firmware can do weired things....
If issue still persists try precompiled bin from http://thehackbox.org/tasmota/sonoff-DE.bin

ascillato2, 2018-12-09:
Hi, your config reset seems to be due your button got pressed for more than 40 seconds. This is a full reset feature.

You can disable this using the proper setoption command (see wiki)

If you are not pressing the button you have a hardware problem. May be you have attached another button without pull up or it is not wired properly.

closed 2018-12-09 by ascillato2