arendst/Tasmota #4572: rule validator

domenpk, 2018-12-09:
**Have you look for this feature in other issues and in the wiki?**
Yes. Was not able to find it.

**Is your feature request related to a problem? Please describe.**
I had this rule:
`rule1 on clock#timer=1 var1 1 endon`
It never got triggered and it took me ages to figure out why.
Correct:
`rule1 on clock#timer=1 do var1 1 endon`
There was never any error, so I looked at everything but the incorrect rule syntax.

**Describe the solution you'd like**
Some sort of feedback when rule is incorrect.

**Describe alternatives you've considered**
"rule validator" could also be a separate webpage (not on the device), although that would be less convenient.

ascillato2, 2018-12-09:
Nice idea.

Please, feel free to do it. Thanks

closed 2018-12-09 by ascillato2

chaosmaster, 2018-12-10:
@ascillato why was this closed?
It does sound like an interesting idea.