arendst/Tasmota #4573: Sonoff T1 not connecting to wifi

askuechler, 2018-12-09:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Atom,espressif8266@1.5.0
- [x] Hardware used : MacBook Pro
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : yes
- [ ] Provide the output of command ``status 0`` :
```
not feasable - see description below
```
**(Please, remember to close the issue when the problem has been addressed)**

Hi, I have a strange behaviour on setting up a Sonoff T1 2channel using the latest development version of tasmota (but I guess this is not a software related issue anyhow). After flashing and reboot the T1 joins my wifi and I am able to configure the sonoff as T1 2channel. In fact as long as it is powered by my ftdi adapter I can do just anything to it and it works (apart from switching any real relays of cause). When I join the board back to the T1 relay board and power it from 230V both button leds first flash bright and then come up dim and the wifi led stays out, the T1 does not connect to wifi. Pressing reset results in a beep and flash of both button leds but still no wifi. Extracting the board again and just attaching vcc,gnd from my ftdi adapter starts the board and it joins wifi. Any help is appreciated.

Andreas

Silas007, 2018-12-09:
I had the exact same problem. I needed to update the Board Info inArduino IDE to Version 2.4.2 to flash correctly. After that, it worked even on 230V

Good luck, Jan

ascillato, 2018-12-09:
Hi, your Tasmota version was not specified in your description. Please, try with latest Tasmota version 6.3.0.16

ascillato2, 2018-12-09:
Closing this issue as it has been answered. If you need more help, please find us in the Tasmota Chat. Thanks. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-09 by ascillato2

askuechler, 2018-12-09:
as stated I am on the latest devel tree - version 6.3.0.16 and just tried
changing the environment to 2.4.2 - no luck, same reaction.

Silas007, 2018-12-10:
Well, strange. Did you check if every paramter is correct like in the tutorial?
While trying to figure out a solution for my problem, I read that some users needed to use 2.3.0. They had to downgrade so that it worked. For me though, 2.4.2 did solve it.