arendst/Tasmota #4575: Armtronix

iodr100, 2018-12-09:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iodr100, 2018-12-09:
Hi
I have used the last version of tasmota to use it with armtronix single dimmer board but it still not work 
I have also flashed atmega328p with ino file https://github.com/armtronix/Wifi-Single-Dimmer-Board/tree/master/Wifi-Single-Dimmer-Board/Arduino_Code/Atmega_Single_dimmerV0_5
Could anyone help me to resolve it i have to integrate this device in my final year project

ascillato, 2018-12-09:
Hi, for support please use the Tasmota Support Chat. Thanks

closed 2018-12-09 by ascillato2