arendst/Tasmota #4580: improve scheduler user interface

gitterman, 2018-12-10:
The scheduler on the web interface is very nice to use smart plugs without an dedicated home automation infrastructure.
I could not find any explanatory examples on the wiki so it took me a while to get it work though.
There should be some hints on the scheduler page e.g. for the +/- function.
Do I need to set a weekday just to trigger a one time action?
Do I need to select all weekdays to trigger repeated everyday action?
As most people live in a rhythm of monday..friday and weekends the day grouping should reflect this.
I know that the week begins with sunday in the US but also US people live according to the above rhythm. 
If a schedule needs a weekday specification there should be a wildcard box for "every day"

gitterman, 2018-12-10:
Tanks for the clarification.
I am not very familiar with wiki editing but I will try to add something perhaps during Christmas break.
I understand that RAM is crucial, so my suggestion then is to have all weekdays enabled by default.
I think this is closer to the standard use case than the current implementation.
If I want the next day I would not need to change anything on the weekday specification and if I want specific days I would only need to untick those not wanted.
BTW: Would arranging the weekdays Mo..Fr,Sa,Su be more programming effort than Su,Mo..Fr,Sa ?

andrethomas, 2018-12-10:
> I think this is closer to the standard use case than the current implementation.

I'm not sure there is a standard for this and just as you might find it having to click all the days being annoying I would find it annoying having to uncheck them all the time.

I think it is fine in its current form.

gitterman, 2018-12-10:
I am not a programmer but rather an expert on ergonomics and psychological aspects
As such I know that people learn more from examples than from rules or algorithms.
Also in everyday use people make assumptions to the reasonable defaults.
E.g. if I said the next meeting would be at 14:00 without specifying a day most people assume the current day or the next day if the time has already passed.
Also if I said the lunch break is always at 11:30 people would assume am and Monday through Friday even if this is not specified.
For scheduler entries this means an active time spec without a day spec would be an everyday spec for most people. 
From the ergonomic point of view that would mean though that all weekdays should be selected by default to show clearly what would happen.
Also from the ergonomic point of view in the worst case I need to tick 7 boxes to get "every day" whereas I would only need to untick 6 or less boxes if I wanted one ore more specific days.

It is your software though and if you think it is fine in its current implementation that is ok.
I just wanted to shine a bit of different light on the subject.

closed 2018-12-10 by gitterman

gitterman, 2018-12-12:
>Thought about installing home assistant software instead of using the webui of the firmware?

Like most I don't live in my own house.
I have just a few smart plugs in my apartment and no other smart components yet.
I wanted to get rid of the different apps I needed to control them and I wanted to become independent from those Chinese based control servers which may go out of order at some point in time.
An infrastructure with a central station would be an overkill for me at the moment.
Thus the web interface is just great for my needs.
I may get myself one of those cheap Echo Dot devices for Xmas to be able to conveniently work with voice control.

ascillato, 2018-12-12:
>An infrastructure with a central station would be an overkill for me at the moment

So, please, visit https://www.home-assistant.io
You will need just a common raspberry pi to install it. That software acts like a local server for all your smart devices, and all your data will be managed locally and over your own wifi network. 

There are also other options for home automation softwares like openhab, domoticz, node red, etc etc.