arendst/Tasmota #4583: NTP time set failure

flexiti, 2018-12-10:
1. You can not check what time is set in the device (now the only place is logs), maybe you can add to the Information box? or any possibility of displaying on the main page such as temperature ?

2. Sometimes it happens that the time from the ntp server is not set when the device is turned on, and only helps reset 

( I have several devices working (6), and sometimes one "refuses" to set NTP time - starts from 00:00:00,
and time does not repair for many hours until reset)

andrethomas, 2018-12-10:
Please ensure you're using nearby NTP servers (or preferably one which your router makes available)

I have 63 devices running Sonoff-Tasmota 6.3.0.16 and never had NTP issues, even with unstable 6.2.1 releases historically.

ascillato2, 2018-12-11:
Hi,

After some tests, the time in main page looks very ugly. Sorry.

You should check time in the console instead.

Please, also check the local NTP server solution to make more reliable the time in your network. Thanks.

closed 2018-12-11 by ascillato2

flexiti, 2018-12-11:
I have the impression that the posts read without understanding and closes without thinking ... just to close;) I changed the NTP server to local, I also wrote that I have 6 ESP8266 and they work at the same time, and the mentioned ESP8266 requires a reset to start work correctly again. It is obviously not the fault of the NTP server, nor is it specific ESP8266 because it happens randomly on different. (every few days)

ascillato, 2018-12-12:
Ok, for your last comment seems that we didn't understand what you were asking/proposing

You ask:

> maybe you can add to the Information box? or any possibility of displaying on the main page such as temperature ?

I spend some time testing that on the main webpage and looks very ugly, besides, that information is already in the console. So, the request is not being added soon or as it is.

> Sometimes it happens that the time from the ntp server is not set when the device is turned on, and only helps reset

You receive 2 comments about the local NTP. We don't have that issue in our setups. I have several devices testing both NTP approaches (local NTP with that hassio ADDON) and the default internet NTP from Tasmota (and testing different Arduino cores). And never had issues. So, as the issue is in your network setup, the concept of local NTP Server was explained.

So, if you think we didn't understand what you wrote, please explain yourself further in order to have more information to properly help you.

tcurdt, 2020-03-29:
I came here for the very same reason. Console would be fine, too. But I didn't really find a command to read the current time and timezone. Set it to `99` now and hope it works OK for Germany.