arendst/Tasmota #4587: Problem Flashing sonoff pow r2 v0.2

woudenbe, 2018-12-11:
I managed to flash th16, basic r2 but the pow r2 v0.2 with date 2018-06-11 i cant get flashed.

I have tried with PC, with raspberry, with pull up resistor between 3.3 and rx none of them worked.

arendst, 2018-12-11:
Pls provide some pictures of the Pow PCA

woudenbe, 2018-12-11:
[image]
[image]

woudenbe, 2018-12-11:
[image]
[image]

ascillato2, 2018-12-11:
@woudenbe 

Can you check with a DMM if the button is indeed connected to GPIO0 ?

Are you pushing the button before powering up the POW with your USB ?

woudenbe, 2018-12-11:
Hi ascilato2

I have not checked with DMM but will check.

I have even connected gpio0 with a wire to ground.

Yes gpio0 connector to ground before powering up.

I saw pull up resistor mentioned as a possible solution for similar issue.

woudenbe, 2018-12-11:
just checked with DMM, and yes button connects the GPIO 0 to GND.

ascillato, 2018-12-11:
Ok, so instead of using a wire, please try keeping the button pressed before and during the flashing process.

woudenbe, 2018-12-11:
[image]
[image]

VCC = 3.264 Vdc
tried with 1 sec press
2 sec press
press until flash
press during flash

all the time same result:

[image]

ascillato, 2018-12-11:
your connections seems ok.

the stock firmware is working fine?

ascillato, 2018-12-11:
From your photos, seems that the issue is a cable that is not making contact.

Could you test all your wires?

I have the same POW and the flashing was fine. Nothing weird. I did just as the wiki explains.

woudenbe, 2018-12-11:
I will check later tonight and update. Thanks for the tip

woudenbe, 2018-12-11:
all cables correct and also connected (measured on pcb side)

jziolkowski, 2018-12-11:
Did you change all the required settings to enable serial port on Rpi3?

https://elinux.org/RPi_Serial_Connection

andrethomas, 2018-12-11:
Also, test the serial interface with termite by connecting a dupont wire from tx to rx on the pi - whatever you type/send should be returned on the screen

Jason2866, 2018-12-11:
Out of ideas...
If i where you i would try a standard USB serial converter with a CH340 chip

woudenbe, 2018-12-11:
i flashed the other sonoff devices in same config...   completely puzzled here. it seems ti be teh new version of the POW R2 that is causing the issue. i have 5 and none of them working....

i have tried with FT231X
i have tried raspberry pi 3b+ (with correct settings)

i will try LH2303HX with an old PC in windows 7

if this will not work i need to wait for delivery (new FTDI) next week

i realy suspect the pow R2 v2.0 to be the issue here and i have FIVE !!  all behaving same

ascillato, 2018-12-11:
May be the measurement MCU is interfering in your flashing ?

Jason2866, 2018-12-11:
Have you measured direct on ESP chip if pressing button really is grounding GPIO 0?

woudenbe, 2018-12-11:
just tried PL2303HX with windows 7 and it worked. now try the other ones :-)

woudenbe, 2018-12-11:
first need to get wifi working :-)

ascillato, 2018-12-11:
Great!!!

So your issue was the Raspberry? Now your issue is solved?

woudenbe, 2018-12-11:
well not sure, as rasp berry & ft231X  gave same problem

woudenbe, 2018-12-11:
seems to work now. Thanks for all the help guys !

ascillato, 2018-12-11:
Great!!!

Nice to know

Please, close this issue. Thanks

closed 2018-12-11 by ascillato2

gavind87, 2018-12-15:
I'm having the same issue trying to flash with an Rpi
What could it be ? 
@woudenbe did you find out more information ? 
Quite lost and disappointed as I have 3 of these currently and none of them will flash

woudenbe, 2018-12-15:
I have flashed 5 sonof pow units now with a PL2303HX.  i still have one left that i cannot flash

gavind87, 2018-12-15:
@woudenbe so I managed to flash it on my Mac with a usb-ftdi 

Weird. Truly weird

seritos, 2018-12-30:
Hello everyone,

I am also having some issues trying to flash the new pow R2 v2.0. I have one FT232x that has given me no issues at all with other devices. 
After reading your coments... what is the correct procedure to do so? 
Thanx!

Jason2866, 2018-12-30:
Buy a usb serial adapter with a CH340 chip, with 3.3 volt support.
FTDI can work but is extremly out of it specs. It is not rated to deliver the needed current on the 3.3v rail

Jason2866, 2018-12-30:
It is the same procedure. Just pressing and holding button before you connect
power and tx rx gnd. After connecting you can release the button
Now start esptool to flash
Flashing process is described in wiki

seritos, 2018-12-31:
The adaptor cable wasnt good enough. Replaced and working perfectly

michalstral, 2019-04-27:
> Buy a usb serial adapter with a CH340 chip, with 3.3 volt support.
> FTDI can work but is extremly out of it specs. It is not rated to deliver the needed current on the 3.3v rail

I did what Jason2866 said and after issues with FTDI cheap adapter I used CH340G on Wifi Witty board and it works without problems. Thanks Jason2866, you saved me a hour at least.

adverserath, 2021-03-29:
FML that was hard to flash. Tried FTDI, Raspberry PI and CH340G. Had no success with any. Could get a minicom though, then checked the pins to the chip on my multimeter and there was a break in the board connection between the chip and the pin! Had to flash it holding my multimeter pin onto the chip with a crocodile clip on a wire to the CH340G. I hope to god I dont need to reflash that one.
Anyway, CH340G is the way to go!

esbenr, 2021-04-17:
I have the same problem Using an Arduino with Reset to GND to disable the Atmel chip.
Any generic solution other than buying specific hardware that "maybe" works?

jziolkowski, 2021-04-17:
Yes, heed to what wiki says and buy a USB adapter based on CH340G.

Belsebub, 2021-08-03:
Hi all

I have a strange problem with exactly this board when trying to flash using a raspi4.
At the moment I connect VDD with 3.3v PIN1 on raspi, the raspi reboots not matter if the button is pushed or not...
What did I wrong ?

Jason2866, 2021-08-03:
Powering from PI is problematic. Pi does not deliver enough current