arendst/Tasmota #4589: Tasmota webserver sometimes does not work

mrteausa, 2018-12-11:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mrteausa, 2018-12-11:
I noticed that a few of my MiniWemo and NodeMCU boards loaded with Tasmota latest firmware sometimes, the webserver does not work.  But the HA MQTT still works.  I have to remove all my connected PINS of the board and power on and off then the Tasmota webserver works again.  This happened pretty often on the boards that I leave it on for days..

ascillato2, 2018-12-11:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-11 by ascillato2

mrteausa, 2018-12-11:
My wi-fi us fine, I use UniFi system.  I noticed that sometimes when I remove the relays connection to all the PINS the webserver works again.  Any output that not suppose to connect to the relays/switches cause it?  From time to time I need to reboot it to get to the Tasmota webserver.

ascillato, 2018-12-11:
Are you powering up the relay coils directly from the nodemcu pins?

NodeMCU does not have enough energy to be used directly with relays, You need transistors in the middle.

onilo09, 2018-12-11:
I have the same problem, but on the sonoff basic on the new firmware 6.3.0, the wi fi connection holds ~ 5 minutes after which it reboots, on 5.14 it works stably.
Now in the house there are 6 sonoff basic, 4 pieces at 5.14 do not lose touch for 3 days, and 2 pieces at 6.3.0 reboot frequently, the light flickers when it is restarted, and sometimes it hangs and does not give access to the web interface, only through mqtt
Translated from Russian through translator

onilo09, 2018-12-11:
ele/momroom/LWT = Он-лайн (нераспред.)
23:03:08 MQT: cmnd/momroom/POWER = 
23:03:30 WIF: Подключение к AP1 Onilo1 в режиме 11N как momroom-6845...
23:03:34 WIF: Соединен
23:03:34 DNS: Инициализировано
23:03:34 MQT: Попытка подключения...
23:03:34 MQT: Соединен
23:03:34 MQT: tele/momroom/LWT = Он-лайн (нераспред.)
23:03:34 MQT: cmnd/momroom/POWER = 
23:04:12 DNS: Инициализировано
23:04:12 MQT: Попытка подключения...
23:04:12 MQT: Соединен
23:04:12 MQT: tele/momroom/LWT = Он-лайн (нераспред.)
23:04:12 MQT: cmnd/momroom/POWER = 
23:04:42 DNS: Инициализировано
23:04:42 MQT: Попытка подключения...
23:04:42 MQT: Соединен

onilo09, 2018-12-12:
thanks, helped

mrteausa, 2018-12-12:
Yes, I power up the relays on the MiniWemo 5v pinout.  Any diagram how to connect to the transistor or can I just connect the external power source?  What fixed on the new update version?

ascillato, 2018-12-12:
You need to connect an external power source. The wemos regulator is designed to power itself only. A relay uses more energy than the available in a wemos. 

For the complete list of fixes for last version please see changelog. Link in the readme (main page of Tasmota repository)

mrteausa, 2018-12-13:
Any easy way to do this with diagram and links correctly using bank of relays with transistor?