arendst/Tasmota #4591: SHT30 shield Not able to get reading from HA

mrteausa, 2018-12-11:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-12-11:
Incomplete issue.

For questions, please use the Tasmota Support Channel.

Github issues are for Bug reports. Thanks

closed 2018-12-11 by ascillato2

mrteausa, 2018-12-11:
I followed Tasmota configuration on SHT30 shield on MiniWemo and it getting the reading on webserver just fine.  But when I put this line value_template: '{{ value_json["SHT30-0x45"].Temperature }}', on the configuration.yaml for HomeAssistant, the return value is unknown.  I use DT22 module just fine but not this SHT30 module.  What do I need to change to get the reading on HA?

mrteausa, 2018-12-11:
Where is the Tasmota Support Channel? I am able to find it.  You respond and closed before I have a chance to finish typing my comments/issues..