arendst/Tasmota #4600: HA 'MQTT JSON Light' component support

radiohe4d, 2018-12-12:
I've looked around and can't seem to find any information about Tasmota supporting the Home Assistant MQTT JSON Light component.

https://www.home-assistant.io/components/light.mqtt_json/

Is this something that could be integrated easily? 

I'm new to Python but a professional .Net developer. Is this something that could be easily added by myself?

emontnemery, 2018-12-12:
The Home Assistant MQTT JSON Light component is deprecated from Hass 0.84, and is replaced by Home Assistant MQTT Light with three supported schemas:
- Basic
- JSON
- Template

Tasmota supports the "Basic" schema.
Details here: https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

ascillato2, 2018-12-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-12 by ascillato2

radiohe4d, 2018-12-12:
I'm sorry but that did not fully answer my question. 

I understand that currently Tasmota only supports 'Basic' but is it possible to add the ability to use the JSON schema also?

radiohe4d, 2018-12-12:
Yes I am aware now that MQTT JSON LIght is deprecated. 

What I am now saying, is if the new 'MQTT Light' component has JSON built in, is it then possible to add this new component and have the 'JSON' schema in it work with Tasmota as well at the 'Basic' schema?

radiohe4d, 2018-12-12:
Simply put, the current MQTT Light component sends up to 3 separate messages to change power, rgb color, brightness, ...

This sometimes leads to Tasmota not always receiving all the messages when it should and sometimes in the wrong order too. This causes lights to flicker and colors to jump and change.

But if we where able to send a JSON message to the Tasmota device with all the settings in 1 message, then it would solve all of these issues of flickering and wrong colors.

If we could send something like this:

'''
**HA Config -**

light:
  - platform: mqtt
     command_topic: "cmnd/<my-light>/set"
'''

'''
**JSON Payload -** 

{
  "brightness": 255,
  "color_temp": 155,
  "color": {
    "r": 255,
    "g": 180,
    "b": 200,
    "x": 0.406,
    "y": 0.301,
    "h": 344.0,
    "s": 29.412
  },
  "effect": "colorloop",
  "state": "ON",
  "transition": 2,
  "white_value": 150
}
'''

emontnemery, 2018-12-12:
@radiohe4d:
> This sometimes leads to Tasmota not always receiving all the messages when it should and sometimes in the wrong order too. This causes lights to flicker and colors to jump and change.

Are you sure about this?

- With the settings documented on the Wiki there should not be any flickering or jumps in color, can you please give an example of when this happens. 
- MQTT guarantees message order, there is no way the messages can be delivered in the wrong order.
- If the WIFI-reception is so poor that connection between MQTT server and Tasmota is dropped, implementing support for the JSON schema will not solve that.

radiohe4d, 2018-12-12:
Ill do more testing and see if I can narrow down and confirm its always happening. Its definitely not the WiFi connection but could be some other interference.

There is mention of this flickering in the HA documentation for MQTT Light component:

```
rgb_command_topic:

(string)(Optional)The MQTT topic to publish commands to change the light’s 
RGB state. Please note that the color value sent by Home Assistant is normalized 
to full brightness if brightness_command_topic is set. Brightness information is 
in this case sent separately in the brightness_command_topic. 
This will cause a light that expects an absolute color value (including brightness) 
to flicker.
```

I was more or less hoping that if we could send the JSON payload then the brightness flicker would not occur because its setting the brightness and color at the same time rather than a few milliseconds apart in different MQTT payload messages.

If you think this wouldn't help the situation then we can just ignore my suggestion :P

Another reason i forgot to mention was the ability to use transitions. This isn't available in the current setup, is it?

emontnemery, 2018-12-12:
Can you share your configuration.yaml section for the light which misbehaves?

radiohe4d, 2018-12-13:
```
- platform: mqtt
  name: "Living Room Front Light"
  availability_topic: "tele/lfl/LWT"
  command_topic: "cmnd/lfl/POWER1"
  state_topic: "stat/lfl/RESULT"
  state_value_template: "{{value_json.POWER}}"
  brightness_command_topic: "cmnd/lfl/Dimmer"
  brightness_state_topic: "stat/lfl/RESULT"
  brightness_value_template: "{{value_json.Dimmer}}"
  brightness_scale: 100
  rgb_command_topic: "cmnd/lfl/Color2"
  rgb_state_topic: "stat/lfl/RESULT"
  rgb_value_template: "{{value_json.Color.split(',')[0:3]|join(',')}}"
  effect_command_topic: "cmnd/lfl/Scheme"
  effect_state_topic: "stat/lfl/RESULT"
  effect_value_template: "{{value_json.Scheme}}"
  effect_list:
    - 0
    - 1
    - 2
    - 3
    - 4
  payload_on: "ON"
  payload_off: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
  qos: 1
```

emontnemery, 2018-12-21:
Ahh, so you are changing brightness and color at the same time?

That's a limitation in in Hass' `light.mqtt` with `basic` schema as it will send two MQTT messages, one for brightness change and one for color change.

Using the `json` schema would solve it, but I'm not sure adding yet another way to set/get light state to Tasmota is a good idea.

@arendst , @ascillato2 What do you think?

arendst, 2018-12-21:
I guess we have more than enough HASS specifics now...

radiohe4d, 2018-12-21:
Ok, thanks for looking into it anyway guys, appreciate your help.

emontnemery, 2018-12-22:
> when I send a Color command, the Dimmer value is set back to 100.

You should use color2 for this reason, please refer to the examples on the wiki.