arendst/Tasmota #4602: Tasmota threw error in HASS on MQTT RGB light turning off

pszafer, 2018-12-12:
So when MQTT light is turned off it throwing an error in HASS if RGB/state topic are the same.
I thought that it is a bug in HASS so made PR https://github.com/home-assistant/home-assistant/pull/19193, but @emontnemery said that it has been already fixed.

Do you mean that it is fix by this issue? https://github.com/arendst/Sonoff-Tasmota/issues/4358

```
- platform: mqtt
  name: "LED light"
  command_topic: "cmnd/led/POWER"
  state_topic: "stat/led/RESULT"
  state_value_template: "{{value_json.POWER}}"
  availability_topic: "tele/led/LWT"
  brightness_command_topic: "cmnd/led/Dimmer"
  brightness_state_topic: "stat/led/RESULT"
  brightness_scale: 100
  on_command_type: "brightness"
  brightness_value_template: "{{value_json.Dimmer}}"
  color_temp_command_topic: "cmnd/led/CT"
  color_temp_state_topic: "stat/led/RESULT"
  color_temp_value_template: "{{value_json.CT}}"
  rgb_command_topic: "cmnd/led/Color2"
  rgb_state_topic: "stat/led/RESULT"
  rgb_value_template: "{{value_json.Color.split(',')[0:3]|join(',')}}"
  effect_command_topic: "cmnd/led/Scheme"
  effect_state_topic: "stat/led/RESULT"
  effect_value_template: "{{value_json.Scheme}}"
  effect_list:
    - 0
    - 1
    - 2
    - 3
    - 4
  payload_on: "ON"
  payload_off: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
```

pszafer, 2018-12-12:
I'll check it after work and let you know, but it seems that you just sending tele message as well with stat, so it has to work :+1:

ascillato2, 2018-12-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-12 by ascillato2

emontnemery, 2018-12-12:
@pszafer Although @ascillato2 was correct in closiong this issue, please report back on if your problem is solved or not. There has been a lot of changes in both Hass and Tasmota recently to improve compatibility between Tasmota and Hass, and I would appreciate feedback.

pszafer, 2018-12-15:
Everything is working except ```tele/led/STATE``` is not triggered by sending Scheme command

emontnemery, 2018-12-15:
@pszafer scheme should be fixed, do you still see this issue on latest version?