arendst/Tasmota #4603: Compile error for Wemos D1 Mini

sdiekstra, 2018-12-12:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Platform IO (Home 2.0.0 / Core 3.6.3b2) @ Atom (1.33.0 x64)
- [x] Hardware used : Wemos D1 Mini (clone)
- [x] If a pre-compiled release or development binary was used, which one? : none
- [x] You have tried latest release or development binaries? : yes
- [x] Provide the output of command ``status 0`` : don't know what this means
```
Yesterday I've received my Wemos D1 Mini clone and I followed the Wiki to install Tasmota on it, which worked like a charm. I was able to install Tasmota succesfully and went back and forth to adjust the code, compile it and upload it to the board without problems. After spending some time at the webgui I wanted to update the code and compile it again, which resulted in lots of errors during the build.
I think Atom and PlatformIO were updated in the background, nothing else was changed.

After the problems started I tried several things, reinstalling Atom, reinstalling PlatformIO, clearing out the cache and lib directories (there are a lot of those...). I even tried compiling it on my Mac, which led to the exact same issues.
Problem is that I have a (somewhat) working Wemos, but are not able to recompile the code, so I can't make any changes are program other Wemos boards.

This it the output I get;
`
Processing wemos-d1-mini (platform: espressif8266; board: esp01_1m; framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)

(...)

Compiling .pioenvs\wemos-d1-mini\src\core_esp8266_timer.c.o
Compiling .pioenvs\wemos-d1-mini\src\core_esp8266_wiring_digital.c.o
Compiling .pioenvs\wemos-d1-mini\src\core_esp8266_wiring_pwm.c.o
Compiling .pioenvs\wemos-d1-mini\src\sonoff.ino.cpp.o

Generating LD script .pioenvs\wemos-d1-mini\ld\eagle.app.v6.common.ld
Compiling .pioenvs\wemos-d1-mini\libfae\PubSubClient-EspEasy-2.6.09\PubSubClient.cpp.o
Compiling .pioenvs\wemos-d1-mini\libc42\SPI\SPI.cpp.o
Compiling .pioenvs\wemos-d1-mini\lib9e8\NeoPixelBus-2.2.9\internal\HsbColor.cpp.o
In file included from D:/Downloads/Sonoff/Sonoff-Wemos/sonoff/sonoff.ino:32:0:
sonoff\my_user_config.h:47:0: warning: "CFG_HOLDER" redefined [enabled by default]
#define CFG_HOLDER             4617              // [Reset 1] Change this value (max 32000) to load SECTION1 configuration parameters to flash
^
In file included from C:\users\sander\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include/osapi.h:29:0,
from C:\users\sander\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:11,
from C:\users\sander\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:256,
from c:\users\sander\appdata\local\temp\tmpw02juw:1:
sonoff/user_config.h:47:0: note: this is the location of the previous definition
#define CFG_HOLDER             4620              // [Reset 1] Change this value (max 32000) to load SECTION1 configuration parameters to flash
^
In file included from D:/Downloads/Sonoff/Sonoff-Wemos/sonoff/sonoff.ino:32:0:
sonoff\my_user_config.h:58:0: warning: "WIFI_GATEWAY" redefined [enabled by default]
#define WIFI_GATEWAY           "192.168.1.1"     // [IpAddress2] If not using DHCP set Gateway IP address
^
In file included from C:\users\sander\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include/osapi.h:29:0,
from C:\users\sander\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:11,
from C:\users\sander\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:256,
from c:\users\sander\appdata\local\temp\tmpw02juw:1:
sonoff/user_config.h:58:0: note: this is the location of the previous definition
#define WIFI_GATEWAY           "192.168.2.254"   // [IpAddress2] If not using DHCP set Gateway IP address
^
In file included from D:/Downloads/Sonoff/Sonoff-Wemos/sonoff/sonoff.ino:32:0:
sonoff\my_user_config.h:60:0: warning: "WIFI_DNS" redefined [enabled by default]
#define WIFI_DNS               "192.168.1.1"     // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)
^
In file included from C:\users\sander\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include/osapi.h:29:0,
[... 565 lines cut ...]
extra_scripts = ${common.extra_scripts}

[env:wemos-d1-mini]
platform = espressif8266
framework = arduino
board = esp01_1m
board_build.flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=1000
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_speed = 115200
; *** Upload Serial reset method for Wemos and NodeMCU
upload_resetmethod = nodemcu
upload_speed = 115200
;upload_port = COM6
; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/strip-floats.py, pio/sftp-uploader.py
; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/strip-floats.py, pio/http-uploader.py`

To be honest, I'm out of options, I've tried all things I could think of and could find on the forums, anyone an idea?

andrethomas, 2018-12-12:
Please don't try a custom config file until you know how to use it.

sdiekstra, 2018-12-12:
Just followed the wiki about editing the platformio.ini file, and it worked fine, before the errors started...

andrethomas, 2018-12-12:
Fix this section in my_user_config.h which you appear to have changed to include "user_config.h"
```
#ifdef USE_CONFIG_OVERRIDE
  #include "user_config_override.h"         // Configuration overrides for my_user_config.h
#endif
```

As per my previous post please do not use custom config file until you know how to use it.

For now ignore the user config override and just make changes in my_user_config.h - Do this until you understand how user config override fits into the equation.

Github Issues is not intended for providing you with a basic understanding of how to compile code. It is for bug and feature request tracking and management.

andrethomas2, 2018-12-12:
Please note that GitHub Issues are for reporting bugs and requesting additional features.

Please search the Wiki for the information you require or visit Discord Chat where general questions and answers are handled.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-12 by andrethomas2