arendst/Tasmota #4604: Topic for each device like FriendlyName

sindudas, 2018-12-12:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
I have 3 Sonoff T1 (1ch , 2ch, 1ch). First 1ch connected to a light, and the 2ch/Button1 needs to manage that light. Then 2ch/button2/relay2 is connected to other light, that the other 1ch needs to manage it.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
I think will be more clear, to set the same Topic for one light on 1ch, and for 2ch/button1. And another Topic for the other Light on 2ch/button2 and other 1ch.
This way, the button status will be sinced with Mqtt. I use Mqtt to monitor the state, and for change it, with Homebridge and/or Openhab.
Now I'm doing it with Rules, but not achieved correctly because the "Remote" button doesn't change it state if main button is used.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-12-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-12 by ascillato2

sindudas, 2018-12-12:
Hi

I've reviewed the Wiki many times. And I may be missing something, and not understanding correctly something.

SwitchTopic is for using external Switch, not Button that Sonoff T1 (1ch/2ch/3ch) have. ButtonTopic changes the Topic, but changes for all Buttons.

Sorry if what I say sounds like a newbie, but I didn't find a clean solution for that.

Thanks.

ascillato, 2018-12-12:
Ok. But using rules should work.

Please, post your rules.
Use command weblog 4 for more debug information and post the output of your console when you use them
And please post also your status 0

Let us help you with the rules configuration to achieve what you want. Thanks

sindudas, 2018-12-12:
Here the rules:

[A] Sonoff T1/1ch Topic[escalera1] 1-[Remote]
`on Button1#State=2 do backlog Power1 2; publish escalera2/cmnd/power2 %value% endon`

[B] Sonoff T1/2ch Topic[escalera2] 1-[Remote] 2-[Local]
`on Button1#State=2 do backlog Power1 2; publish pasillo1/cmnd/power %value% endon`

[C] Sonoff T1/1ch Topic[pasillo1] 1-[Local]
no rules

When pressing Button1 on [A], triggers Relay2 on [B]. But if I press Button2 on [B], Relay1 on [A] (also LedButton) stays the same.

Maybe adding the same rules in a cross way? Like this:
[B] 
`on Button2#State=2 do backlog Power2 2; publish escalera1/cmnd/power1 %value% endon`

[C]
`on Button1#State=2 do backlog Power1 2; publish escalera2/cmnd/power1 %value% endon`

If needed the all other information, I'll post when I'm at home
Thanks

ascillato, 2018-12-12:
Well, you have just post your solution also.

Please, try that and let us know if now it is working as you wanted. Thanks.

sindudas, 2018-12-13:
Hi
It worked as expected!

One last question. When writing Rules, what event is triggered when pressed de Web GUI button? is the same that relay, "Power" ?

Thanks for all.