arendst/Tasmota #4608: RFbridge not recieving any signals anymore

woudenbe, 2018-12-12:
i updated the rf chip firmware with the latest file (20181127) and now the bridge is not able to see any rf signals anymore.
is there any way to get the original RF chip firmware back ?

thanks Hans

meingraham, 2018-12-12:
I believe that the ITEAD firmware is available here - https://github.com/arendst/Sonoff-Tasmota/tree/development/tools/fw_efm8bb1

ascillato2, 2018-12-12:
@woudenbe You can go back to the stock firmware using the link provided by @meingraham 

But anyway, The firmware from Portish works really good. If you want, you can try flash it again.

If you need further assistance with that, please find us in the Tasmota Support Chat.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-12 by ascillato2