arendst/Tasmota #4609: decode-config.py - @v retain minor rev level as previous version of the tool

meingraham, 2018-12-12:
Norbert (@curzon01),

**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
In a recent (October 27, 2018) prior version of decode-config.py, when one used the @v parameter for the output file name, the version would include the minor rev level of the development code version (e.g., 6.2.1.3). The current version only inserts the major release (e.g., 6.3.0) when in fact the firmware version is 6.3.0.14.

**Describe the solution you'd like**
Insert the full release version when using @v for the output file name.

**Describe alternatives you've considered**
None

Regards.

Mike

curzon01, 2018-12-13:
Mike, thanks for reporting: That's not intended, `@v` has to expand to full version at all.

Fixed with PR #4612 - wait until Theo has merged it

ascillato2, 2018-12-13:
Closing this issue as it has been fixed. 

Thanks a lot for your report. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-13 by ascillato2