arendst/Tasmota #4614: Sonoff TH always 2 degree to much

naesle, 2018-12-13:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._
Sonoff TH10 always 2 degree more than other temperature devives tested with my home wetherstation and a homematic thermostat device - all 3 beside weatherstation and homematic device have the same temperature ,sonoff 2 is degree above. Same when copared with a analog temperature device
Tested with both 2 differnt sensors  AM2301 and SI7021 - both alway s2 degree beside

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_yes
- [ ] _Hardware used_ : Sonoff TH10 with AM2301 and SI7021
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._
Maybe add an offset function that value can be corrected

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-12-13:
Hi,

You can correct your temperature value in your home automation software, but if you want to do in Tasmota, please use the example provided at https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#8b-adjust-the-value-of-a-sensor-and-send-it-by-mqtt

Thanks

ascillato2, 2018-12-13:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-13 by ascillato2

naesle, 2018-12-14:
Hello

 

Many thanks for reply

Just a further small question

 

I tried that but nothing happens:

 

What is wrong?

 

Would be gals about a further answer

 

Thanks and regards

 

naesele