arendst/Tasmota #4617: poweronstate 0, but device turns on

prankousky, 2018-12-13:
Hi everybody,

I am running `Sonoff-Tasmota 6.2.1.3 von Theo Arends` on a Sonoff S20. This S20 is connected to multiple socket, which is powered by a Gosund SP1. Let's call the Gosund "Computer" and the S20 "green".

When I turn on computer, green will also turn on (after a short delay). I do not want this. It should always be off, unless manually switched on via mqtt.

I checked in the console: `poweronstate 3` and `powerretain ON`. Poweronstate should be 0, correct? I don't think powerretain should have anything to do with this, but I set it to 0 anyway. So `{poweronstate: 0, powerretain: 0}`. 

Switched computer off and back on again. Short delay, then green turned itself on. Mhhhh. Perhaps some controller (I am using node-red and ioBroker to control my tasmota devices via mqtt) had a retain flag for green's power state. So I manually turned off green. Then I manually turned off computer. Then turned computer back on... Short delay, then green turned itself on as well.

I am not sure how to fix this, but hopefully you guys have some ideas.

Usually, all my Tasmota devices work "on their own" (not sure how to describe this in English; I mean that there is one Tasmota device for one or more devices, but they are not connected to each other in a row; this also means they are always "on" as in "the relay might be off, but the device itself is able to catch commands". In this case, the "green" tasmota is completely cut off from power as soon as the "computer" one shuts off its relay).

Quick addition: I read on another issue and tried the `weblog 4`  command. This is the output

```
00:00:06 MQT: home/tasmota/tasmota/tele/LWT = online (beibehalten)
00:00:06 MQT: home/tasmota/tasmota/cmnd/POWER = 
00:00:06 MQT: abonniere home/tasmota/tasmota/cmnd/#
00:00:06 MQT: abonniere sonoff-/cmnd/#
00:00:06 MQT: abonniere cmnd/tasmota/#
00:00:06 SRC: MQTT
00:00:06 RSL: empfangenes topic home/tasmota/tasmota/cmnd/POWER, Datengröße 2, Daten ON
00:00:06 RSL: Gruppe 0, Index 1, Befehl POWER, Daten ON
00:00:07 MQT: stat/tasmota/RESULT = {"POWER":"ON"}
00:00:07 MQT: stat/tasmota/POWER = ON
00:00:07 MQT: home/tasmota/tasmota/tele/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.1.3","FallbackTopic":"tasmota","GroupTopic":"sonoff-"}
00:00:07 MQT: home/tasmota/tasmota/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"home/tasmota/tasmota","IPAddress":"1"}
00:00:07 MQT: home/tasmota/tasmota/tele/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: home/tasmota/tasmota/stat/RESULT = {"POWER":"ON"}
00:00:07 MQT: home/tasmota/tasmota/stat/POWER = ON
00:00:08 HTP: Hauptmenü
00:00:08 HTP: Konsole
00:00:11 APP: (UTC) Thu Dec 13 15:47:25 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 01:00:00 2018
17:47:29 MQT: home/tasmota/tasmota/tele/STATE = {"Time":"2018-12-13T17:47:28","Uptime":"0T00:00:15","Vcc":3.123,"POWER":"ON","Wifi":{"AP":1,"SSId":"z","BSSId":"8","Channel":11,"RSSI":70}}
```

This does not help me determine why the relay switches on once powered on, though. I have checked, and there is only one MQTT node in node-red pointing to this device. I made sure to set retain to false, so this should not be it. There was also an object in ioBroker pointing to this device, which had the retain flag on; I made sure to set it to false as well. Tested by going through the same steps as mentioned above, still no success. 

Thanks in advance for your input 👍

ascillato, 2018-12-13:
Hi,

Thanks for sharing all your information.

In your log said that the source of the TURN ON is MQTT `SRC: MQTT`
So, that means that, as you had retain to true, it still has the retained values. You need to clear you mqtt broker database. Your issue is in your broker.

_Delete the mosquitto.db and restart the mosquitto service to fix the issue._

Super duplicated issue: https://github.com/arendst/Sonoff-Tasmota/issues/4376, https://github.com/arendst/Sonoff-Tasmota/issues/3339, https://github.com/arendst/Sonoff-Tasmota/issues/2192, https://github.com/arendst/Sonoff-Tasmota/issues/3185, https://github.com/arendst/Sonoff-Tasmota/issues/3066, https://github.com/arendst/Sonoff-Tasmota/issues/2995, https://github.com/arendst/Sonoff-Tasmota/issues/2809, https://github.com/arendst/Sonoff-Tasmota/issues/2140, https://github.com/arendst/Sonoff-Tasmota/issues/739, https://github.com/arendst/Sonoff-Tasmota/issues/628, among others.

ascillato2, 2018-12-13:
Closing this issue as it is duplicated.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-13 by ascillato2